espressif/openocd-esp32 #165: JTAG programming ESP32-C3 devKit M

Wasabinary, 2021-07-12:
Hi,

After successfully flashing my code using a JLINK and JTAG pins( TDO, TDI, TCK and TMS) of an ESP32-S2-SAOLA-1 board, I tried to do the same with an ESP32-C3-DevKitM-1, following the instruction given [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html)

Unfortunately, I am not able to use JTAG with this ESP32-C3 board, even after downloading the newest version of openocd-esp32 and checking my connections multiple times.

I keep receiving an error when launching the command 
> C:/path/to/openocd.exe -f C:/path/to/interface/jlink.cfg -f C:/path/to/target/esp32c3.cfg -c "program_esp test.bin 0x10000 verify exit"

Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32c3.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors

I tried changing the adapter frequency in the jlink.cfg file, but it does not change anything.

Any idea?

igrr, 2021-07-12:
Could you please run `espefuse.py summary` to check eFuse settings and the chip revision? ESP32-C3 starting from revision 3 defaults to JTAG over the built-in USB-SERIAL-JTAG adapter. If you'd like to use it with J-Link, an eFuse should be burned. Sorry that this is not mentioned in the linked document.

Wasabinary, 2021-07-12:
Thanks for your help! I tried running espefuse summary, which outputs a lot of information I don't really understand...

I get the following Config: 

Config fuses:
DIS_RTC_RAM_BOOT (BLOCK0)             Disables boot from RTC RAM                                            = False R/W (0b0)
DIS_ICACHE (BLOCK0)                           Disables ICache                                                                 = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)     Disables Icache when SoC is in Download mode              = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)      Disables forcing chip into Download mode                       = False R/W (0b0)
DIS_CAN (BLOCK0)                                Disables the TWAI Controller hardware                             = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                 Set this bit to vdd spi pin function as gpio                        = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)               Enable btlc gpio                                                                 = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                 Set this bit to enable power glitch function                      = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)       Sample delay configuration of power glitch                     = 0 R/W (0b00)
DIS_LEGACY_SPI_BOOT (BLOCK0)          Disables Legacy SPI boot mode                                        = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)         Selects the default UART for printing boot msg                = UART0 R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)         Sets the default UART boot message output mode          = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)          Force ROM code to send a resume command during SPI = False R/W (0b0) 
... 
Efuse fuses:
WR_DIS (BLOCK0)                          Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                          Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

I also have a part of the output which says something about JTAG:

Jtag Config fuses:
JTAG_SEL_ENABLE (BLOCK0)      = False R/W (0b0)           
Set this bit to enable selection between usb_to_jtag and pad_to_jtag through strapping gpio10 when both reg_dis_usb_jtag and reg_dis_pad_jtag are equal to 0. 

SOFT_DIS_JTAG (BLOCK0)          = 0 R/W (0b00)       
Software disables JTAG. When software disabled, JTAG can be activated temporarily by HMAC peripheral

DIS_PAD_JTAG (BLOCK0)             = False R/W (0b0)       
Permanently disable JTAG access via pads. USB JTAG is controlled separately.

What's the part I need to check ?

Wasabinary, 2021-07-12:
BTW, i checked the chip revision using:
>esptool read_mac

 And i got the following output:
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 3)

Isn't it strange that it says the chip is unknown ?

igrr, 2021-07-12:
> Chip is unknown ESP32-C3 (revision 3)
> Isn't it strange that it says the chip is unknown ?

This should be solved by https://github.com/espressif/esptool/commit/dbbf83fa04032cdede2928266c0d48518da54fa2, but ESP-IDF still uses a slightly older version of esptool, so the fix hasn't propagated there yet.

ESP32-C3, starting from revision 3, supports a built-in USB to SERIAL/JTAG adapter. It should be useable both for console (flash/monitor) and OpenOCD out of the box, provided you connect a USB cable to specific pins, [described here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html). On Windows, there is an extra step to install the driver to make JTAG work.

If you want to use JTAG via GPIOs using an external adapter such as J-link, there are two options.

One option  is to burn `DIS_USB_JTAG` efuse. This is a one-time (irreversible) operation, afterwards JTAG will only be accessible over GPIOs. Built-in USB to Serial part of USB_SERIAL_JTAG peripheral will still work.

The other option is to burn `JTAG_SEL_ENABLE` efuse. This is also an irreversible operation. When this efuse is burned, JTAG connection will be determined by the state of GPIO10 pin on power-on reset. If GPIO10 is low on reset, USB_SERIAL_JTAG will be used. With it is high on reset, JTAG via GPIOs will be used.

Wasabinary, 2021-07-13:
Thank you  @igrr for these very clear explanations! I am now able to connect to my esp32c3 board using my JLINK.

I tried running 
> C:/path/to/openocd.exe -f C:/path/to/interface/jlink.cfg -f C:/path/to/target/esp32c3.cfg -c "program_esp test.bin 0x10000 verify exit"

but I have now another issue:
> .espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts/target/esp_common.cfg: 100: Error: invalid command name "esp"

Something is probably wrong with my config, I'm trying to figure this out. I'll let you know when I'm able to flash my ESP32-C3-DevKitM-1 using JTAG

igrr, 2021-07-13:
I'm sorry i missed the fact you are trying to flash an ESP32-C3 over JTAG, not debug it! This feature was implemented in OpenOCD recently (https://github.com/espressif/openocd-esp32/commit/e950ede18b8e5e7cc6b8a846497fb1817ad47589) but is not part of a release yet. We'll make a new release soon; in the meantime you can use these binaries provided by @gerekon in another issue: https://github.com/espressif/openocd-esp32/issues/162#issuecomment-870041993.

Wasabinary, 2021-07-13:
Oh that's why! I will wait for the new release then. Thank you for your help! 

Closing the issue now

Wasabinary, 2021-08-03:
Now that openocd-esp32 supports JTAG programming on ESP32-C3, I successfully flashed my ESP32-C3-DevKitM-1 after burning the JTAG_SEL_ENABLE efuse.

The thing is, I also wanted to flash ESP32-C3 modules placed on custom PCBs. The thing is, I cannot use USB to burn the JTAG_SEL_ENABLE efuse of these modules, therefore I cannot use my ESP32-C3...

@igrr what's the best way to proceed to successfully use JTAG to flash my modules without using USB ?

gerekon, 2021-08-03:
@Wasabinary I think you do not need USB to program efuses. It is done over UART0 (@igrr @Spritetm please correct me). So you need to have TXD/RXD connected to your PC in some way (e.g. via external USB-UART bridge). Having that you can program, efuses to use ESP32-C3 JTAG pads to connect external JTAG adapter. So you need TXD, RXD and JTAG pins to be accessible on your custom PCB for external connection.

gerekon, 2021-08-03:
To clear it a bit...
When you use `esptool.py` efuses are read/programmed by the stub or ROM code. Communication with stub/ROM is done over UART0.

Spritetm, 2021-08-04:
Almost correct. It's done using either ROM or USB serial/JTAG; the ROM and stub understand both. But yeah, you need to zap the efuse either way before JTAG-over-GPIOs work.

Wasabinary, 2021-08-04:
@gerekon @Spritetm, thank you for your help ! As far as I understand, I could use an FTDI cable connected to TXD/RXD pins in order to burn the efuse ?

Spritetm, 2021-08-04:
Yes. In order to do it fully automatic, you'll also need to connect to EN and IO0 to get into bootloader mode, however.

Wasabinary, 2021-08-04:
@Spritetm I'm using ESP32-C3, don't you mean EN and IO9 ? It seems like IO9 is the one connected to the boot button on ESP32-C3-DevKitM-1

Spritetm, 2021-08-04:
Egh, I keep doing that wrong, mostly because I only use USB nowadays. Yes, it's IO9 on the C3.

Wasabinary, 2021-08-04:
Got it. 
I found this [tutorial](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection) about ESP32 boot mode selection. It is said that IO0 (so IO9 in my case) must be connected to DTR and EN to RTS... But I don't have DTR pin on my FTDI cable, so I connected IO9 to GROUND and successfully activated bootloader mode. Since I manually activated it, I did not connect RTS either, and used the command:
>espefuse --port COM_PORT_FTDI --before no_reset_no_sync summary

It starts connecting, tries detecting the chip type... and fails:
>Connecting...
>Detecting chip type...

>A fatal error occured: Timed out waiting for packet header

I think I miss something, but it looks like I'm close... Please save me @Spritetm!

Spritetm, 2021-08-04:
Not sure what happens there... try adding --no-stub in there perhaps?

Wasabinary, 2021-08-04:
Same behavior with --no-stub option...

Wasabinary, 2021-08-04:
Is there any hardware modification to perform on the ESP32-C3-DevKitM-1 before doing this ?

Spritetm, 2021-08-04:
Ah, that might possibly be the problem: the DevkitM-1 has an USB-serial converter connected to TxD/RxD that interfere. If you remove R21 and R22 and it works, that's your issue.

Wasabinary, 2021-08-04:
Nice! Seems like it's going better without R21 and R22. I also changed
>--before no_reset_no_sync

to 

>--before no_reset

Though, espefuse is still not able to recognize the chip type:
>espefuse --port COM8 --before no_reset_no_sync summary
>Connecting...
>Detecting chip type...

>A fatal error occurred: Unexpected CHIP magic value 0x1b31506f. Failed to autodetect chip type.

So I also added --chip option: 
>espefuse --port COM8 --before no_reset_no_sync --chip esp32c3 summary

Now it finally gives me the expected output with the list of the efuses! 👍 👍 👍  Thank you @Spritetm.
I'll try with an ESP32-C3 module now and I will tell you how it's going.

One question remains though: why is the program not able to determine the chip type ?

Spritetm, 2021-08-04:
Just speculating, but I think I recall something minor changed in the C3 ROM between ECO2 and ECO3... the esptool.py version you have may be new enough to recognize ECO2 but not new enough to recognize ECO3.

Wasabinary, 2021-08-04:
Ok, so nothing to worrying right now. Thank a lot, I will try with the module itself now!

Wasabinary, 2021-08-04:
Well, works like a charm on the module itself. Using an FTDI cable and the reference design given in the ESP32-C3-MINI-1 datasheet, I was able to burn the efuse JTAG_SEL_ENABLE, and the next step is to use JTAG to program the module. 
Thanks a lot for helping me resolving this issue, I think this journey trying to use JTAG with ESP32-C3 could be useful for other people! Great support

closed 2021-08-04 by Wasabinary

andrew-elder, 2022-01-20:
> ESP32-C3, starting from revision 3, supports a built-in USB to SERIAL/JTAG adapter. It should be useable both for console (flash/monitor) and OpenOCD out of the box, provided you connect a USB cable [...]

So, why is there an FDTI USB<-> serial chip on the ESP32-C3-DevKitM-1 board?

igrr, 2022-01-20:
Hi @andrew-elder, indeed ESP32-C3 DevKitM-1 currently comes with a CP2102 USB-UART bridge.

The design of this development board was done when ESP32-C3 support in ESP-IDF was in early stages, and we didn't yet support the "serial" part of USB_SERIAL_JTAG well enough to fully rely on it.

Since then, support for this feature in ESP-IDF has improved. Given that, I think Espressif will eventually remove the CP2102 from the future version of DevKitM board.

With the current generation of DevKitM-1 boards you can use USB_SERIAL_JTAG either through a USB breakout cable (such as [this one](https://www.adafruit.com/product/4448)) or through the built-in USB connector by swapping resistors R1 and R4 into positions of R2 and R3. The latter will disconnect CP2102 from USB and connect ESP32-C3's D+ and D- pins instead.

andrew-elder, 2022-01-20:
@igrr - thanks for the super fast response! I'll be trying your resistor swapping soon here. After the resistors are "moved", would I expect to see two USB derived devices in Linux? I currently have just the /dev/ttyACM0 one.

igrr, 2022-01-20:
You should see one /dev/tty(something) device for the "serial" port. I'm not sure if on Linux it's going to be called ttyACM or not (I'm using macOS). The "jtag" part of the USB_SERIAL_JTAG peripheral doesn't create a /dev/tty* device file, but it will be available to OpenOCD via libusb, should you decide to run OpenOCD.

andrew-elder, 2022-01-23:
At deployment time, if I set the efuse to disable JTAG, will the serial interface still work over OpenOCD, or are they intertwined in such a way that the serial output will be disabled as well.

andrew-elder, 2022-01-26:
I get
```
Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
```
when I try to flash. I'm unsure of the next step(s).