espressif/openocd-esp32 #166: JTAG flash fails using WROVER (DevKitC_V4) and ESP-PROG

TravisRo, 2021-07-24:
This occurs with the 20210721 release binaries on windows 10 using Visual Studio Code.
The 20210401 release binaries work.  

It appears to be a flash writing problem because If I flash with the 20210401 binaries then I can debug with the 20210721.

OpenOCD Log
---------------
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x40134D06, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40134D06, debug_reason=00000000
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing e:/ESP32/bt_spp_acceptor/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 100 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 406 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32.cpu0.flash' size 8192 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 100 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 406 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 8192 KB
Info : Using flash bank 'esp32.cpu1.flash' size 8192 KB
** Programming Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : PROF: Data transferred in 30.929 ms @ 129.328 KB/s
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Programming Finished **
** Verify Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Verified OK **
** Flashing done for partition_table/partition-table.bin **
Flashing e:/ESP32/bt_spp_acceptor/build/bootloader/bootloader.bin at 0x1000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : PROF: Data transferred in 77.385 ms @ 310.138 KB/s
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Programming Finished **
** Verify Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Verified OK **
** Flashing done for bootloader/bootloader.bin **
Flashing e:/ESP32/bt_spp_acceptor/build/bt_spp_acceptor_demo.bin at 0x10000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : PROF: Data transferred in 2358.23 ms @ 257.821 KB/s
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Programming Finished **
** Verify Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Verified OK **
** Flashing done for bt_spp_acceptor_demo.bin **
** Resetting Target **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
0
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Target halted, PC=0x400803C0, debug_reason=00000001
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x400803C0, debug_reason=00000001
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing e:/ESP32/bt_spp_acceptor/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
❌ Error: timed out while waiting for target halted / 4 - 2
Info : esp32.cpu0: Target halted, PC=0x40000DE8, debug_reason=00000000
❌ Error: xtensa❌ _wait_algorithm: not halted 0, pc 0x40000de8, ps 0x60035
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Warn : not e❌ nough working area available(requested 10433)
Error: no working area available, can't alloc space for stub code!
Warn : not enough working area available(reques❌ ted 10433)
Error: no working area available, can't alloc space for stub code!
** Programming Finished **
** Verify Started **
Warn : not e❌ nough working area available(requested 10433)
Error: no working area❌  available, can't alloc space for stub code!
Error: **** Verification failure! ****
Error: target_hash 8cea98...000
Error: file_hash: 7f0b6...b7c820
[Stopped] : OpenOCD Server

gerekon, 2021-07-24:
Hi @TravisRo 

Could you capture the verbose OpenOCD log and attach it here? You need to add `-d3 -l oocd.log` options to its command line.

gerekon, 2021-07-24:
BTW. As I see from your current log. You successfully flashed all binaries
```
Flashing e:/ESP32/bt_spp_acceptor/build/partition_table/partition-table.bin at 0x8000
** Flashing done for partition_table/partition-table.bin **
Flashing e:/ESP32/bt_spp_acceptor/build/bootloader/bootloader.bin at 0x1000
** Flashing done for bootloader/bootloader.bin **
Flashing e:/ESP32/bt_spp_acceptor/build/bt_spp_acceptor_demo.bin at 0x10000
** Flashing done for bt_spp_acceptor_demo.bin **
```
And after that you try to flash it again and that attempt fails.
```
Flashing e:/ESP32/bt_spp_acceptor/build/partition_table/partition-table.bin at 0x8000
x Error: timed out while waiting for target halted / 4 - 2
```
Are you doing this second flashing attempt intentionally?

TravisRo, 2021-07-25:
I'm using visual studio code.  I'm not sure how to change the open ocd command line it uses to do a jtag flash.

I'm not sure why my log is showing 2 flash attempts.  I've tried it several time since and just shows the one failed attempt:

Open OCD Log
---------------
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
HELLO WASHINGTON
oocd.log
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000001
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing e:/ESP32/bt_spp_acceptor/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 94 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 409 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32.cpu0.flash' size 8192 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 94 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 409 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 8192 KB
Info : Using flash bank 'esp32.cpu1.flash' size 8192 KB
** Programming Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : PROF: Data transferred in 30.982 ms @ 129.107 KB/s
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Programming Finished **
** Verify Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Verified OK **
** Flashing done for partition_table/partition-table.bin **
Flashing e:/ESP32/bt_spp_acceptor/build/bootloader/bootloader.bin at 0x1000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : PROF: Data transferred in 76.318 ms @ 314.474 KB/s
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
** Programming Finished **
** Verify Started **
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
❌ Error: **** V❌ erification failure! ****
Error: target_hash 563095...a9c71
Error: file_hash: c5114c...568d39
** Flashing Failed **
-1

TravisRo, 2021-07-25:
[oocd_2021_04_01.log](https://github.com/espressif/openocd-esp32/files/6873442/oocd_2021_04_01.log)
[oocd_2021_07_21_fail.log](https://github.com/espressif/openocd-esp32/files/6873443/oocd_2021_07_21_fail.log)

I think I did this correctly.  I used the following command from a terminal:
`openocd -f board/esp32-wrover-kit-1.8v.cfg -d3 -l oocd.log -c "program_esp bt_spp_acceptor_demo.bin 0x10000 verify exit"
`
The verification *appears* to fail on both of them but I know the code start and runs perfectly with 2021-04-01.  The only difference I see is that 2021-04-01 doesn't actually show it as an "Error" message while 2021-07-21 does.

2021-04-01:
```
Debug: 2831 5227 command.c:143 script_debug(): command - echo echo ** Verify Failed **
User : 2833 5228 command.c:770 jim_echo(): ** Verify Failed **

```
2021-07-21
```
Error: 3128 3749 esp_flash.c:1458 esp_flash_verify_bank_hash(): **** Verification failure! ****
Error: 3129 3749 esp_flash.c:1461 esp_flash_verify_bank_hash(): target_hash fb4a1...7ba71b
Error: 3130 3749 esp_flash.c:1464 esp_flash_verify_bank_hash(): file_hash: 41e475...75c21f

```

TravisRo, 2021-07-25:
long story short, when I use ESP-IDF directly with with it's command prompt that sets all the environment variables, it appears that everything works with both versions.

Sorry for the trouble. There seems to be something else going on here that isn't necessarily related to openocd's ability to flash.

[oocd_2021-04-01.log](https://github.com/espressif/openocd-esp32/files/6874210/oocd_2021-04-01.log)
[oocd_2021-07-21.log](https://github.com/espressif/openocd-esp32/files/6874211/oocd_2021-07-21.log)

closed 2021-07-25 by TravisRo

gerekon, 2021-07-26:
> long story short, when I use ESP-IDF directly with with it's command prompt that sets all the environment variables, it appears that everything works with both versions.

ESP-IDF set `OPENOCD_SCRIPTS` env var to point to OpenOCD scripts dir. So if you want to use non-default version your (not one coming with ESP-IDF) you need to override that variable or speciidy that dir on command line like
`openocd -s <path_to_scripst dir>`.

The newest OpenOCD is not added as default for ESP-IDF yet, so it looks like yuo ran it with `OPENOCD_SCRIPTS` poiting to  `2021-04-01`

TravisRo, 2021-07-26:
I understand why it appears that way but no.  When swapping between the two, I was replacing the entire `openocd-esp32` sub folder.

The problem was the flash voltage.  I was under the impression that it had a 1.8v flash IC under the hood and was using the wrong config file. `esp32-wrover-kit-1.8v.cfg` instead of `esp32-wrover-kit-3.3v.cfg`.  However, even with the right config file, I was still having a lot of inconsistent issues.

I ended up burning the efuses so that it ignored the GPIO12 strapping pin and that ultimately solved the problems.
`espefuse.py --port COM5 set_flash_voltage 3.3V`

For anyone having esp32 JTAG issues, this would be my intial 'automated' response to them:
1) Determine whether your flash IC is 1.8v or 3.3v
2) Use espefuse.py to permanently set the correct voltage.  :)