espressif/openocd-esp32 #167: 【Bug Report/ESP32-C3】JTAG built-in hung after flashing with JTAG built-in

xiaolongba, 2021-07-28:
Hi experts,

Recently, I tested and found that one bug, that is JTAG built-in hung after flashing with JTAG built-in, the details as below:
```c
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing f:/BLE_WIFI/Espressif/SDK/blink/blink/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 28 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 88 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
** Programming Started **
Info : PROF: Data transferred in 107.204 ms @ 37.312 KB/s
** Programming Finished **
** Verify Started **
** Verified OK **
** Flashing done for partition_table/partition-table.bin **
Flashing f:/BLE_WIFI/Espressif/SDK/blink/blink/build/bootloader/bootloader.bin at 0x0
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
** Programming Started **
Info : PROF: Data transferred in 437.879 ms @ 54.8097 KB/s
** Programming Finished **
** Verify Started **
** Verified OK **
** Flashing done for bootloader/bootloader.bin **
Flashing f:/BLE_WIFI/Espressif/SDK/blink/blink/build/blink.bin at 0x10000
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
** Programming Started **
Info : PROF: Data transferred in 2822.58 ms @ 59.5201 KB/s
** Programming Finished **
** Verify Started **
** Verified OK **
** Flashing done for blink.bin **
** Resetting Target **
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
0
❌ Error: esp_usb_jtag: u❌ sb sent only 0 out of 17 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to read DMSTATUS (-4)!
❌ Error: esp_usb_jtag: usb sent only 0 out of 34 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
❌ Error: esp_usb_jtag: usb sent only❌  0 out of 51 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to read DMSTATUS (-4)!
❌ Error: esp_usb_jtag: usb sent only 0 out of❌  68 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
[... 366 lines cut ...]
❌ Error: esp_usb_jtag: usb sent only 0 out of 1394 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag sca❌ n
Error: failed read at 0x11, status=2
❌ Error: esp_usb_jtag: usb sent only 0 out of 1408 bytes.
❌ Error: esp_usb_jtag: usb sent only ❌ 0 out of 1411 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to rea❌ d DMSTATUS (-4)!
Error: esp_usb_jtag: usb sent only 0 o❌ ut of 1428 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
❌ Error: esp_usb_jt❌ ag: usb sent only 0 out of 1445 bytes.
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to read DMSTATUS (-4)!
❌ Error: esp_usb_jtag: usb sent only 0 out of 1462 bytes.
Error: missing data from bitq interface
❌ Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
```
the way to fix this issue is that ESP-IDF Monitor device again, it doesn't work at this moment even if the hardware reset.

gerekon, 2021-07-28:
Hi @xiaolongba 

I tested this on Linux and had no problems.
What does device UART show when you reset targets after programming?

xiaolongba, 2021-07-29:
Hi @gerekon bro

I have tested it with the ESP-IDF extension in the vscode. it can flash with JTAG built-in. however, it hung after resetting the device, as I mentioned above. But it works well if you always open the UART port and check the log information via series tool during the flashing process with JTAG built-in., all information can be outputted normally and there is not hanging issue, while it will hang if you don't open the UART port for monitoring the log information after flashing with JTAG built-in.

can you try it with the ESP-IDF extension in the vscode? it can easy to be duplicated.

thanks.

gerekon, 2021-07-29:
Can not reproduce your problem on Windows. Neither manually nor via VsCode.
It looks like USB_JTAG device gets disconnected in your case for some reason. What esp32c3 board do you use?
Also, I can suggest you avoid using hubs between device and host and/or try different USB ports. And I assume that you do not run Windows on a virtual machine.

xiaolongba, 2021-07-30:
Hi @gerekon Bro,

> What esp32c3 board do you use?
ESP32-C3-DevKitM-1

> Also, I can suggest you avoid using hubs between device and host and/or try different USB ports

No, I plug in the USB port directly, not hubs. Also, I tried the different USB ports, but the results are the same.

> And I assume that you do not run Windows on a virtual machine.

yes

Has the Jtag built-in driver impacted this behavior? also, could you share with me the way you used by manually? and which ways you used in vscode? ESP-IDF Flash device or ESP-IDF build, Flash, and monitor?

gerekon, 2021-07-30:
> Has the Jtag built-in driver impacted this behavior? 

Not sure.

> could you share with me the way you used by manually?

You need:
1) Run the last OpenOCD release from command line, e.g. `c:\Temp\openocd-esp32\bin\openocd.exe -s c:\Temp\openocd-esp32\share\openocd\scripts -f board/esp32c3-builtin.cfg`
2) Open Telnet connection to it. `telnet localhost 4444`
3) Run flashing command as VsCode does: `program_esp_bins c:/projects/esp-idf/examples/get-started/blink/build flasher_args.json reset verify`

> and which ways you used in vscode? ESP-IDF Flash device or ESP-IDF build, Flash, and monitor?

In VsCode I used `ESP-IDF: Build your project` and `ESP-IDF: Flash (with Jtag)`

xiaolongba, 2021-07-30:
> * Run the last OpenOCD release from command line, e.g. `c:\Temp\openocd-esp32\bin\openocd.exe -s c:\Temp\openocd-esp32\share\openocd\scripts -f board/esp32c3-builtin.cfg`
> * Open Telnet connection to it. `telnet localhost 4444`
> * Run flashing command as VsCode does: `program_esp_bins c:/projects/esp-idf/examples/get-started/blink/build flasher_args.json reset verify`

I followed these instructions, the issue still exists, the log information is shown as the following gif file:

[JTAG_Buit-in Hung.zip](https://github.com/espressif/openocd-esp32/files/6908688/JTAG_Buit-in.Hung.zip)

gerekon, 2021-07-30:
Hmm. Do you do anything after running `program_esp_bins`? I see that problem happens after some period of time.
Can you you collect OpenOCD log by running it with `-d3 -l oocd.log`?

gerekon, 2021-07-30:
@xiaolongba Is it possible to share your `bootloader.bin`, `partition-table.bin` and `blink.bin` here?

xiaolongba, 2021-07-30:
> Hmm. Do you do anything after running `program_esp_bins`?

no, I am waiting for the flashing process until the error occurred.

> I see that problem happens after some period of time

yes

> Can you you collect OpenOCD log by running it with `-d3 -l oocd.log`?

sure, [oocd.log](https://github.com/espressif/openocd-esp32/files/6908863/oocd.log)

> Is it possible to share your `bootloader.bin`, `partition-table.bin` and `blink.bin` here?

no problem,
[all bin files.zip](https://github.com/espressif/openocd-esp32/files/6908874/all.bin.files.zip)

xiaolongba, 2021-07-30:
@gerekon Bro

according to your words above, it seems that it is related to the following settings:
- it is normal
[image]

- it is abnormal
[image]

could you double confirm? if it is the same as me, any workarounds about it?

thanks in advance.

gerekon, 2021-07-30:
I reproduced the problem with your binaries. Your app hangs during startup for some reason and get reset by RTC_WDT after some time.
```
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
```
I can reproduce this with connected IDF monitor which shows this weird behavior. UART log is below (includes reset after flashing and one triggered by RTC_WDT). If you have not introduced any changes to `blink` example I suggest you to update your  IDF `master` branch or stick to the latest release tag `4.3`.
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x20000348
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x20e0
load:0x403ce000,len:0xd00
load:0x403d0000,len:0x2e88
entry 0x403ce000
I (53) boot: ESP-IDF v4.4-dev-1849-g8e3e65a47 2nd stage bootloader
I (53) boot: compile time 23:47:26
D (53) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (60) boot: chip revision: 3
D (63) qio_mode: Probing for QIO mode enable...
D (68) qio_mode: Raw SPI flash chip id 0x204016
D (72) qio_mode: Manufacturer ID 0x20 chip ID 0x4016
I (77) qio_mode: Enabling default flash chip QIO
D (83) qio_mode: Initial flash chip status 0x0
D (87) qio_mode: Updated flash chip status 0x2
D (92) qio_mode: Enabling QIO mode...
D (96) boot.esp32c3: magic e9
D (99) boot.esp32c3: segments 03
D (102) boot.esp32c3: spi_mode 02
D (105) boot.esp32c3: spi_speed 0f
D (109) boot.esp32c3: spi_size 02
I (112) boot.esp32c3: SPI Speed      : 80MHz
I (117) boot.esp32c3: SPI Mode       : QIO
I (122) boot.esp32c3: SPI Flash Size : 4MB
D (127) boot: Enabling RTCWDT(9000 ms)
I (130) boot: Enabling RNG early entropy source...
D (136) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (145) boot: mapped partition table 0x8000 at 0x3c008000
D (150) flash_parts: partition table verified, 4 entries
I (156) boot: Partition Table:
I (159) boot: ## Label            Usage          Type ST Offset   Length
D (167) boot: load partition table entry 0x3c008000
D (172) boot: type=1 subtype=2
I (175) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (182) boot: load partition table entry 0x3c008020
D (187) boot: type=1 subtype=1
I (191) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (198) boot: load partition table entry 0x3c008040
D (203) boot: type=0 subtype=0
I (206) boot:  2 factory          factory app      00 00 00010000 00100000
I (214) boot: End of partition table
D (218) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (224) esp_image: reading image header @ 0x10000
D (229) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (236) esp_image: image header: 0xe9 0x06 0x02 0x02 403802d2
V (242) esp_image: loading segment header 0 at offset 0x10018
V (247) esp_image: segment data length 0x73f8 data starts 0x10020
V (254) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3c020020
I (262) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=073f8h ( 29688) map
D (270) esp_image: free data page_count 0x0000003f
D (275) bootloader_flash: mmu set paddr=00010000 count=1 size=73f8 src_addr=10020 src_addr_aligned=10000
V (289) esp_image: loading segment header 1 at offset 0x17418
D (291) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
V (297) esp_image: segment data length 0x13e8 data starts 0x17420
V (304) esp_image: segment 1 map_segment 0 segment_data_offs 0x17420 load_addr 0x3fc8aa00
I (312) esp_image: segment 1: paddr=00017420 vaddr=3fc8aa00 size=013e8h (  5096) load
D (320) esp_image: free data page_count 0x0000003f
D (325) bootloader_flash: mmu set paddr=00010000 count=1 size=13e8 src_addr=17420 src_addr_aligned=10000
V (336) esp_image: loading segment header 2 at offset 0x18808
[... 48 lines cut ...]

I (598) cpu_start: Pro cpu up.
I (613) cpu_start: Pro cpu start user code
I (613) cpu_start: cpu freq: 160000000
I (614) cpu_start: Application information:
I (616) cpu_start: Project name:     blink
I (621) cpu_start: App version:      1
I (625) cpu_start: Compile time:     Jul 22 2021 23:47:20
I (632) cpu_start: ELF file SHA256:  67b607a8aa2a5a6e...
I (638) cpu_start: ESP-IDF:          v4.4-dev-1849-g8e3e65a47
I (644) heap_init: Initializing. RAM available for dynamic allocation:
I (651) heap_init: At 3FC8CC60 len 000333A0 (204 KiB): DRAM
I (657) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (664) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x20e0
load:0x403ce000,len:0xd00
load:0x403d0000,len:0x2e88
entry 0x403ce000
I (31) boot: ESP-IDF v4.4-dev-1849-g8e3e65a47 2nd stage bootloader
```

gerekon, 2021-07-30:
> according to your words above, it seems that it is related to the following settings:
> 
> *

Yes it sems to be the reason why your app hangs trying to write to `USB Serial on USB JTAG controller`

xiaolongba, 2021-08-01:
> Yes it sems to be the reason why your app hangs trying to write to `USB Serial on USB JTAG controller`
> 
> To use it you need to burn some efuses as far as I know,

thanks bro. but could you give me more tips about it? there are several bits about it, But I cannot sure which ones should be burned.

gerekon, 2021-08-02:
@xiaolongba It seems that it needs only selecting an option in menuconfig. But there is a bug in IDF that can lead to execution blocking. The fix is pended and should appear in IDF `master` soon.

xiaolongba, 2021-08-03:
@gerekon thanks for your replies. got it now. what can I only do is to wait and see at this moment. pls, close this ticket if there are not more supplements from your side.

closed 2021-08-03 by gerekon

Tomblarom, 2024-11-01:
So I'm having exactly the same issue on esp-idf 5.3.1! The only workaround is to kill openocd, before flashing. This sucks!

```
❌ Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32c3] Failed DMI read at 0x11; status=2
Error: [esp32c3] polling failed!
```
[image]

erhankur, 2024-11-01:
@Tomblarom If you are a Windows user, we fixed the recovery issue in this release. https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20240726

Please update your OpenOCD with the latest one. 

> jtag/esp_usb_jtag: Re-init usb port when replug or hard reset detected. On Windows, libusb_reset() is not functional. Libusb prints the error AbortPipe (pipe address 02) failed: [22] The device does not recognize the command and returns with a success code. So, OpenOCD still thinks that the revive_device has been completed successfully. But still, continue to get write_error from the endpoint.