espressif/openocd-esp32 #168: Programming ESP32-C3-MINI-1 module using JTAG

Wasabinary, 2021-08-04:
Hi,

Continuing my [journey trying to use JTAG on ESP32-C3](https://github.com/espressif/openocd-esp32/issues/165), I am now trying to flash an ESP32-C3-MINI-1, which is connected following [the reference design of this module](https://www.espressif.com/sites/default/files/documentation/esp32-c3-mini-1_datasheet_en.pdf). 
After successfully burning the **JTAG_SEL_ENABLE** efuse, and pulling it to 3.3V in order to enable JTAG over GPIOs, I'm now trying to flash the "Hello World" ESP-IDF example using a JLINK and the latest openocd-esp32 version.

Unfortunately, I get stuck with error messages that I did not encountered using an ESP32-C3 devkit when running **openocd -f interface\jlink.cfg -f target\esp32c3.cfg -c "program_esp hello-world.bin 0x10000 verify reset exit"**:
>...
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Error: Failed to get flash maps (4294967290)!
Warn : Failed to get flash mappings (-4)!
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
** Programming Started **
Info : PROF: Data transferred in 2375.62 ms @ 63.9832 KB/s
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
shutdown command invoked

Even if it seems like programming succeed, I'm confused by this error message:
>Error: Failed to get flash maps (4294967290)!
Warn : Failed to get flash mappings (-4)!

Moreover, after connecting an FTDI cable to RXD and TXD pins, all I this error printing again and again: 
>invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

Obviously, I did something wrong to my chip...
I thought using JTAG would be very straight forward after burning the efuse, but I'm really struggling to finally succeed at programming ESP32 modules for custom PCB...

Any help would be very welcomed...

igrr, 2021-08-04:
@Wasabinary aside from programming the application binary, have you also programmed the 2nd stage bootloader and the partition table binaries? `invalid header: 0xffffffff` possibly indicates that the 2nd stage bootloader is not written into Flash.

Wasabinary, 2021-08-04:
Oh so I definitely missed a step somewhere... Indeed, I only compiled the hello-world example and tried to flash the binary into my ESP32-C3, and I'm completely unaware of the steps you are referring to... Where can I find the resources about these in order to properly prepare the chip to host my firmware ?

Wasabinary, 2021-08-04:
I think I got it, building the app also creates a bootloader folder and a partition_table folder into the build folder of my projet. Do I need to use JTAG the same way in order to upload bootloader.bin and partition-table.bin into my chip ?

erhankur, 2021-08-04:
@Wasabinary yes you can with -c "program_esp_bins build flasher_args.json verify reset exit":

Wasabinary, 2021-08-04:
Yaaay it works now! The command given by @erhankur did the trick:
>openocd -f interface\jlink.cfg -f target\esp32c3.cfg -c "program_esp_bins build flasher_args.json verify exit" 

It uploaded partition-table.bin, bootloader.bin and hello-world.bin one after the other, and now my application is running fine: no more **flash mapping error** and no **more invalid header error**.

Once again, thanks a lot!

closed 2021-08-04 by Wasabinary

Spatens, 2022-02-07:
I use a esp32dev and it works perfectly not using SPIFFS - the "Hello" project
When I try to embed SPIFFS - I get the same symptoms with not finding the flash maps.
I DID "Build Filesystem image" and selected "Upload Fielsystem image" in PlatformIO. 
And then I get "Error: Failed to get flash maps (4294967290)!"
When I try to issue this one  (I tried to enter my device...):
openocd -f interface\jlink.cfg -f target\esp32dev.cfg -c "program_esp_bins build flasher_args.json verify exit"
Then I get this one:   **Error: Can't find target\esp32dev.cfg**
How do I proceed ? Please any help appreciated...

Wasabinary, 2022-02-07:
Hi,

I can't find the file "target\esp32dev.cfg" you are refering to in my own openocd-esp32 folder. But may be you should try with "target\esp32", that should be fine

Spatens, 2022-02-09:
Hi - thanks for your reply. I have trying to get it to work - I ran into other problems and have not been able to get back yo you.
I had two versions of python, two Espressif IDF`s on top of VS-code with platformIO, my IDF installation is on E: - with numerous irritating references to my windows userID. I still have not solved how to set my path  correct - but I keep trying. I WILL return to you. I deleted ALL old python references and installed all over. 

Now I am simply not able to execute the command: "openocd" from my project directory but have to change to the tool-openocd-esp32" directory and then I execute this:
openocd -f interface\jlink.cfg -f target\esp32.cfg -c "program_esp_bins build flasher_args.json verify exit

The result is this:
_E:/ESP/ESP_IDF/packages/tool-openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:191: Error: build/flasher_args.json: No such file or directory in procedure 'program_esp_bins'
at file "E:/ESP/ESP_IDF/packages/tool-openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 191_

I am pretty sure that it could be my setup - wrong path - but I don't know how to fix it. Some os us needs more... :-)
Hope you will help...

Wasabinary, 2022-02-10:
I think the best way to make sure you don't have path related problem is to use absolute path for everything (including openocd.exe, which is in *.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\bin fodler* in my case. Same think for *.cfg* files and *flasher_args.json*, use absolute paths so you will not have any problem linked to that.

On the other hand, I remember having problems with *program_esp_bins* command, and I managed to flash my ESP32 by flashing *build/bootloader/bootloader.bin*, *build/partition_table/partition-table.bin* and *build/application.bin* one by one with the corresponding offsets. These offsets can be found in *build/flasher_args.json*

gerekon, 2022-02-10:
@Spatens 

> openocd -f interface\jlink.cfg -f target\esp32.cfg -c "program_esp_bins build flasher_args.json verify exit

To make this working you either need to invoke this command from your project dir (implying that `build` dir is there) or use @Wasabinary advice to use absolute path.

Spatens, 2022-02-10:
Thanks Guys.... :-) I'm sure I can make this work for me. Else "I'll be back" as Sshwarzenegger says....

SanjanaNibgoor, 2022-12-27:
Hi,
I have designed a custom PCB for ESP32-C3Mini 1 module. The problem is that the **communication is not happening** to the chip and I am unable to upload the program in Arduino IDE and also in ESP IDF platforms. The hardware setup is fine with respect to this and I have used necessary capacitors and pull-up resistors wherever required. 
I am stuck at this point from from so long.
Please suggest me the necessary measures to resolve this.
When I try to upload, I get the following ERROR.

Sketch uses 646892 bytes (49%) of program storage space. Maximum is 1310720 bytes.
Global variables use 33420 bytes (10%) of dynamic memory, leaving 294260 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3
Serial port COM4
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C3: **No serial data received.**
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
the selected serial port For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
 **does not exist or your board is not connected**

Thanks in advance.