espressif/openocd-esp32 #170: Couldn't halt target before SoC reset

Rajssss, 2021-08-18:
### Environment
- Development Kit: ESP32-DevKitC V4
- Module or chip used: ESP32-WROVER-B
- Debug Adapter: Jlink Edu
- OpenOCD version: v0.10.0-esp32-20210401
- Operating System: Xubuntu 20.04
- Using an IDE?: Nope
- OpenOCD command line: openocd -f interface/jlink.cfg -f board/esp32-wrover.cfg -c "program_esp build/main.bin 0x10000 verify exit"
- JTAG Clock: default
- ESP-IDF version: v4.4.0

### Problem Description
Steps to reproduce the behavior:
1. export idf
2. openocd -f interface/jlink.cfg -f board/esp32-wrover.cfg -c "program_esp build/main.bin 0x10000 verify exit"
3. See error
Jlink is detecting the esp32 fine, but can't reset it for some reasons.

### Debug Logs
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/7004379/openocd_log.txt)

### Expected behavior
Flash successful 

### Screenshots
Wiring: 
[image]

gerekon, 2021-08-18:
Hi @Rajssss 

It looks like chips is continuously reset. Could you try with disconnected TRST. Toggling (pull-down + pull-up) `EN` pin resets entire chip not TAP, so it should not be connected to TRST (TAP reset). OpenOCD handles chip reset via JTAG.

Rajssss, 2021-08-20:
> Hi @Rajssss
> 
> It looks like chips is continuously reset. Could you try with disconnected TRST. Toggling (pull-down + pull-up) `EN` pin resets entire chip not TAP, so it should not be connected to TRST (TAP reset). OpenOCD handles chip reset via JTAG pins.

I did with disconnected TRST, same error.

gerekon, 2021-08-20:
Can you ensure that there is good app image is in flash? You can flash blink example just for tests.

Rajssss, 2021-08-20:
Yes, I use this board regularly, there's nothing wrong with that.

gerekon, 2021-08-20:
From your log I see that cores can not be examined
```
Debug: 637 42 xtensa.c:685 xtensa_examine(): xtensa_examine coreid=0
Debug: 656 42 xtensa.c:698 xtensa_examine(): OCD_ID = fffffffc
```
Can you check that JTAG is not disabled via efuses?
```bash
 espefuse.py -p /dev/ttyUSB1 summary
 ...
 JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = False R/W (0b0)
 ...
```

gerekon, 2021-08-20:
fom docs for esp32-wrover chip
```
Internal pull-up resistor (R9) for MTDI is not populated in the module, as the flash and SRAM in ESP32-WROVER-B
and ESP32-WROVER-IB only support a power voltage of 3.3 V (output by VDD_SDIO).
```
So I think you need to use `-f board/esp32-wroom-32.cfg` instead of `-f board/esp32-wrover.cfg` for your module.

Rajssss, 2021-08-20:
> ```shell
> espefuse.py -p /dev/ttyUSB1 summary
> ```

Here is the o/p:

`
...
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = False R/W (0b0)
...
`

Rajssss, 2021-08-20:
> set ESP32_FLASH_VOLTAGE 3.3

you mean esp-wroom-32.cfg right?
I got the same error with that too.

Rajssss, 2021-08-30:
@gerekon any update on this?

gerekon, 2021-08-30:
@Rajssss There are not much to suggest.

When I tested OpenOCD with Infineon XMC Link (based on SEGGER Jlink) with [Cortex Debug 10 pin connection](https://www.keil.com/support/man/docs/ulinkpro/ulinkpro_cs_connectors.htm) I used:
1)  5 wire connections. 
     - TMS -> IO14
     - TCK -> IO13
     - TDI -> IO12
     - TDO -> IO15
     - VTref -> 3V3
     - GND -> GND (BTW not shown on your picture)
 2) 1-3 Mhz adapter clock

What are the lengths of your wires? Could you check the connection and post OpenOCD verbose log (generated with `-d3 -l oocd.log`) for the corrected command line (using `esp-wroom-32.cfg`)?

Rajssss, 2021-09-01:
> @Rajssss There are not much to suggest.
>
> When I tested OpenOCD with Infineon XMC Link (based on SEGGER Jlink) with [Cortex Debug 10 pin [...]

So, It turned out to be my mistake, that the firmware which was running on the target, have already used the GPIO12 pin (TDI). Sigh...
Thank you, and sorry for wasting your time.

closed 2021-09-01 by Rajssss