espressif/openocd-esp32 #172: JTAG debugging on the Ublox NINA-W106 (ESP32) using EFM32 (JLink)

reiniercoetzer, 2021-08-20:
### Environment
- Development Kit: Ublox NINA-W106
- Module or chip used: ESP32
- Debug Adapter: EFM32 Giant Gecko (JLink)
- OpenOCD version: v0.10.0-esp32-20210721
- Operating System: Ubuntu
- Using an IDE?: Eclipse
- OpenOCD command line: openocd -f board/efm32.cfg "program_esp build/hello-world.bin 0x10000 verify exit"
- JTAG Clock: 1000 Hz
- ESP-IDF version: v4.3

### Problem Description
When running the above command line, I get this error message:

> invalid command name "program_esp"

I am basically trying to determine how to debug the ESP32 with JTAG from EFM32 (Giant Gecko) eval kit, which has a Segger J-Link debbugger. However, the documentation on how this can be done is all a bit sketchy.

I read on the Segger website that using OpenOCD is possible, but that you need a different USB driver. However, I am not sure if this is just applicable to Windows. I could not figure out if it's necessary for Linux as well.

### Debug Logs
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/7020306/openocd_log.txt)

### Expected behavior
I suspect it must be possible to somehow flash and debug the ESP32 chip using JLink interface, but I am just unfamiliar with the inner workings, so I just need some pointers.

### Screenshots
If applicable, add screenshots to help explain your problem.

gerekon, 2021-08-20:
EFM32 is cortex-m based board. `program_esp` is available for ESP32xx chips only

gerekon, 2021-08-20:
If you want to use 3rd party debug adapter with ESP32 chips you need to use the following command line (Jlink example)
`openocd -f interface/jlink.cfg -f target/esp32.cfg -c "program_esp build/hello-world.bin 0x10000 verify exit"`

reiniercoetzer, 2021-08-22:
@gerekon Thanks for clarification, I was mixing up the interface and target. I had to modify the jlink.cfg slightly for working with the EFM32 board, since it can only work at a max clock speed of 1000 kHz. From there on things looked much better.

My next problem is that the target is now not being detected (_Error: JTAG scan chain interrogation failed: all ones_) , which typically indicates faulty wiring. However, I briefly checked that out and the connection from the EFM32 board to the NINA-W106 (ESP32) on a connector level it looks fine. I will have to double check the jumpers on the NINA board.

In the meantime, is there anything specific that needs to be set on the ESP32 to enable JTAG debugging from a SW point of view? I flashed the ESP32 with the get-started blink example, so I assume that should just work if the wiring checks out?

gerekon, 2021-08-23:
> In the meantime, is there anything specific that needs to be set on the ESP32 to enable JTAG debugging from a SW point of view?

If you did not disable JTAG in refuses blink example should work with proper wiring.

closed 2023-12-13 by erhankur