espressif/openocd-esp32 #177: 【ESP32-C3/JTAG Built-In Invalid】Jtag Built-in after burning JTAG_SEL_ENABLE bit

xiaolongba, 2021-09-01:
# Environment
Development Kit: Custom Board 
Module or chip used: ESP32-C3-WROOM-02
Debug Adapter: Jtag Built-in
OpenOCD version: v0.10.0-esp32-20210721
Operating System: Win10
Using an IDE?: Vscode + ESP-IDF Extension
OpenOCD command line: 
```c
openocd-esp32\\bin\\openocd.exe -s tcl -f board/esp32c3-builtin.cfg
```
JTAG Clock: default
ESP-IDF version: ESP-IDF v4.4-dev-1849-g8e3e65a47
# Problem Description
Steps to reproduce the behavior:

1. Type in the OpenOCD command line: openocd-esp32\\bin\\openocd.exe -s tcl -f board/esp32c3-builtin.cfg
2. Error occurred
# Debug Logs
```c
 F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210721\\openocd-esp32\\bin\\openocd.exe -s tcl -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32c3.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Info : Listening on port 3333 for gdb connections
```

# Expected behavior
I followed the instructions mentioned [here](https://github.com/espressif/openocd-esp32/issues/165#issuecomment-878397446), as @igrr Bro said, it could be switched to USB_SERIAL_JTAG again if GPIO10 is low on reset, but the actual situation is impossible to use the Jtag Bulit-in any more. also, the JTAG via GPIOs is always enabled no matter which GPIO levels after burning the JTAG_SEL_ENABLE  bit. again, I looked into the datasheet of ESP32-C3, it described as below:
> EFUSE_STRAP_JTAG_SELSet this bit to enable selection between usb_to_jtag and pad_to_jtagthrough strapping gpio10 when both reg_dis_usb_jtag and reg_dis_pad_jtag are equal to 0. (RO)

However, I tried to burn this bit, the error occurred, as below:
```c
PS F:\BLE_WIFI\Espressif\SDK\blink\blink> F:/Tools/Espressif/IDF_Tools/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe F:\BLE_WIFI\Espressif\SDK\esp-idf-c3\components\esptool_py\esptool\espefuse.py -p COM50 burn_efuse STRAP_JTAG_SEL 1
Connecting....
Detecting chip type... ESP32-C3

A fatal error occurred: Invalid the efuse name 'STRAP_JTAG_SEL'. Available the efuse names: ['WR_DIS', 'RD_DIS', 'DIS_RTC_RAM_BOOT', 'DIS_ICACHE', 'DIS_USB_JTAG', 'DIS_DOWNLOAD_ICACHE', 'DIS_USB_DEVICE', 'DIS_FORCE_DOWNLOAD', 'DIS_USB', 'DIS_CAN', 'JTAG_SEL_ENABLE', 'SOFT_DIS_JTAG', 'DIS_PAD_JTAG', 'DIS_DOWNLOAD_MANUAL_ENCRYPT', 'USB_EXCHG_PINS', 'VDD_SPI_AS_GPIO', 'BTLC_GPIO_ENABLE', 'POWERGLITCH_EN', 'POWER_GLITCH_DSENSE', 'WDT_DELAY_SEL', 'SPI_BOOT_CRYPT_CNT', 'SECURE_BOOT_KEY_REVOKE0', 'SECURE_BOOT_KEY_REVOKE1', 'SECURE_BOOT_KEY_REVOKE2', 'KEY_PURPOSE_0', 'KEY_PURPOSE_1', 'KEY_PURPOSE_2', 'KEY_PURPOSE_3', 'KEY_PURPOSE_4', 'KEY_PURPOSE_5', 'SECURE_BOOT_EN', 'SECURE_BOOT_AGGRESSIVE_REVOKE', 'FLASH_TPUW', 'DIS_DOWNLOAD_MODE', 'DIS_LEGACY_SPI_BOOT', 'UART_PRINT_CHANNEL', 'FLASH_ECC_MODE', 'DIS_USB_DOWNLOAD_MODE', 'ENABLE_SECURITY_DOWNLOAD', 'UART_PRINT_CONTROL', 'PIN_POWER_SELECTION', 'FLASH_TYPE', 'FLASH_PAGE_SIZE', 'FLASH_ECC_EN', 'FORCE_SEND_RESUME', 'SECURE_VERSION', 'MAC', 'SPI_PAD_CONFIG_CLK', 'SPI_PAD_CONFIG_Q', 'SPI_PAD_CONFIG_D', 'SPI_PAD_CONFIG_CS', 'SPI_PAD_CONFIG_HD', 'SPI_PAD_CONFIG_WP', 'SPI_PAD_CONFIG_DQS', 'SPI_PAD_CONFIG_D4', 'SPI_PAD_CONFIG_D5', 'SPI_PAD_CONFIG_D6', 'SPI_PAD_CONFIG_D7', 'WAFER_VERSION', 'PKG_VERSION', 'BLOCK1_VERSION', 'OPTIONAL_UNIQUE_ID', 'BLOCK2_VERSION', 'BLOCK_USR_DATA', 'BLOCK_KEY0', 'BLOCK_KEY1', 'BLOCK_KEY2', 'BLOCK_KEY3', 'BLOCK_KEY4', 'BLOCK_KEY5', 'BLOCK_SYS_DATA2']
```

so what I expected is that the Jtag Built-in functionality can be switched through GPIO10, as said as  [here](https://github.com/espressif/openocd-esp32/issues/165#issuecomment-878397446)
# Screenshots
[image]

# Complain
 In terms of environment setup, I encountered lots of issues, and took me lots of time sometimes,  the deeper research, the more questions. I tried to resolve the issues by myself, but I found that it is difficult to do that as the contents docs mentioned are wrong or missing sometimes. also, the debug issue for ESP32-C3 with Jtag Built-in,  & esp-idf extension, I found that lots of guys faced this issue, but it is very strange for me as it is general for users for this issue. anyway, I'm going crazy.

At last, Can you help me double-check this situation?

big thanks.

gerekon, 2021-09-01:
Hi @xiaolongba 

Could you post the output of  `espefuse.py -p COM50 summary`?

xiaolongba, 2021-09-02:
@gerekon 

sure, I have checked these efuse bits, the details as below:
```c
PS F:\BLE_WIFI\Espressif\SDK\blink\blink> F:/Tools/Espressif/IDF_Tools/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe F:\BLE_WIFI\Espressif\SDK\esp-idf-c3\components\esptool_py\esptool\espefuse.py -p COM50 summary
Connecting....
Detecting chip type... ESP32-C3
espefuse.py v3.2-dev
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_RTC_RAM_BOOT (BLOCK0)                Disables boot from RTC RAM                         = False R/W (0b0)
DIS_ICACHE (BLOCK0)                      Disables ICache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)             Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)              Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                         Disables the TWAI Controller hardware              = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                 Set this bit to vdd spi pin function as gpio       = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                Enable btlc gpio                                   = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                  Set this bit to enable power glitch function       = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)             Sample delay configuration of power glitch         = 0 R/W (0b00)
DIS_LEGACY_SPI_BOOT (BLOCK0)             Disables Legacy SPI boot mode                      = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)              Selects the default UART for printing boot msg     = UART0 R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)              Sets the default UART boot message output mode     = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)               Force ROM code to send a resume command during SPI = False R/W (0b0)
                                          bootduring SPI boot
BLOCK_USR_DATA (BLOCK3)                  User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                          Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                          Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Flash Config fuses:
FLASH_TPUW (BLOCK0)                      Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                          unit is (ms/2). When the value is 15, delay is 7.
                                         5 ms
FLASH_ECC_MODE (BLOCK0)                  Set this bit to set flsah ecc mode.
   = flash ecc 16to18 byte mode R/W (0b0)
FLASH_TYPE (BLOCK0)                      Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                 Flash page size                                    = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                    Enable ECC for flash boot                          = False R/W (0b0)

Identity fuses:
SECURE_VERSION (BLOCK0)                  Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                         ure)
MAC (BLOCK1)                             Factory MAC Address
   = 7c:df:a1:bb:e4:08: (OK) R/W
WAFER_VERSION (BLOCK1)                   WAFER version                                      = 3 R/W (0b011)
PKG_VERSION (BLOCK1)                     Package version                                    = 8 R/W (0x8)
BLOCK1_VERSION (BLOCK1)                  BLOCK1 efuse version                               = 0 R/W (0b000)
OPTIONAL_UNIQUE_ID (BLOCK2)(0 errors):   Optional unique 128-bit ID
   = 0c 91 de 66 c8 ce de ca af 5e f1 7e 4d fb 9c a8 R/W
BLOCK2_VERSION (BLOCK2)                  Version of BLOCK2                                  = No calibration R/W (0b000)

Jtag Config fuses:
JTAG_SEL_ENABLE (BLOCK0)                 Set this bit to enable selection between usb_to_jt = True R/W (0b1)
                                         ag and pad_to_jtag through strapping gpio10 when b
                                         oth reg_dis_usb_jtag and reg_dis_pad_jtag are equa
                                         l to 0.
SOFT_DIS_JTAG (BLOCK0)                   Software disables JTAG. When software disabled, JT = 0 R/W (0b00)
                                         AG can be activated temporarily by HMAC peripheral
DIS_PAD_JTAG (BLOCK0)                    Permanently disable JTAG access via pads. USB JTAG = False R/W (0b0)
                                          is controlled separately.

Security fuses:
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)     Disables flash encryption when in download boot mo = False R/W (0b0)
                                         des
SPI_BOOT_CRYPT_CNT (BLOCK0)              Enables encryption and decryption, when an SPI boo = Disable R/W (0b000)
                                         t mode is set. Enabled when 1 or 3 bits are set,di
                                         sabled otherwise
[... 42 lines cut ...]
SPI_PAD_CONFIG_CLK (BLOCK1)              SPI CLK pad                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_Q (BLOCK1)                SPI Q (D1) pad                                     = 0 R/W (0b000000)
[... cut ...]
DIS_USB (BLOCK0)                         Disables the USB OTG hardware                      = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                  Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)           Disables use of USB in UART download boot mode     = False R/W (0b0)

Vdd_Spi Config fuses:
PIN_POWER_SELECTION (BLOCK0)             GPIO33-GPIO37 power supply selection in ROM code   = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                   Selects RTC WDT timeout threshold at startup       = False R/W (0b0)
```

gerekon, 2021-09-03:
Hi @xiaolongba 

Thanks for your findings. I can confirm that issue exists.

xiaolongba, 2021-09-03:
Hi @gerekon Bro

this bug is related to  HW or SW? if HW, that means that users can only select the internal JTAG or external JTAG and it is impossible to use both through GPIO10.

gerekon, 2021-09-03:
Our HW team confirmed that this is HW bug in silicon.  So this functionality does not work in rev3 chips. The docs will be updated. Sorry for the inconvenience.

gerekon, 2021-09-03:
@igrr Should we move this issue to IDF repo? Since it does not belong to OpenOCD.

xiaolongba, 2021-09-03:
@gerekon Got it. clear, Big thanks.