espressif/openocd-esp32 #178: ESP32S3 openocd debugging

skj2021, 2021-09-20:
I am trying to configure openocd for debugging with ESP32S3 module. I am using ESP32-S3-DevKitC-1. I am using eclipse for my debugging. My configuration to start server on cmd prompt is openocd.exe -f interface\esp_usb_jtag.cfg -f board\esp32s3-builtin.cfg and I use GDB Hardware debugging onn eclipse to set the gdb client. GDB command xtensa-esp32s3-elf-gdb and GDB Connection STring is localhost:3333.
But when I run debug configuration I get
Error: esp_usb_jtag: usb sent only 0 out of 31 bytes.
Error: missing data from bitq interface
Error: esp_usb_jtag: usb sent only 0 out of 50 bytes.
Error: missing data from bitq interface
Error: Failed to exec JTAG queue!
Error: Failed to read debug stubs address location (-104)!

I understand that all the openocd features are not supported currently for ESP32S3 but I assume that the basic debugging part is supported. So i was expecting it to work. I am not sure if I am missing any configuration? Any inputs on this? Thanks.

gerekon, 2021-09-27:
Hi @skj2021 

It looks like problem with driver on Winodws. Can you try to remove it and install using [IDF Tools Installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/windows-setup.html#esp-idf-tools-installer).

NateZimmer, 2021-10-09:
Out of curiosity, did you find the documentation for that command? `openocd.exe -f interface\esp_usb_jtag.cfg -f board\esp32s3-builtin.cfg` The latest documentation is wrong at https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#run-openocd citing files that don't exist in the latest openocd release by espressif or in the latest IDF Tools Installer (unlike your command which is correct). So... hunting correct documentation
I'm stuck a bit earlier in the process. I see the CDC USB driver but I can't find any interface for JTAG despite installing the latest tools. Nothing visible via zadig either.

gerekon, 2021-10-11:
@NateZimmer 

> citing files that don't exist in the latest openocd release

Sorry for that. This is a mistake in docs. There are two board config files for S3: 
- `board\esp32s3-builtin.cfg`  for working via builtin USB_JTAG
- `board\esp32s3-ftdi.cfg`  for working via externally connected FTDI-based probe like ESP-PROG

> I see the CDC USB driver but I can't find any interface for JTAG despite installing the latest tools. Nothing visible via zadig either.

Maybe it was mentioned as `Unknown device` in the system because driver have not been installed properly?

BTW. Could you post efuse configs for your board (output of `espefuse.py -p COMxx summary`)?

gerekon, 2021-11-03:
Closed by dbede94268ee26360524d418d6a27bd544d38e49 and 9e06f3e25ec37159124fec9163e27de858ca41bf

closed 2021-11-03 by gerekon