espressif/openocd-esp32 #181: riscv32-esp-elf-gdb requires libpython2.7

jeancf, 2021-10-01:
### Environment
- chipset: esp32-c3
- Operating System: linux
- ESP-IDF version: master

### Problem Description
Steps to reproduce the behavior:
1. Install idf master branch on a system that does not have python 2.7 installed
2. execute `idf.py gdb` in a console
```
Executing action: gdb
riscv32-esp-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
```

### Expected behavior
Since python 2 reached End-of-life on Jan 1 2020 and is no longer supported, I did not expect a dependency on it. I  would prefer not to have to install it on my system just to run gdb.
I see that a similar issue (https://github.com/espressif/esp-idf/issues/5284) has already been reported for extensa.
Can `libpython2.7` be included in the gdb package maybe?

thanks.

gerekon, 2021-10-01:
Hi @jeancf,

Yes, this is a bit bgger issue. In general GDB Python support may require not only libraries but also packages for Python with respective version. We plan to solve this problem maybe by providing several GDB binaries built for all supported Python versions. Minor Python versions are not compatible too.
arm-none-eabi- toolchain suggests similar approach https://xpack.github.io/arm-none-eabi-gcc/python/.

igrr, 2021-10-01:
This issue is currently tracked in https://github.com/espressif/esp-idf/issues/6334 (although the title is about xtensa-esp32-elf-gdb, it's the same problem).
@jeancf I'm going to close this issue as duplicate; please click "subscribe" button in the linked issue to get notified when the issue is resolved.

closed 2021-10-01 by igrr