espressif/openocd-esp32 #190: "Memory protection is enabled. Reset target to disable it..". (debugging works, memory protection disabled)

ad7718, 2021-11-05:
### Environment
- Development Kit: Ai-thinker esp32-c3
- Module or chip used: esp32-c3-32s
- Debug Adapter: builtin esp32 jtag
- OpenOCD version: v0.10.0-esp32-20210902-33-g65b7bf50
- Operating System: debian 11 (MX linux 21)
- Using an IDE?: VS Code + platformio
- OpenOCD command line: openocd -f board/esp32c3-builtin.cfg
- JTAG Clock: 5000 kHz
- ESP-IDF version: v4.3

### Problem Description
Steps to reproduce the behavior:
1. Disable memory protection in menuconfig
2. Start debugging
3. Open debug terminal
4. See warning: "Memory protection is enabled. Reset target to disable it..."

### Debug Logs
Strange, but no warning with "-l openocd_log.txt -d3" options.
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/7485830/openocd_log.txt)

### Expected behavior
No warning

ad7718, 2021-11-05:
Also there is "Warn : No symbols for FreeRTOS!" warning, and seems it's also doesn't affect debugging.

gerekon, 2021-11-05:
@ad7718 
That warning was added intentionally to notify the user that the target will be reset by OpenOCD to disable memory protection. It does not affect functionality.
Is that warning is the only problem you see?

ad7718, 2021-11-05:
@gerekon
Yes, I think it must not say that memory protection is enabled if it's enabled, or we need different message that only reminds about memory protection.

gerekon, 2021-11-05:
Ok. Got it. It looks like some memprot bits remain set https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp32c3.cfg#L121. I think we need to limit checked bits for ESP32-C3 to ones in `SENSITIVE_CORE_0_xxx_PMS_MONITOR_1_REG`

Thanks, for the findings!

ad7718, 2021-11-05:
Thank you, glad it helped. Maybe it's offtop, but what do you think about the FreeRTOS warning? Is it openocd or IDF related?

closed 2021-11-06 by espressif-bot