espressif/openocd-esp32 #192: ESP32-C3 built-in JTAG port only showing when in download mode

miroslavpetrov, 2021-11-08:
### Environment
- Development Kit: [ESP32-C3-DevKitM-1]
- Module or chip used: [ESP32-C3-MINI]
- Debug Adapter: [USB JTAG]
- OpenOCD version: [v0.10.0-esp32-20210902]
- Operating System: [MacOS and Windows]
- Using an IDE?: [VScode]
- OpenOCD command line: [openocd -f board/esp32c3-builtin.cfg]
- JTAG Clock: [default]
- ESP-IDF version: [master]

### Problem Description

I am using GPIO 18 and GPIO 19 on the devkitM and a usb cable to connect to Openocd.
The problem is that the Serial/JTAG debug device is only detected by the PC or Mac when the ESP32-C3 is in download mode(powered on while GPIO9 is LOW)

```
 USB JTAG/serial debug unit:
  Product ID:	0x1001
  Vendor ID:	0x303a
  Version:	1.01
  Serial Number:	7C:DF:A1:66:DD:1C
  Speed:	Up to 12 Mb/s
  Manufacturer:	Espressif
  Location ID:	0x14310000 / 24
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```

I can successfully upload firmware from both the Serial and JTAG ports but nothing else. Its impossible to debug the firmware simply because the JTAG port is not presented to the PC or Mac when the ESP32-C3 is in normal boot mode.

Is that some limitation of I am missing something? I double checked the documentation and this weird behaviour is not mentioned anywhere.

igrr, 2021-11-08:
Hi @miroslavpetrov, which example/application and which ESP-IDF commit are you using?

miroslavpetrov, 2021-11-08:
Hi @igrr! I am the ESP-IDF commit is d0dd9ce1b9825c29ae2b462556c4bf3965ab774b.

My application is based on the switch example in ESP Rainmaker - https://github.com/espressif/esp-rainmaker/tree/master/examples/switch

igrr, 2021-12-13:
@miroslavpetrov I've overlooked this at first, but it seems that the `switch` example configures GPIO19 as an output: https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_driver.c#L25. This is one of the USB pins on ESP32-C3, so USB interface doesn't work after the pin is reconfigured.

closed 2023-01-17 by erhankur