espressif/openocd-esp32 #193: Error: Invalid offset! when flashing at offset 0

balix-microej, 2021-11-09:
### Environment
- Development Kit: ESP32-S2-Saola-1
- Module or chip used: ESP32-S2-WROVER
- Debug Adapter: Segger J-Link
- OpenOCD version: v0.10.0-esp32-20210401
- Operating System: Windows
- Using an IDE?: No
- OpenOCD command line: openocd.exe -f .\interface\jlink.cfg -f .\target\esp32s2.cfg -c "program C:/../application-combined.bin 0x0 verify exit"
- JTAG Clock: default
- ESP-IDF version: v4.2.2

### Problem Description
Steps to reproduce the behavior:
1. Combine binary files produced by ESP IDF to be able to flash to offset 0 (script to combine files is attached).
[combine_binaries.py.txt](https://github.com/espressif/openocd-esp32/files/7503317/combine_binaries.py.txt)
2. Connect a Segger J-Link probe to the dev kit
3. Run the following command (with the binary with files combined): `openocd.exe -f .\interface\jlink.cfg -f .\target\esp32s2.cfg -c "program C:/../application-combined.bin 0x0 verify exit"`
4. Programmation of the device failed with error `Invalid offset`.

### Debug Logs
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/7503311/openocd_log.txt)

### Expected behavior
The device is programmed with the binary provided.

### Comments
Maybe my combined binary file is wrong? Is there any other way provided "officially" to flash the entire binary files at once? One workaround I found is to program the bootloader, partition table, ota data and application sequentially using the different binary files and offsets. But this doesn't always work because sometimes the device will reset indefinitely.

gerekon, 2021-11-09:
@balix-microej 
Not sure that it will work in this way.
What binaries did you combine?  What command did you use to create it?
If you want to combine  `bootloader.bin @ 0x1000` + `partition-table.bin @ 0x8000` + `app @ 0x10000` it is better to combine them shifted by `0x1000` and write combined binary at `0x1000` because the first 4KB of flash are reserved for secure boot digest file at offset `0x0`. For Expressif Xtensa chips OpenOCD forbids to write flash at offsets below `0x1000`.

balix-microej, 2021-11-17:
A custom script is used to combine: the bootloader, partition table, ota initial data and application binaries.

I will try with this shift of 4KB thanks!

I will let you know if this solves the issue.

erhankur, 2023-01-19:
@balix-microej Can we close this issue?

erhankur, 2023-02-11:
Closed because of the inactivity.

closed 2023-02-11 by erhankur