espressif/openocd-esp32 #194: Need To Start Debug Twice Because First Attempt Always Fails

vindicatorr, 2021-11-12:
### Environment
- Development Kit: []
- Module or chip used: [ESP32-C3]
- Debug Adapter: [C3 Embedded USB]
- OpenOCD version: [1.040c2f021-dirty]
- Operating System: [Linux]
- Using an IDE?: [Visual Studio Code]
- OpenOCD command line: <pathTo>/openocd-esp32/src/openocd --search <pathTo>/openocd-esp32/tcl/ --file <pathTo>/openocd-esp32/tcl/board/esp32c3-builtin.cfg
- JTAG Clock: [default]
- ESP-IDF version: [v5.0-dev-266-g58283e385]

### Problem Description
I have to restart debugging after the initial attempt fails. This happens every time.
Ref: https://github.com/microsoft/vscode-cpptools/issues/1075
Ref: https://gojimmypi.blogspot.com/2017/05/vscode-jtag-debugging-of-esp32-part-1.html

Steps to reproduce the behavior:
1. Build/Flash any simple app
Launch.json:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "(openocd) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/<project>.elf",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "debugServerPath": "<pathTo>/openocd-esp32/src/openocd",
            "debugServerArgs": "-d3 --search <pathTo>/openocd-esp32/tcl/ --file <pathTo>/openocd-esp32/tcl/board/esp32c3-builtin.cfg",
            "serverStarted": ".*Listening\\ on\\ port\\ 3333.*",
            "serverLaunchTimeout": 10000,
            "MIMode": "gdb",
            "miDebuggerPath": "<pathTo>/crosstool-NG/builds/riscv32-esp-elf/bin/riscv32-esp-elf-gdb",
            "setupCommands": [
                {
                    "text": "set remotelogfile ${workspaceFolder}/gdb_failed_log.txt"
                },
                {
                    "description": "enable pretty printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": false
                },
                {
                    "text": "file '${workspaceFolder}/build/<project>.elf'"
                },
                {
                    "text": "target remote :3333"
                },
                {
                    "text": "set remote hardware-watchpoint-limit 2",
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "flushregs"
                }
                // {
                //     "text": "tb app_main"
                // }
                // {
                //     "text": "c"
                // }
                //Debugging ALWAYS fails when I have these set
            ],
            // "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "externalConsole": false,
            "logging": {
                "engineLogging": true,
                "exceptions": true,
                "trace": true
            }
        }
    ]
}
```
2. Launch VSCode debugger
3. See error
4. Launch VSCode debugger
5. Continue debugging successfully

### Debug Logs
[gdb_failed_log.txt](https://github.com/espressif/openocd-esp32/files/7530795/gdb_failed_log.txt)
[openocd_failed_log.txt.tar.gz](https://github.com/espressif/openocd-esp32/files/7530838/openocd_failed_log.txt.tar.gz)

### Expected behavior
It should only take one attempt to begin debugging.

### Screenshots
If applicable, add screenshots to help explain your problem.

gerekon, 2021-11-17:
HI @vindicatorr 
1) It looks like you use some custom version of OpenOCD `1.040c2f021-dirty`. Could you try with the latest released version [v0.10.0-esp32-20211111](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20211111)?
2) You use "cppdbg". As far as I know it has issues when stepping over code, so we use our own DAP server in our [ESP-IDF extension](https://github.com/espressif/vscode-esp-idf-extension) for debugging. Could you try it?

vindicatorr, 2021-11-17:
1. ? I fetched/pulled and saw there was only one change. The only reason it's "dirty" is because I edit `configure.ac` so it builds.
It was from a previous issue relating to `[openocd], [ ]` which I see hasn't been corrected yet (https://github.com/espressif/openocd-esp32/issues/187).

2. I haven't had much issue with cppdbg (as far as I know) and prefer to go as "native" as I reasonably can. Same reason why I avoid "west" when it comes to zephyr (which I also pulled back from, since it was lacking).
And if I remember right, I had issues with the extension dealing with my custom locations or something. In short, I left the extension because I wasn't pleased with it, and I'm fine without it, other than having this little quirk which I just deal with knowing what it's about.

gerekon, 2021-11-18:
> I haven't had much issue with cppdbg (as far as I know) and prefer to go as "native" as I reasonably can.

Ok. So since I do not see any OpenOCD error reported in your logs. Can I ask you, in this case, to localize the problem and try debugging from the command line? If the problem still exists I can dig into it if not I will ask our VsCode guys to have a look at it.

vindicatorr, 2021-11-18:
Is there something specific I should be doing with gdb that you have in mind in order to flesh this out?
Terminal 1:
```
<pathTo>/openocd-esp32/src/openocd --search <pathTo>/openocd-esp32/tcl/ --file <pathTo>/openocd-esp32/tcl/board/esp32c3-builtin.cfg
```

Terminal 2:
```
$ EDITOR=/usr/bin/nano <pathTo>/riscv32-esp-elf/bin/riscv32-esp-elf-gdb -ex "file <pathTo>/<project>.elf" -ex "target remote :3333" -ex "set remote hardware-watchpoint-limit 2" -ex "monitor reset halt" -ex "flushregs"
GNU gdb (crosstool-NG esp-2021r2) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Undefined command: "-enable-pretty-printing".  Try "help".
Reading symbols from <pathTo>/<project>.elf...
Remote debugging using :3333
0x40000000 in ?? ()
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Register cache flushed.
(gdb) list
1       /*
2        * SPDX-FileCopyrightText: 2017-2021 Espressif Systems (Shanghai) CO LTD
3        *
4        * SPDX-License-Identifier: Apache-2.0
5        */
6
7       #include <assert.h>
8       #include <sys/param.h>
9       #include "esp_ota_ops.h"
10      #include "esp_attr.h"
(gdb) bt
#0  0x40000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) tbreak app_main
Temporary breakpoint 1 at 0x42005b4a: file <pathTo>/main/base.c, line 6.
(gdb) list
11      #include "sdkconfig.h"
12
13      // Application version info
14      const __attribute__((section(".rodata_desc"))) esp_app_desc_t esp_app_desc = {
15          .magic_word = ESP_APP_DESC_MAGIC_WORD,
16      #ifdef CONFIG_APP_EXCLUDE_PROJECT_VER_VAR
17          .version = "",
18      #else
19          .version = PROJECT_VER,
20      #endif
(gdb) c
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
[New Thread 1070134084]
[New Thread 1070129232]
[Switching to Thread 1070133740]

Thread 1 hit Temporary breakpoint 1, app_main ()
    at <pathTo>/main/base.c:6
6           ESP_LOGW("testTag", "test message");
(gdb) bt
#0  app_main () at <pathTo>/main/base.c:6
#1  0x4201231e in main_task (args=<optimized out>)
    at <pathTo>/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127
#2  0x40385da6 in vPortSetInterruptMask ()
    at <pathTo>/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:327
Backtrace stopped: frame did not save the PC
```
From this, it doesn't look like it just drops out.
And after every test, I always pressed the reset button on the board before starting up openocd and then gdb again.

tonyjackson-srt, 2022-03-12:
Interesting.  I have almost identical setup (clean v0.10.0-esp32-20210401) and experience the same - have to start the debugger twice from vscode for it connect successfully.

erhankur, 2023-01-17:
Closing due to inactivity. If the issues exist with the latest vscode extension and OpenOCD release, feel free to re-open.

closed 2023-01-17 by erhankur

vindicatorr, 2023-01-20:
The issue isn't fixed, but I guess it can remain closed based on the `latest vscode extension` part of your statement.
I and the others (I presume) aren't using the extension.
The issue still remains open in the first reference in my OP.
```
GNU gdb (crosstool-NG esp-2022r1) 9.2.90.20200913-git
Open On-Chip Debugger v0.11.0-esp32-20221026-2-g3f08cc59 (2023-01-19-20:06)
ms-vscode.cpptools v1.13.3-main
```