espressif/openocd-esp32 #195: GDB connection rejected

CodeOn-ArK, 2021-11-25:
### Environment
- Development Kit: [ ESP32-DevKitC]
- Module or chip used: [ ESP32-WROOM-32]
- Debug Adapter: [ FT232H]
- OpenOCD version: [v0.10.0-esp32-20210401]
- Operating System: [ Linux]
- Using an IDE: [ Eclipse]
- OpenOCD command line: [-s ${openocd_path}/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover-kit-1.8v.cfg]
- JTAG Clock: [ default]
- ESP-IDF version: [ v4.2]

### Problem Description
Debugging Using the default parameters set by eclipse IDE. Whenever I debug using eclipse IDE and FT232H based JTAG debugger I get some errs. And the error is untracable. Somedays I get the error, somedays It just debugs fine without any issue.

### Debug Logs
``` Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
Flashing /home/ark/ESP32-Projects/001tasks/build/partition_table/partition-table.bin at 0x8000
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Error: Couldn't halt target before SoC reset
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Error: Target not examined yet
Error executing event gdb-attach on target esp32.cpu0:
embedded:startup.tcl:449: Error: 
in procedure 'program_esp' called at file "/home/ark/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 189
in procedure 'program_error' called at file "/home/ark/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 85
at file "embedded:startup.tcl", line 449
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked ```

### Expected behavior
Debug session should start, which normally does

CodeOn-ArK, 2021-11-25:
Here attaching another set of logs I get upon testing the latest release

``` ❯ ../../../bin/openocd  -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover-kit-1.8v.cfg -c 'ftdi_vid_pid 0x0403 0x6014'

Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2000 kHz

Warn : Interface already configured, ignoring
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections

CodeOn-ArK, 2021-11-25:
It works again 😂😂

``` Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
Flashing /home/ark/ESP32-Projects/hello_world/build/partition_table/partition-table.bin at 0x8000
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x400E3082, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu1.flash' size 4096 KB
** Programming Started **
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : PROF: Data transferred in 22.53 ms @ 177.541 KB/s
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
** Programming Finished **
** Verify Started **
Info : PROF: Data transferred in 19.736 ms @ 152.006 KB/s
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
** Verified OK **
** Flashing done for partition_table/partition-table.bin **

gerekon, 2021-11-25:
@CodeOn-ArK 

This is from esp32-wroom datsheet 
```
The module integrates a 3.3 V SPI flash, so the pin MTDI cannot be set to 1 when the module is powered up
```
Could you try to use proper board config file?
```bash
-s ${openocd_path}/share/openocd/scripts -f board/esp32-wrover-kit-3.3v.cfg]
```

CodeOn-ArK, 2021-11-30:
> @CodeOn-ArK 
> 
> This is from esp32-wroom datsheet 
> ```
> The module integrates a 3.3 V SPI flash, so the pin MTDI cannot be set to 1 when the module is powered up
> ```
> Could you try to use proper board config file?
> ```bash
> -s ${openocd_path}/share/openocd/scripts -f board/esp32-wrover-kit-3.3v.cfg]
> ```

Mine one works, only that I have to debug it many times before getting it working

gerekon, 2021-11-30:
@CodeOn-ArK 
So was the problem solved?
In the case you reported it working there are JTAG connection errors. What debug adapter and JTAG wire length do you use?
```
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
```

CodeOn-ArK, 2021-12-01:
Yea problem solved, wire length is the normal jumper wires only, and debugger ft232h.
It seems there is no connection issue rather I have to press the debug button 2-3 times before it fires up.

closed 2021-12-05 by CodeOn-ArK