espressif/openocd-esp32 #196: On configure, enabling --enable-sysfsgpio option will break make

MU-Software, 2021-11-25:
### Environment
- Development Kit: Ai-Thinker ESP32-CAM
- Module or chip used: ESP32-WROOM-32
- Debug Adapter: `Raspberry Pi Zero W GPIO as Debugger`
- OpenOCD version: `v0.10.0-esp32-20211111`, `latest on this repo`
- Operating System: Raspberry Pi OS Bullseye (Kernel version: 5.10)
- Using an IDE?: None
- OpenOCD command line: `make`
- JTAG Clock: default
- ESP-IDF version: None

### Problem Description
Steps to reproduce the behavior:
1. `git clone -d v0.10.0-esp32-20211111 https://github.com/espressif/openocd-esp32.git`
2. `cd openocd-esp32`
3. `./bootstrap`
4. `./configure --enable-sysfsgpio --enable-bcm2835gpio`
5. `make`

This problem raises when we enable sysfsgpio options.  
`bitbang_swd` is used both on `bitbang.c` file and `stsfsgpio.c` file, so that variable must be a extern variable, but on `src/jtag/drivers/bitbang.h`, `bitbang_swd` is just a `const struct`.  
On latest OpenOCD repo, `bitbang_swd` is `extern const struct variable`, so they fixed this at some point.

### Easy fix
In `src/jtag/drivers/bitbang.h` file, Changing
```CPP
const struct swd_driver bitbang_swd;
```
to
```CPP
extern const struct swd_driver bitbang_swd;
```
will fix this.

### The reason why I submit a issue instead of PR
When someone pulls OpenOCD version to latest, then this fix can be treated as a conflict to merge.

### Debug Logs
```
...
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformcd.la  ./jimtcl/libjim.a  -lpthread -lz -lz -ldl
libtool: link: gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra tcl/libjim.a -lpthread -lz -ldl
/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/pi/opesp32/src/jtag/drivers/bitbang.h:60: first defined here
/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-sysfsgpio.o):/home/pi/o-esp32/src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:3142: src/openocd] 오류 1
make[2]: 디렉터리 '/home/pi/openocd-esp32' 나감
make[1]: *** [Makefile:4524: all-recursive] 오류 1
make[1]: 디렉터리 '/home/pi/openocd-esp32' 나감
make: *** [Makefile:2111: all] 오류 2
```
(`오류`, `디렉터리`, `나감` mean `error`, `directory`, `left`. Those are Korean, tho)

### Expected behavior
We should successfully build a OpenOCD for ESP32

gerekon, 2021-11-29:
Hi @MU-Software 

We plan to release the next version with merged upstream changes. SO it will be fixed there. 
Thanks!

MU-Software, 2021-11-29:
Great! If you don't mind, Can you close this issue when upstream merged or next version releases? Maybe I won't be noticed when those happens.

gerekon, 2021-11-29:
> If you don't mind, Can you close this issue when upstream merged or next version releases? Maybe I won't be noticed when those happens.

Yes, sure

erhankur, 2023-01-17:
Closing. We are in sync with the upstream.

closed 2023-01-17 by erhankur