espressif/openocd-esp32 #197: Custom WROVER-E board

ndjur, 2021-11-25:
### Environment
- Development Kit: None - it is a custom WROVER-E based board
- Module or chip used: WROVER-E
- Debug Adapter: ESP-PROG
- OpenOCD version: v0.10.0-esp32-20191114
- Operating System: Windows
- Using an IDE?: VS Code
- OpenOCD command line: openocd -f board/esp32-wrover-kit-3.3v.cfg
- JTAG Clock: [e.g. default]
- ESP-IDF version: v4.1-dirty

OpenOCD error log:

Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: type 'esp32' is missing virt2phys
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
shutdown command invoked
Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of esp32.irom does not support free_driver_priv()
Warn : Flash driver of esp32.drom does not support free_driver_priv()

gerekon, 2021-11-25:
@ndjur 
Looks like JTAG connection problem. Please see https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ#openocd-reports-jtag-scan-chain-interrogation-failed
```
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
```

Also I recommend you to update OpenOCD version.

ndjur, 2021-11-25:
I'll try and report, thanks a lot.

ndjur, 2021-11-26:
I really feel bad for asking this probably trivial question, but I installed the new OpenOCD from the https://github.com/espressif/openocd-esp32/releases and changed entries in the VS Code settings.json from v0.10.0-esp32-20191114 to v0.10.0-esp32-20211111 in these two lines:

"idf.customExtraPaths":  (long string here) and
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\ESP-IDF\TOOLS\.espressif\tools\openocd-esp32\v0.10.0-esp32-20211111\openocd-esp32\share\openocd\scripts\"}",

but I'm now getting two issues:
a) the whole idf.customExtraVars line has a red squiggly underline and a warning that there is an invalid escape character in the string which I don't see and,
b) VS Code still invokes the same old version as before:

Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

However, I changed Windows PATH environment variable and if I just run "openocd -v" in Windows cmd I see the new version:

Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

Can you please tell me where is this openocd setting hiding from me in VS Code? Settings.json in .vscode doesn't even have any openocd related entry.

Thank you very much.

ndjur, 2021-11-26:
Just to report that I got rid of the warning by changing some slash characters  into backslash in the string:
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:/ESP-IDF/TOOLS/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts\"}",
However, the old openocd is still called.

ndjur, 2021-11-26:
Finally - all that was necessary was to restart the VS Code. I'm sorry for wasting your time.
However, the problem is still here as I get this output from the new openocd:

openocd -f board/esp32-wrover-kit-3.3v.cfg

Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x240068cb (mfg: 0x465 (Dosilicon Co Ltd), part: 0x4006, ver: 0x2)
Warn : JTAG tap: esp32.cpu0       UNEXPECTED: 0x240068cb (mfg: 0x465 (Dosilicon Co Ltd), part: 0x4006, ver: 0x2)
Error: JTAG tap: esp32.cpu0  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x240068cb (mfg: 0x465 (Dosilicon Co Ltd), part: 0x4006, ver: 0x2)
Warn : JTAG tap: esp32.cpu1       UNEXPECTED: 0x240068cb (mfg: 0x465 (Dosilicon Co Ltd), part: 0x4006, ver: 0x2)
Error: JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x03 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : esp32.cpu0: Core was reset.
Info : Listening on port 3333 for gdb connections

gerekon, 2021-11-26:
Hi @ndjur 

It looks like problems with JTAG connection. Have you check everything suggested from the link I sent you?

gerekon, 2021-11-26:
Since you are using custom board design, could it be that there are some problems with JTAG pins wiring?

ndjur, 2021-11-26:
Thanks a lot gerekon,

After your first answer I contacted our board designer and it turned out that I connected the 10 pin ribbon cable in the wrong direction to our board 10 pin header connector. This was the reason for my first post.
Later, after connecting correctly and running the latest OpenOCD, I got all correct diagnostics signs from the board LEDs but I still can't connect - as shown in my last post. I don't know if I managed to break something when I connected the cable in the wrong direction.
Another (remote) developer who works with the same board managed to debug his using Eclipse, so I hope that originally everything was OK with the board design.
I'll contact the board designer to get his opinion.
Thank you very much.

ndjur, 2021-11-29:
Here are my most recent attempts to configure the system for debugging:
1. I used ESP32- DevKitC -VE WROVER-E based 8MB board in order to eliminate any issues that I may have with the custom board.
2 Connected ESP-PROG to the development board according to the diagram here: https://www.visualmicro.com/pics/Debug-Help-ESP32-PROG-ALL-Connections.png

VS Code terminal was displaying all serial output so that connection with ESP-PROG was OK.
After this I tried the JTAG connection. 
The result was somewhat different from the one I got with the custom board:
OpenOCD now displays the following:

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED                     d "ftdi_tdo_samp
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"                                                , part: 0x2003, 
Info : clock speed 20000 kHz                                                    , part: 0x2003, 
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : Listening on port 3333 for gdb connections

When I start the debugging in VS Code, I see the following command:

& 'c:\Users\ndjur\.vscode\extensions\ms-vscode.cpptools-1.7.1\debugAdapters\bin\WindowsDebugLauncher.exe' '--stdin=Microsoft-MIEngine-In-0i4c2bc4.e1d' '--stdout=Microsoft-MIEngine-Out-kbdntwgy.q3a' '--stderr=Microsoft-MIEngine-Error-gl10u0wp.b00' '--pid=Microsoft-MIEngine-Pid-bctferze.pm2' '--dbgExe=C:/ESP-IDF/TOOLS/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb.exe' '--interpreter=mi'

and I a debug console pops-up with the gdb output that ends with:

ERROR: Unable to start debugging. Failed to find thread 1 for break event

I don't know how to direct gdb log into a file, otherwise I would attach it here. I'll keep trying.

At the same time, OpenOCD terminal shows this:

Error: ftdi device did not accept all data: 0, tried 84
Error: unable to purge ftdi rx buffers: LIBUSB_ERROR_IO
Error: error while flushing MPSSE queue: -4
Error: libusb_handle_events() failed with LIBUSB_ERROR_IO
Error: unable to purge ftdi rx buffers: LIBUSB_ERROR_IO
Error: error while flushing MPSSE queue: -4
Error: Failed to exec JTAG queue!
Error: Failed to read debug stubs address location (-4)!
Polling target esp32.cpu0 failed, trying to reexamine
Error: libusb_handle_events() failed with LIBUSB_ERROR_IO
Error: unable to purge ftdi rx buffers: LIBUSB_ERROR_IO
Error: error while flushing MPSSE queue: -4
Examination failed, GDB will be halted. Polling again in 100ms

At some point it also shows this line:

Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.

I would really appreciate if you have any further suggestions what to try.
Thank you

erhankur, 2023-12-13:
@ndjur If you still have an issue, feel free to re-open.

closed 2023-12-13 by erhankur