espressif/openocd-esp32 #198: add support for the cmsis-dap debugger in prebuilt openocd binary

wuxx, 2021-11-28:
hello, can we add support for the cmsis-dap debugger? it's a open source debugger and commonly used in arm world, but it also can use to debug esp32 series chip,  and compiling openocd under windows seems very troublesome, so support cmsis-dap in prebuilt openocd binary will be great, thanks~!

gerekon, 2021-12-01:
Hi @wuxx 

It looks like upstream got rid of using hidapi library and enabled `cmsis-dap` and `cmsis-dap_v2` by default in 55821bba1c0a0ba75e9539a9cf0ba3e8d80664aa. We merged upstream changes and plan to release new OpenOCD with them soon.

WangShuoran, 2021-12-28:
Dear @gerekon 

Does this issus already publish with v0.11.0-esp32-20211220? And Does this publish will add in ESP-IDF installer or already add in.
I have a question, but I am doubtful about the value of create issue that my CMSIS-DAP is version 2. And I Google this debugger is or not adapt debug esp32. I find https://www.esp32.com/viewtopic.php?t=22336. It tell me ESP-IDF installed OpenOCD don't support CMSIS-DAP version 2. Is this correct?
And does this release will adapt CMSIS-DAP version 2?

Think you!

igrr, 2022-01-18:
This could also be useful to support hs-probe (https://shop.probe.rs/products/hs-probe-v1-3) which seems to be popular with the folks developing in Rust.

gerekon, 2022-01-27:
> Dear @gerekon
>
> Does this issus already publish with v0.11.0-esp32-20211220? And Does this publish will add in ESP-IDF installer or already add in. I have a question, but I am doubtful about the [...]

Yes, it should be in that release, but I am not sure it is included into IDF Windows installer. In any case if you check out IDF [v4.4](https://github.com/espressif/esp-idf/blob/v4.4/tools/tools.json#L584) or [master](https://github.com/espressif/esp-idf/blob/master/tools/tools.json#L584) it can be installed via script `instal.bat`

wuxx, 2022-06-13:
hello, I try the released openocd-esp32-win32-0.11.0-esp32-20220411, it still fail (can not recognize the CMSIS-DAP debugger), another version I built was succ.
fail:
[image]

succ:
[image]

erhankur, 2022-08-25:
Closed with https://github.com/espressif/openocd-esp32/issues/237 

Feel free to create a new one if you see an issue later on.

closed 2022-08-25 by erhankur