espressif/openocd-esp32 #199: Build fails on Pi

ithinkido, 2021-12-01:
### Environment

- OpenOCD version: pulled direct from git today. 
- Operating System: Pi OS Lite (Debian Bullseye) fresh install - updated and upgraded
- Hardware : Pi 4 

### Problem Description
Building OCD-ESP32 fails on Pi
I am following the build instructions [here](https://blog.wokwi.com/gdb-debugging-esp32-using-raspberry-pi/l).
These are as follows :
```
sudo apt-get update
sudo apt-get install -y git make pkg-config autoconf libtool libusb-1.0-0 libusb-1.0-0-dev

git clone https://github.com/espressif/openocd-esp32 ~/openocd-esp32
cd ~/openocd-esp32
./bootstrap
./configure --enable-sysfsgpio --enable-bcm2835gpio
make
```

The build is not successful and throws the following errors  

`In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:22,
                 from src/flash/nor/jtagspi.c:22:
src/flash/nor/jtagspi.c: In function ‘jtagspi_write_enable’:
./src/helper/log.h:130:2: error: ‘status’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  130 |  log_printf_lf(LOG_LVL_ERROR, __FILE__, __LINE__, __func__, expr)
      |  ^~~~~~~~~~~~~
src/flash/nor/jtagspi.c:262:11: note: ‘status’ was declared here
  262 |  uint32_t status;
      |           ^~~~~~
src/flash/nor/jtagspi.c: In function ‘jtagspi_wait’:
src/flash/nor/jtagspi.c:249:15: error: ‘status’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  249 |   if ((status & SPIFLASH_BSY_BIT) == 0) {
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3550: src/flash/nor/jtagspi.lo] Error 1
make[2]: Leaving directory '/home/pi/openocd-esp32'
make[1]: *** [Makefile:4524: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/openocd-esp32'
make: *** [Makefile:2111: all] Error 2
`

### Debug Logs
Make log attached.
[make_log.txt](https://github.com/espressif/openocd-esp32/files/7633725/make_log.txt)
 
This issue seems to be OCD-ESP32 specifc, I did try to make a generic OCD and this looks to have worked

igrr, 2021-12-01:
Seems like the issue is in src/flash/nor/jtagspi.c, which is not esp-specific. I think this will be fixed shortly, we are about to merge latest upstream changes into our fork. See https://github.com/espressif/openocd-esp32/issues/196#issuecomment-981957942. We'll close this issue once the merge happens.

ithinkido, 2021-12-01:
Perfect, Thanks.

erhankur, 2022-05-06:
With the latest release we are almost up to date with the upstream. Closing this issue.

closed 2022-05-06 by erhankur