espressif/openocd-esp32 #201: esp32-s3 GDB is not connecting and openocd stopping

ndinsmore, 2021-12-11:
### Environment
- Development Kit: ESP32-S3-DecvKitC-1
- Module or chip used: ESP32-S3 WROOM-1
- Debug Adapter: two USB cables
- OpenOCD version: v0.10.0-esp32-20211111
- Operating System:osx
- Using an IDE?: vscode
- OpenOCD command line: openocd launched from vscode [setup such that . openocd -f board/esp32s3-builtin.cfg]
- JTAG Clock: default
- ESP-IDF version: v4.4

### Problem Description
Using the basic blinky example.  Please note that while all this output is from using vscode, I have attempted it through the command line and get the same results.

1.) flash code
2.)Start openocd with the following setup commands

```
        { "text": "set remotelogfile ${workspaceFolder}/build/gdb_log.txt"},
        { "text": "target remote :3333" },
        { "text": "set remote hardware-watchpoint-limit 2"},
        { "text": "mon reset halt" },
        { "text": "thb app_main" },
        { "text": "flushregs" },
        { "text": "c" }
```

3.) Start GDB

when gdb is started the board does halt as indicated by the led no longer flashing,

### Debug Logs

<details>
<summary>openocd output in vscode</summary>
<p>

```
Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 1 options.c:57 configuration_output_handler(): debug_level: 3
User : 14 1 options.c:57 configuration_output_handler(): 
Debug: 15 1 options.c:181 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 1 options.c:182 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 1 options.c:183 add_default_dirs(): exepath=~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin
Debug: 18 1 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding ~/.openocd
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts
Debug: 21 1 configuration.c:42 add_script_search_dir(): adding ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin/../share/openocd/site
Debug: 22 1 configuration.c:42 add_script_search_dir(): adding ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin/../share/openocd/scripts
Debug: 23 1 configuration.c:97 find_file(): found ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts/board/esp32s3-builtin.cfg
Debug: 24 1 configuration.c:97 find_file(): found ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/share/openocd/scripts/interface/esp_usb_jtag.cfg
Debug: 25 1 command.c:143 script_debug(): command - interface interface esp_usb_jtag
Debug: 27 1 command.c:355 register_command_handler(): registering 'espusbjtag'...
Debug: 28 1 command.c:355 register_command_handler(): registering 'espusbjtag'...
Debug: 29 1 command.c:355 register_command_handler(): registering 'espusbjtag'...
Debug: 30 1 command.c:355 register_command_handler(): registering 'espusbjtag'...
Info : 31 1 transport.c:117 allow_transports(): only one transport option; autoselect 'jtag'
Debug: 32 1 command.c:355 register_command_handler(): registering 'jtag_flush_queue_sleep'...
Debug: 33 1 command.c:355 register_command_handler(): registering 'jtag_rclk'...
Debug: 34 1 command.c:355 register_command_handler(): registering 'jtag_ntrst_delay'...
Debug: 35 1 command.c:355 register_command_handler(): registering 'jtag_ntrst_assert_width'...
Debug: 36 1 command.c:355 register_command_handler(): registering 'scan_chain'...
Debug: 37 1 command.c:355 register_command_handler(): registering 'jtag_reset'...
Debug: 38 1 command.c:355 register_command_handler(): registering 'runtest'...
Debug: 39 1 command.c:355 register_command_handler(): registering 'irscan'...
Debug: 40 1 command.c:355 register_command_handler(): registering 'verify_ircapture'...
Debug: 41 1 command.c:355 register_command_handler(): registering 'verify_jtag'...
[... 997 lines cut ...]
w -
r <Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -+$vMustReplyEmpty#3a
r <Timeout: 2 seconds>
w $vMustReplyEmpty#3a
r <Timeout: 2 seconds>
w $vMustReplyEmpty#3a
r <Timeout: 2 seconds>
w $vMustReplyEmpty#3a
r <Timeout: 2 seconds><Timeout: 2 seconds>
w -
r <Error: Connection reset by peer>
End of log
```

</p>
</details>

### Expected behavior
GDB should be properly attached, and openocd should not stop.

### Screenshots
If applicable, add screenshots to help explain your problem.

erhankur, 2021-12-12:
@ndinsmore We added a solution for the ESP32-S3 builtin USB-JTAG into the latest commit.  

You can find the built binary here.

[openocd-esp32-macos-0.11.0-esp32-20211209_test-12-gc8a308b47.tar.gz](https://github.com/espressif/openocd-esp32/files/7698672/openocd-esp32-macos-0.11.0-esp32-20211209_test-12-gc8a308b47.tar.gz)

ndinsmore, 2021-12-12:
@erhankur That seems to work, I had to hack it into thee v0.10 directory as I could not figure out how to repoint idf_tools.   is there a correct way to do that?  

That said gdb does seem a little temperamental and different than my experience on the pi pico.  for example it:
1.) won't let me pause, only breakpoints are working. 
2.) From vscode the restart button doesnt work
3.) Can't add breakpoints while running i get "attempting to bind breakpoint"
4.) I have to hit the start debugging button twice to actually get it to start.

erhankur, 2021-12-13:
Hi @ndinsmore can you please share verbose openocd logs for each item?

ndinsmore, 2022-01-05:
@erhankur I am just getting back up to speed for the new year.   Should I still use the binary included here or are those changes included in the latest release?   If so I will use that.

erhankur, 2022-01-05:
@ndinsmore yes fix is included.You can use the latest release build.

gerekon, 2022-01-28:
@ndinsmore Can we close this issue?

closed 2022-02-04 by gerekon