espressif/openocd-esp32 #202: esp32s2 openocd halting problem

ktaylan, 2021-12-18:
Hi,

I am getting halting error when i start to debug esp32 s2 wroom module with FT2232H.
Openocd console output as follows;

Info : clock speed 2000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s2: Debug controller was reset.
Info : esp32s2: Core was reset.
Error: esp32s2_soc_reset: Couldn't halt target before SoC reset
** Flashing Failed **
-1

Is there an any suggestions about the root cause?

erhankur, 2021-12-19:
Hi @ktaylan can you please add all details mentioned here https://github.com/espressif/openocd-esp32/blob/master/.github/ISSUE_TEMPLATE/issue-report.md

ktaylan, 2021-12-28:
Hi,

I solved the problem. It was bad connection between FT2232H and custom made ESP32-S2 board. I improved JTAG signaling lanes by making direct connection between boards. Now I am starting debugging session without any problem.

closed 2021-12-28 by ktaylan