espressif/openocd-esp32 #204: Cannot read ESP32-C3 memory when the target is running

sysprogs, 2021-12-28:
We are trying to get the background memory reading working on ESP32-C3 (i.e. reading variable values while the CPU is running) and it looks like the chip is not handling the read requests properly despite OpenOCD following the RISC-V debug manual. It would be great if you could clarify whether there is a reasonable workaround to get it working.

The easiest way to reproduce it is to add a global variable to the "blink" project (e.g. `volatile static int s_Counter`), find out its address, and try reading it while the target is running vs. stopped, e.g.:

```
halt
mdw 0x3fc8c474
resume
mdw 0x3fc8c474
```

Reading the memory while the target is stopped works, while resuming it causes all reads to return the value of 0. 

We have tried instrumenting the `read_memory_bus_v1()` function in **riscv-013.c gets** to directly dump the values of **SDBDATA0** and **DBCS** right after starting a read with the following code:

```
uint32_t sbdata, sbcs;
int err0;
err0 = dmi_read(target, &sbdata, DM_SBDATA0);
err0 = dmi_read(target, &sbcs, DM_SBCS);
LOG_DEBUG("SBDATA0=0x%08x, SBCS=0x%08x", sbdata, sbcs);
```

It shows the expected values when the target is stopped:

```
Debug: 103 8555 riscv-013.c:2545 read_memory_bus_v1(): reading burst starting at address 0x3fc8c474
Debug: 104 8559 riscv-013.c:2560 read_memory_bus_v1(): SBDATA0=0x00000386, SBCS=0x20150404
```

When the target is resumed, the SBCS has the same value, but SBDATA0 is always 0:

```
Debug: 139 34179 riscv-013.c:2545 read_memory_bus_v1(): reading burst starting at address 0x3fc8c474
Debug: 140 34184 riscv-013.c:2560 read_memory_bus_v1(): SBDATA0=0x00000000, SBCS=0x20150404
```

If you could suggest any workarounds to get this working, we would very much appreciate it, since inspecting global variables without stopping the application simplifies a lot of debugging scenarios.

igrr, 2022-01-07:
Hi @sysprogs, I'm sorry for the late reply, we didn't have time to look into this issue yet. I believe reading through SBA while the target is running should work, in general, since we use this method for apptrace/SystemView features on ESP32-C3. We'll try to pick up this issue and investigate it soon.

sysprogs, 2022-01-07:
Thanks, any insights into this will be very much appreciated. Just to double-check: on the regular ESP32, the apptrace uses a special memory region that is accessible while running. Can you confirm that on ESP32-C3 it uses the regular RAM? 

Also apptrace in ESP-IDF 4.3.1 appears unimplemented/broken for ESP32-C3. If you could confirm that another version of ESP-IDF supports it reliably, we can try comparing it on our side.

erhankur, 2022-01-07:
Hi @sysprogs Can I ask what is your openocd version? I can read without halting using the latest release v0.11.0-esp32-20211220 

```
$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> mdw 0x3fc8c478
0x3fc8c478: 00000102

> mdw 0x3fc8c478
0x3fc8c478: 00000104

> mdw 0x3fc8c478
0x3fc8c478: 00000105
```

igrr, 2022-01-08:
> Can you confirm that on ESP32-C3 it uses the regular RAM?

Yes, that's correct.

> Also apptrace in ESP-IDF 4.3.1 appears unimplemented/broken for ESP32-C3. If you could confirm that another version of ESP-IDF supports it reliably, we can try comparing it on our side.

apptrace for esp32-c3 is currently available in `master` branch of ESP-IDF, we are currently backporting it to release/v4.4.

sysprogs, 2022-01-11:
@igrr and @erhankur, thanks for your feedback. It turns out, the background memory reads work just fine with the **master** branch of the SDK (commit df1f9ec26e81138494462451889e7cb05dfc4ad4), but don't work with v4.3.2 (commit 8bf14a9238329954c7c5062eeeda569529aedf75).

The variable addresses in the master branch vs. 4.3.2 are 0x3fc8db54 vs, 0x3fc8c6f8, i.e. apparently in the same address space. Using the same gdb and the same OpenOCD version yields a successful readout in the master branch, and produces a value of 0 (unless the target is stopped) on 4.3.2.

It looks like the new SDK might be configuring some part of the device differently. If you have any ideas what that could be, they would be appreciated. If not, we could wait until the current master branch makes it into a stable release, but it would be great if we could find a way to enable this functionality on existing releases.

igrr, 2022-01-11:
@sysprogs now that you mention this difference, I remember what it is related to. We need to prevent the CPU from going into a WFI mode if we want to use SBA. This line is responsible: https://github.com/espressif/esp-idf/blob/df1f9ec26e81138494462451889e7cb05dfc4ad4/components/hal/esp32c3/include/hal/cpu_ll.h#L183

sysprogs, 2022-01-13:
OK, thanks. Applying this patch to ESP-IDF 4.3.2 did the trick.

closed 2022-01-13 by sysprogs