espressif/openocd-esp32 #205: Openocd with cmsis-dap does not work properly for esp32c3 dev board

asypost, 2021-12-29:
### Environment
- Development Kit: ESP32-C3-DevKitM-1
- Debug Adapter: cmsis-dap
- OpenOCD version: github master
- Operating System: Windows 10
- Using an IDE: VS Code
- OpenOCD command line:openocd -f board/esp32c3-cmsis-dap.cfg -c "program_esp_bins build flasher_args.json verify reset exit"
- JTAG Clock: default
- ESP-IDF version: v4.3.2

### Problem Description
Steps to reproduce the behavior:

1. build openocd from source with configuration:

```
./configure --disable-werror 
```

2. create an openocd file esp32c3-cmsis-dap.cfg in openocd board directory with content:
```
# Source the JTAG interface configuration file
source [find interface/cmsis-dap.cfg]
# Source the ESP32-S3 configuration file
source [find target/esp32c3.cfg]
```
3. run openocd in idf terminal
```
openocd -f board/esp32c3-cmsis-dap.cfg -c "program_esp_bins build flasher_args.json verify reset exit"
```
then the result shows as Debug logs and end without any other message.

### Debug Logs
```
Open On-Chip Debugger v0.11.0-esp32-20211220 (2021-12-29-10:42)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Flashing build/partition_table/partition-table.bin at 0x8000
```

### Expected behavior
The openocd program the device with bootloader,application and partition table.

### Screenshots
1.CMSIS-DAP has been installed properly.
[image]
2.OpenOCD has been built and installed properly.
[image]
[image]

erhankur, 2021-12-29:
Hi @asypost ,

Can you please configure with `--enable-cmsis-dap` option and try again. If there is still an issue please share the debug logs  running openocd with -d3

asypost, 2021-12-29:
> Hi @asypost ,
> 
> Can you please configure with `--enable-cmsis-dap` option and try again. If there is still an issue please share the debug logs running openocd with -d3

Here's the log:
```
Debug: 118 475 target.c:1661 handle_target_init_command(): Initializing targets...
Debug: 119 479 esp32c3.c:125 esp32c3_init_target(): enter
Debug: 120 489 semihosting_common.c:100 semihosting_common_init():  
Debug: 121 518 cmsis_dap_usb_bulk.c:111 cmsis_dap_usb_open(): could not open device 0x10de:0x1aec: Operation not supported or unimplemented on this platform
Debug: 122 528 cmsis_dap_usb_bulk.c:111 cmsis_dap_usb_open(): could not open device 0x8086:0xa36d: Operation not supported or unimplemented on this platform
```

asypost, 2021-12-29:
> Hi @asypost ,
> 
> Can you please configure with `--enable-cmsis-dap` option and try again. If there is still an issue please share the debug logs running openocd with -d3

Replace the driver of JTag adapter to WinUSB,I got the log below:
```
Debug: 119 267 target.c:1661 handle_target_init_command(): Initializing targets...
Debug: 120 269 esp32c3.c:125 esp32c3_init_target(): enter
Debug: 121 276 semihosting_common.c:100 semihosting_common_init():  
Debug: 122 304 cmsis_dap_usb_bulk.c:111 cmsis_dap_usb_open(): could not open device 0x10de:0x1aec: Operation not supported or unimplemented on this platform
Debug: 123 310 cmsis_dap_usb_bulk.c:111 cmsis_dap_usb_open(): could not open device 0x8086:0xa36d: Operation not supported or unimplemented on this platform
Debug: 124 315 cmsis_dap_usb_bulk.c:156 cmsis_dap_usb_open(): found product string of 0xc251:0xf001 'CMSIS-DAP-MuseLab'    
Debug: 125 318 cmsis_dap_usb_bulk.c:176 cmsis_dap_usb_open(): enumerating interfaces of 0xc251:0xf001
Debug: 126 321 cmsis_dap_usb_bulk.c:223 cmsis_dap_usb_open(): found interface 0 string 'CMSIS-DAP CDC'
Debug: 127 324 cmsis_dap_usb_bulk.c:236 cmsis_dap_usb_open(): skipping interface 0, has only 1 endpoints
Debug: 128 327 cmsis_dap_usb_bulk.c:223 cmsis_dap_usb_open(): found interface 1 string 'CMSIS-DAP DCI'
Warn : 129 329 cmsis_dap_usb_bulk.c:273 cmsis_dap_usb_open(): Using CMSIS-DAPv2 interface 1 with wrong class 10 subclass 0 or protocol 0
Info : 130 332 cmsis_dap_usb_bulk.c:324 cmsis_dap_usb_open(): Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf001, serial=0001A0000001
Warn : 131 335 cmsis_dap_usb_bulk.c:348 cmsis_dap_usb_open(): could not claim interface: Operation not supported or unimplemented on this platform
Error: 132 339 cmsis_dap_usb_bulk.c:419 cmsis_dap_usb_write(): error writing data: Entity not found
Error: 133 342 cmsis_dap.c:412 cmsis_dap_cmd_DAP_Info(): CMSIS-DAP command CMD_INFO failed.
Debug: 134 344 command.c:628 run_command(): Command 'init' failed with error code -107
Debug: 135 346 command.c:146 script_debug(): command - echo ** OpenOCD init failed **
User : 137 348 command.c:769 jim_echo(): ** OpenOCD init failed **
Debug: 138 349 command.c:146 script_debug(): command - shutdown error
User : 140 351 server.c:755 handle_shutdown_command(): shutdown command invoked
Debug: 141 353 command.c:628 run_command(): Command 'shutdown' failed with error code -4
User : 142 357 command.c:694 command_run_line():
Debug: 143 358 riscv.c:483 riscv_deinit_target(): riscv_deinit_target()
Debug: 144 360 riscv-013.c:1543 deinit_target(): riscv_deinit_target()
Debug: 145 364 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
```

erhankur, 2021-12-29:
To validate you connection and usb driver is working, any chance to test with upstream openocd and maybe with another device that you are sure to work.  Then we can focus on to the diff with the upstream. 

We have already merged upstream v11 release into our fork.  I don't have a test environment right now. So not sure any update needed specific to cmsis-dap

asypost, 2021-12-30:
I used it for debugging rpi-pico.So I assume the adapter is fine,the COM port works fine too so there's no problem with the connection.

asypost, 2021-12-30:
@erhankur I final fixed this. Here's the solution:
1.Set the adapter driver as bellow by right click the adapter in Device Manager and choose Update Device Driver,then 
 
[image]

erhankur, 2021-12-30:
@asypost thanks for sharing the solution. Then, I am closing this issue.

closed 2021-12-30 by erhankur