espressif/openocd-esp32 #206: openocd wierd behavior while stepping through the code

ralphde, 2022-01-08:
### Environment
- Development Kit: Kaluga
- Module or chip used: [e.g. ESP32-WROOM-32]
- Debug Adapter: ESP-PROG  as well
- OpenOCD version: [e.g v0.10.0-esp32-202109 , 20211111, 20211220
- Operating System: Win10
- Using an IDE?: Eclipse and VScode
- OpenOCD command line: openocd -f board/esp32-wrover-kit-3.3v.cfg and esp-prog
- JTAG Clock: 20000
- ESP-IDF version:  v4.2

### Problem Description
after nailing down this nightmare https://esp32.com/viewtopic.php?f=40&t=25437) I just have to fix the debug/flashing problem.
I will continue using Eclipse plug in as there are the settings cleaner (and easier) to maintain.
Both using the same ESP-IDF location and have almost the same issue.

I am using the simple http server example untouched. Compiles seamlessly, no issues. Flashing OK.

Boards: Kaluga and via Jtag-PROG.

In eclipse the debug session does not halt at app_main() at start. It halts somewhere before in an area where no source file information is available. However if I restart the session while debugging it lands at app_main().

From here on both VScode and Eclipse behave the same.
The 'jump over' and 'jump in' button do not behave like they should. The jump over button jumps most of the time into the function and the jump in button behaves like the run button or jump out button.

Both the Kaluga and our own board worked well until the extension 1.3 was installed in VScode automaticly. Since then the problem occurred.

Remarkable is as well, that the Kaluga is not even flashable with the kaluga cfg file. Only selecting the ESP-PROG cfg file allows to flash the kaluga.

this looks to me like a timing problem. Question, did the extension install 1.3 update the open OCD scripts or touched somehow libusb ?

I have tested with the 3 latest openocd versions. 
a really quick answer would be lifesaving

erhankur, 2023-01-19:
@ralphde Sorry for the too late response. if you still have an issue, can you please share openocd verbose logs?

erhankur, 2023-02-11:
Closed due to inactivity.

closed 2023-02-11 by erhankur