espressif/openocd-esp32 #209: ESP32C3 - USB openocd  - Hart 0 unexpectedly reset

andrew-elder, 2022-01-28:
### Development Kit

ESP32-C3-DevKitM-1

### Module or chip used

ESP32-C3-MINI-1 module

### Debug Adapter

USB D+/D- connection

### OpenOCD version

v0.11.0-esp32-20211220 (2021-12-20-15:42)

### Operating System

Linux

### Using an IDE ?

No, used command line

### OpenOCD command line

$ openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

v4.4

### Problem Description

I have moved R1 -> R2 and then R4 -> R3 so that the USB cable connects directly to USB_DN and USB_DP.

I am running on a Linux virtual machine (VMware) that is running on a Windows host.

This is what I get
```
$ openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Hart 0 unexpectedly reset
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
....
```

[... 701 lines cut ...]
Debug: 721 7638 riscv-013.c:3334 write_memory_bus_v1(): transferring burst starting at address 0x600080a8
Debug: 722 7646 target.c:2742 target_write_u32(): address: 0x60008090, value: 0x00000000
Debug: 723 7662 riscv-013.c:3334 write_memory_bus_v1(): transferring burst starting at address 0x60008090
Debug: 724 7669 FreeRTOS.c:1157 FreeRTOS_post_reset_cleanup(): FreeRTOS_post_reset_cleanup
Debug: 725 7669 riscv-013.c:3715 riscv013_get_register(): [0] reading register dcsr on hart 0
Debug: 726 7669 riscv-013.c:802 execute_abstract_command(): command=0x2207b0; access register, size=32, postexec=0, transfer=1, write=0, regno=0x7b0
Debug: 727 7689 riscv-013.c:1510 register_read_direct(): {0} dcsr = 0xc3
Debug: 728 7689 esp32c3.c:266 esp32c3_core_ebreaks_enable(): DCSR: c3
Debug: 729 7689 riscv-013.c:3741 riscv013_set_register(): [0] writing 0xb0c3 to register dcsr on hart 0
Debug: 730 7689 riscv-013.c:1320 register_write_direct(): {0} dcsr <- 0xb0c3
Debug: 731 7697 riscv-013.c:802 execute_abstract_command(): command=0x2307b0; access register, size=32, postexec=0, transfer=1, write=1, regno=0x7b0
Debug: 732 7709 esp32c3.c:323 esp32c3_poll(): Resume core
Debug: 733 7733 esp32c3.c:327 esp32c3_poll(): resumed core
Info : 734 7741 riscv-013.c:3950 riscv013_is_halted(): Hart 0 unexpectedly reset
User : 736 7749 server.c:755 handle_shutdown_command(): shutdown command invoked
Debug: 737 7749 riscv.c:483 riscv_deinit_target(): riscv_deinit_target()
Debug: 738 7749 riscv-013.c:1543 deinit_target(): riscv_deinit_target()
Debug: 739 7750 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas

```
```

### Expected behavior

I expect openocd to start and connect to the device.

igrr, 2022-01-28:
Hi @andrew-elder, this part of the log looks okay from OpenOCD perspective:
```
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
```
Usually this means that the CPU was reset — either by the program itself (`esp_restart` call) or by one of the watchdogs.
You can check what is happening with the program by observing serial output. The serial port will be visible when you connect a USB cable to the built-in USB peripheral of the ESP32-C3.

andrew-elder, 2022-01-28:
Hi @igrr, Well you are correct. Thanks for the prompt response. See below
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d1546
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d40
entry 0x403ce000
I (35) boot: ESP-IDF v4.4 2nd stage bootloader
I (35) boot: compile time 08:47:20
I (35) boot: chip revision: 3
I (36) boot.esp32c3: SPI Speed      : 80MHz
I (41) boot.esp32c3: SPI Mode       : DIO
I (46) boot.esp32c3: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 0013d620
I (89) boot: End of partition table
E (94) esp_image: image at 0x10000 has invalid magic byte (nothing flashed here?)
E (102) boot: Factory app partition is not bootable
E (107) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c3-api1-20210207
...
```
It seems like openocd JTAG does not support recovery from this really bad state (which I don't know how I got on to). My only path forward is reflashing using serial? I don't have easy access to the fully functional serial port because I've moved a couple of resistors to bypass the CP210.

andrew-elder, 2022-01-28:
@igrr - if I was using JTAG with a JLink adapter, would I have this issue?

igrr, 2022-01-28:
@andrew-elder When you connect ESP32-C3 over the built-in USB, you get JTAG _and_ serial (it's a composite device). So you should be able to reflash via USB CDC — just point esptool to the port which appears in your system. It is not necessary to return to CP2102 for this.

That said, what do you mean by "JTAG does not support recovery from this really bad state"? It looks like you don't have an application flashed yet. Are you trying to flash the application with OpenOCD and is that failing somehow?

Normally if you'd like to run your program on the ESP32-C3, you need to:

1) build the program first, usually using `idf.py build`
2) when the program is built, flash it to the device. You can do this either with:
   - `idf.py -p PORT flash` — this will use the serial port. It can be either UART (via CP2102) or USB-CDC (provided on the same port as USB-JTAG)
   - `openocd` with `program_esp` command — this will flash over JTAG. Note that this is usually slower than flashing with `idf.py flash`, since the latter supports compression.
3) run the program. You can use `gdb` or `idf.py monitor`, or both.

andrew-elder, 2022-01-31:
@igrr - here is the result of trying to flash using `idf.py`

```
logix@ubuntu:~/sw/krado/ledc$ idf.py -p /dev/ttyACM0 flash
Executing action: flash
Running ninja in directory /home/logix/sw/krado/ledc/build
Executing "ninja flash"...
[1/5] cd /home/logix/sw/krado/ledc/build/esp-idf/esptool_py && /home/logix/.espressif/python_env/idf4.4_py3..../home/logix/sw/krado/ledc/build/partition_table/partition-table.bin /home/logix/sw/krado/ledc/build/ledc.bin
ledc.bin binary size 0x271f0 bytes. Smallest app partition is 0x100000 bytes. 0xd8e10 bytes (85%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/logix/sw/krado/ledc/build/bootloader/esp-idf/esptool_py && /home/logix/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/logix/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/logix/sw/krado/ledc/build/bootloader/bootloader.bin
Bootloader binary size 0x4d80 bytes. 0x3280 bytes (39%) free.
[2/3] cd /home/logix/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/logix/esp/esp-id...ORY="/home/logix/sw/krado/ledc/build" -P /home/logix/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32c3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 ledc.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.2-dev
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:06:f2:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x00037fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 19840 bytes to 12002...
Writing at 0x00000000... (100 %)
Wrote 19840 bytes (12002 compressed) at 0x00000000 in 0.4 seconds (effective 384.4 kbit/s)...

A fatal error occurred: Packet content transfer stopped (received 25 bytes)
CMake Error at run_serial_tool.cmake:56 (message):
  /home/logix/.espressif/python_env/idf4.4_py3.8_env/bin/python
  /home/logix/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip
  esp32c3 failed

FAILED: CMakeFiles/flash 
cd /home/logix/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/logix/esp/esp-idf" -D SERIAL_TOOL="/home/logix/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/logix/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32c3" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/logix/sw/krado/ledc/build" -P /home/logix/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

And this is what `idf.py monitor` shows

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d1546
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d40
entry 0x403ce000
I (24) boot: ESP-IDF v4.4 2nd stage bootloader
I (24) boot: compile time 07:54:07
I (24) boot: chip revision: 3
I (25) boot.esp32c3: SPI Speed      : 80MHz
I (30) boot.esp32c3: SPI Mode       : DIO
I (35) boot.esp32c3: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 0013d620
I (78) boot: End of partition table
E (82) esp_image: image at 0x10000 has invalid magic byte (nothing flashed here?)
E (90) boot: Factory app partition is not bootable
E (96) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d1546
...
```

I'm building examples/ledc. It seems like the flashing operation is not completing correctly?

I have never successfully flashed an application to the DevKit. It was running the default Rainmaker image that it shipped with from the factory. If it is no long running bootable code, it is because all my attempts to flash an application across JTAG have resulted in a non-bootable/corrupt state.

igrr, 2022-01-31:
@andrew-elder Thanks for the details, that's really interesting! Do you have a chance to try some combination of: different USB cable, different USB port (e.g. internal/external), maybe different OS (if you have another computer)? We did get some reports of ESP32-C3 built-in USB peripheral not working well with certain USB hubs or cables, but didn't find enough information to narrow down the issue.

(cc @radim.karnis for esptool flashing issue)

andrew-elder, 2022-01-31:
@iggr - ok, so I tried a different cable - no joy. I tried a completely different machine and had success.

```
logix@andrew-ub1:~/projects/krado/sw/ledc/ledc_basic$ idf.py -p /dev/ttyACM0 flash
Executing action: flash
Running ninja in directory /home/logix/projects/krado/sw/ledc/ledc_basic/build
Executing "ninja flash"...
[1/5] cd /home/logix/projects/krado/sw/ledc/ledc_basic/build/esp-idf/esptool_py ...partition-table.bin /home/logix/projects/krado/sw/ledc/ledc_basic/build/ledc.bin
ledc.bin binary size 0x261d0 bytes. Smallest app partition is 0x100000 bytes. 0xd9e30 bytes (85%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/logix/projects/krado/sw/ledc/ledc_basic/build/bootloader/esp-idf/esptool_py && /home/logix/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/logix/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/logix/projects/krado/sw/ledc/ledc_basic/build/bootloader/bootloader.bin
Bootloader binary size 0x4d80 bytes. 0x3280 bytes (39%) free.
[2/3] cd /home/logix/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_...ic/build" -P /home/logix/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32c3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 ledc.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.2-dev
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:06:f2:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x00036fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 19840 bytes to 12000...
Writing at 0x00000000... (100 %)
Wrote 19840 bytes (12000 compressed) at 0x00000000 in 0.4 seconds (effective 408.0 kbit/s)...
Hash of data verified.
Compressed 156112 bytes to 83113...
Writing at 0x00010000... (16 %)
Writing at 0x00019ce0... (33 %)
Writing at 0x00020689... (50 %)
Writing at 0x00027d49... (66 %)
Writing at 0x0002e55b... (83 %)
Writing at 0x000358ea... (100 %)
Wrote 156112 bytes (83113 compressed) at 0x00010000 in 2.3 seconds (effective 538.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 451.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

My other (failing) setup used a Lenovo X1 Extreme laptop running Windows 10. The esp-idf setup was running on a Linux Ubuntu 20.04 installation on VMware. I tried both a USB laptop port and another port in the Thinkpad dock and neither of them worked.

Note: I haven't yet tried the debugging step, but I'm sure it will work. This is much further than I got before.

andrew-elder, 2022-01-31:
I set up the ESP32C3 tools on the same Lenovo laptop mentioned above using the "host" Windows OS, and I successfully flashed across USB openocd port. I would have to conclude there is an issue running openocd in a Linux VM. I'm using VMware Workstation 15.5.7 for future reference.

closed 2023-12-13 by erhankur