espressif/openocd-esp32 #210: OpenOCD V11 IDF V4.4 Eclipse Debugger steps into when using stepover command.

MPC-BlackBox, 2022-02-02:
### Development Kit

Customer Product

### Module or chip used

ESP32-wroom-32E

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

ubuntu

### Using an IDE ?

Eclipse

### OpenOCD command line

openocd -s ${openocd_path}/share/openocd/scripts -c 'set ESP32_FLASH_VOLTAGE 3.3' -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg

### JTAG Clock Speed

default

### ESP-IDF version

V4.4

### Problem Description

When stepping over code the debugger will step into instead of over the code

The attached screen recording shows the debugger steps into SPI_lus_initialise()

When I resume the program to a breakpoint then step the debugger steps over some routines but then steps into nvs_close(nvs_handle)

I also get the error 
Error: FreeRTOS uxTaskNumber seems to be corrupted!

### Debug Logs

```shell
Info : esp32.cpu0: Target halted, PC=0x400D7E80, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x40168F66, debug_reason=00000000
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Info : esp32.cpu0: Target halted, PC=0x400D7FDD, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x400D45F5, debug_reason=00000000
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Error: FreeRTOS uxTaskNumber seems to be corrupted!
Info : esp32.cpu0: Target halted, PC=0x400D7FFF, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x4016C04C, debug_reason=00000000
```

### Expected behavior

Debugger should reliably step over routines

https://user-images.githubusercontent.com/59446384/152205957-0c451624-37ab-4f56-9a2d-33dcf0b98644.mp4

.

gerekon, 2022-02-02:
Hi @MPC-BlackBox 

Could you collect verbose OpenOCD and GDB logs and attach them here? You need to add `-d3 -l oocd.log` option to OpenOCD command line and add GDB startup command `set remotelogfile gdb_log.txt` as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb)

MPC-BlackBox, 2022-02-03:
Hi @gerekon,

I tried the -l command in OpenOCD but just got a timeout. OpenOCD does not seem to like the -l command.

I dropped the -l command and just copied the verbose output from the console.

This morning I am not getting the :
Error: FreeRTOS uxTaskNumber seems to be corrupted!

I have looked through the log files but I no idea what I am looking at.

Cheers
[oocd.txt](https://github.com/espressif/openocd-esp32/files/7993312/oocd.txt)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/7993314/gdb_log.txt)

.

MPC-BlackBox, 2022-02-03:
I tried using the command **n** in the debugger console which I believe is the GDB command for step over. But the GDB still step into code.

gerekon, 2022-02-08:
Seems to be the same issue as  https://github.com/espressif/esp-idf/issues/8065 related to fixes in GDB from `esp-2021r2-8.4.0` toolcahin

closed 2022-02-08 by gerekon