espressif/openocd-esp32 #211: Platformio Registry update to latest

Jason2866, 2022-02-04:
### Development Kit

every

### Module or chip used

esp32

### Debug Adapter

ESP-PROG

### OpenOCD version

0.10.

### Operating System

Windows, Linux MacOS

### Using an IDE ?

Plattformio

### OpenOCD command line

not relevant

### JTAG Clock Speed

not relevant

### ESP-IDF version

IDF 4.4

### Problem Description

With Platformio only a outdated version can be used https://registry.platformio.org/tools/platformio/tool-openocd-esp32
Please upload latest version to Platformio registry

### Expected behavior

Using latest version with Platformio

igrr, 2022-02-06:
As mentioned in https://github.com/espressif/esp-idf/issues/8352#issuecomment-1030905576 — i _think_ espressif can't update this package in platformio registry directly, as it is under `platformio` namespace, unlike the toolchain packages.

Jason2866, 2022-02-06:
@igrr  You (espressif) can. It will appear in the ownership `espressif`

Jason2866, 2022-04-05:
Is there a chance to upload under the namespace `espressif`?

igrr, 2022-04-05:
@Jason2866 Indeed there doesn't seem to be any issue blocking this, just the team had been working on other things and we haven't had time to look into automating PlatformIO registry uploads.

Jason2866, 2022-04-06:
Thanks for your promising answer. Lets help us the Ukraine PlatformIO team, as best we can.

ivankravets, 2022-04-06:
@Jason2866 please forward this issue to https://github.com/platformio/platform-espressif32/issues
We will add support for the new openOCD.

closed 2022-09-19 by Jason2866