espressif/openocd-esp32 #213: OpenOCD fails to attach to ESP32-C3 under macOS 12.2.1

jeamland, 2022-02-15:
### Development Kit

ESP32-C3-DevKitM-1

### Module or chip used

ESP32-C3-MINI-1

### Debug Adapter

On-board (GPIO18/19 via AdaFruit USB breakout cable)

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

macOS Monterey 12.2.1

### Using an IDE ?

No

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

N/A

### ESP-IDF version

5.0 (from git, revision 84d11d1c3ed08d4013dd9397bbefc1d75da71215)

### Problem Description

I have connected my board to the host system both via the micro-USB connector (for console) and via the JTAG GPIO pins (using an [Adafruit cable](https://www.adafruit.com/product/4448) connected to D+, D- and GND). The board boots up and I can see messages on the console.

I can see both the CP2102N bridge device and the JTAG device connected to the system:

[image]

When I run openocd however I see the following:

```
$ openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_get_string_descriptor_ascii() failed with -9
$
```

-9 appears to map to `LIBUSB_ERROR_PIPE`.

Out of curiosity I replaced the call to `libusb_get_string_descriptor_ascii` with a call to `libusb_get_string_descriptor` passing the language ID for en_US (0x0409) and that successfully retrieved the serial number, albeit with some extra bytes in front. I'm not sure if this is expected or not. However it then failed in the call to `jtag_libusb_control_transfer` that is used to fetch a capabilities descriptor. While `jtag_libusb_control_transfer` returns 0, it seems that this is due to it squashing the error code from a call to `libusb_control_transfer` which returned -9 again.

### Debug Logs

```shell
$ openocd -d -f board/esp32c3-builtin.cfg
[... 109 lines cut ...]
Info : 118 15 server.c:312 add_service(): Listening on port 4444 for telnet connections
Debug: 119 15 command.c:146 script_debug(): command - init
Debug: 121 15 command.c:146 script_debug(): command - target init
Debug: 123 15 command.c:146 script_debug(): command - target names
Debug: 124 15 command.c:146 script_debug(): command - esp32c3 cget -event gdb-flash-erase-start
Debug: 125 15 command.c:146 script_debug(): command - esp32c3 configure -event gdb-flash-erase-start reset init
Debug: 126 15 command.c:146 script_debug(): command - esp32c3 cget -event gdb-flash-write-end
Debug: 127 15 command.c:146 script_debug(): command - esp32c3 configure -event gdb-flash-write-end reset halt
Debug: 128 15 command.c:146 script_debug(): command - esp32c3 cget -event gdb-attach
Debug: 129 15 target.c:1661 handle_target_init_command(): Initializing targets...
Debug: 130 15 esp32c3.c:125 esp32c3_init_target(): enter
Debug: 131 38 semihosting_common.c:100 semihosting_common_init():
Error: 132 47 libusb_helper.c:381 jtag_libusb_get_serial(): libusb_get_string_descriptor_ascii() failed with -9
Debug: 133 47 command.c:629 run_command(): Command 'init' failed with error code -4
User : 134 47 command.c:694 command_run_line():
Debug: 135 47 riscv.c:483 riscv_deinit_target(): riscv_deinit_target()
Debug: 136 47 riscv-013.c:1543 deinit_target(): riscv_deinit_target()
Debug: 137 50 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
$
```
```

### Expected behavior

OpenOCD starts up and lets me debug the application on the board.

Spritetm, 2022-02-16:
What program is running on the ESP32C3? Do you also get this running a simple 'Hello, world' program?

gerekon, 2022-02-16:
@jeamland It looks like you use OpenOCD config for builtin USB JTAG (`$ openocd -f board/esp32c3-builtin.cfg
`) while connected your JTAG cable to JTAG I\O pads. By default, ESP32-C3 has built-in JTAG enabled and do not need any external JTAG adapter. To make use of it you need to connect USB cable to IO18 and IO19 pins. See docs [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html)

jeamland, 2022-02-16:
> @jeamland It looks like you use OpenOCD config for builtin USB JTAG (`$ openocd -f board/esp32c3-builtin.cfg `) while connected your JTAG cable to JTAG I\O pads. By default, ESP32-C3 has built-in JTAG adapter enabled and do not need any external one. To make use of it you need to connect USB cable to IO18 and IO19 pins. See docs [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html)

Sorry, I wasn't clear: When I say JTAG GPIO pins, I mean GPIO18 and GPIO19, i.e. the builtin JTAG-to-USB functionality.

jeamland, 2022-02-16:
> @jeamland It looks like you use OpenOCD config for builtin USB JTAG (`$ openocd -f board/esp32c3-builtin.cfg `) while connected your JTAG cable to JTAG I\O pads. By default, ESP32-C3 has built-in [...]

Interesting. I built the `get-started/hello_world` example out of esp-idf and flashed that and it now appears to be working. Would you be able to expand on what kinds of applications are likely to run into issues here?

Spritetm, 2022-02-16:
@jeamland What firmware did you originally run, and more important, did it do something with WiFi/BLE?

jeamland, 2022-02-16:
@Spritetm It was the start of trying to do some stuff via wifi from Rust. Primary console was via the micro-USB port so I'm assuming that's UART0. How do I attach the console to the JTAG interface?

Spritetm, 2022-02-16:
In esp-idf, idf.py menuconfig and then Component config → ESP System Settings → Channel for console output.

GiulioDallaVecchia, 2022-03-04:
Hi everyone,
I have the same issue. I'm using USB-JTAG build in. I flashed correctly official AT binary firmware for esp32-c3. After that I'm not able to connect to the microcontroller anymore.

Do you know if the official AT command firmware uses or reinitialize the PIN 18-19?

Thanks a lot in advance.

QGB, 2022-05-08:
```
C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: Unable to execute JTAG queue
Error: Failed to write ESP32C3_TIMG1WDT_PROTECT (-4)!
Error: Failed to disable WDTs (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: Failed to enable EBREAKS handling (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x10, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to read dmstatus!
Error: Failed to resume core (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
```

erhankur, 2022-05-08:
@QGB @GiulioDallaVecchia what is your primary console? UART0 or USB-serial-JTAG? 

Please set it as USB-serial-JTAG and try again.

`idf.py menuconfig  -> Component config → ESP System Settings → Channel for console output`

If you still have a problem can you send a test binary to reproduce the issue?

erhankur, 2022-05-08:
And do you see usb errors after application reset or communication suddenly lost while the application is running? 

OpenOCD tries to re-connect to the usb device after external reset happens. This might not be working as expected on MacOS Monterey. I will check it.

QGB, 2022-05-08:
@erhankur sorry,I use Win10.

GiulioDallaVecchia, 2022-05-09:
Hi @erhankur,
my original problem was [https://github.com/espressif/esp-at/issues/615](url).

erhankur, 2023-01-17:
I prepared test packages linking with the latest libusb(1.0.26) release and tested it internally at the M1 machine. Looks fine. 

If anyone is willing to test, here are the packages; 

[openocd-esp32-win32-0.11.0-esp32-20221026-87-g6960c72f5.zip](https://github.com/espressif/openocd-esp32/files/10435459/openocd-esp32-win32-0.11.0-esp32-20221026-87-g6960c72f5.zip)

[openocd-esp32-macos-0.11.0-esp32-20221026-87-g6960c72f5.tar.gz](https://github.com/espressif/openocd-esp32/files/10435406/openocd-esp32-macos-0.11.0-esp32-20221026-87-g6960c72f5.tar.gz)

[openocd-esp32-macos-arm64-0.11.0-esp32-20221026-87-g6960c72f5.tar.gz](https://github.com/espressif/openocd-esp32/files/10435435/openocd-esp32-macos-arm64-0.11.0-esp32-20221026-87-g6960c72f5.tar.gz)

erhankur, 2023-02-11:
If test packages do not work for you, feel free to re-open.

closed 2023-02-11 by erhankur