espressif/openocd-esp32 #214: Issue connecting ESP32 with 16Mb FLASH since ESP-IDF 4.4

tonyjackson-srt, 2022-03-11:
### Development Kit

Custom board

### Module or chip used

ESP32-WROVER-IE

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows 10

### Using an IDE ?

No

### OpenOCD command line

openocd -s %userprofile%\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "init; reset halt"

### JTAG Clock Speed

default

### ESP-IDF version

v4.4

### Problem Description

Launch OpenOCD with:

**openocd -s %userprofile%\\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "init; reset halt"**

Launch gdb:

**xtensa-esp32-elf-gdb**

On GDB enter: **target remote :3333**

Result from GDB:

```
Remote debugging using :3333
Remote communication error.  Target disconnected.: (undocumented errno 10061).
```

### Debug Logs

```shell
Result from OpenOCD:

Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Too many flash mappings 1852140917! Must be 2.
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Failed to get flash size!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

Log with -d3 and -l to file is too long to attach to this issue, will attempt to add as later comment.
```

### Expected behavior

If I use OpenOCD from the previous ESP-IDF 4.3 release, v0.10.0-esp32-20210401 then it connects and detects flash size correctly, with OpenOCD reporting:

```
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 255 KB
Info : Flash mapping 1: 0x50020 -> 0x400d0020, 967 KB
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32.cpu0.flash' size 16384 KB
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 255 KB
Info : Flash mapping 1: 0x50020 -> 0x400d0020, 967 KB
Info : Using flash bank 'esp32.cpu0.irom' size 968 KB
Info : esp32.cpu0: Target halted, PC=0x40091C2A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 255 KB
Info : Flash mapping 1: 0x50020 -> 0x400d0020, 967 KB
Info : Using flash bank 'esp32.cpu0.drom' size 256 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
```

tonyjackson-srt, 2022-03-11:
Full OpenOCD log with -d3 and -l:

```
Debug: 17 26 log.c:251 handle_log_output_command(): set log_output to "openocd.log"
Debug: 18 26 configuration.c:42 add_script_search_dir(): adding C:\Users\tonyj\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts
Debug: 19 26 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 20 26 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 21 27 options.c:246 add_default_dirs(): exepath=C:/Users/tonyj/.espressif_4.4/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 22 27 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 23 27 configuration.c:42 add_script_search_dir(): adding C:\Users\tonyj\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts
Debug: 24 27 configuration.c:42 add_script_search_dir(): adding C:/Users/tonyj/AppData/Roaming/OpenOCD
Debug: 25 27 configuration.c:42 add_script_search_dir(): adding C:/Users/tonyj/.espressif_4.4/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 26 27 configuration.c:42 add_script_search_dir(): adding C:/Users/tonyj/.espressif_4.4/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 27 27 configuration.c:97 find_file(): found C:\Users\tonyj\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/interface/ftdi/esp32_devkitj_v1.cfg
Debug: 28 29 command.c:146 script_debug(): command - adapter driver ftdi
Debug: 30 29 command.c:146 script_debug(): command - ftdi_vid_pid 0x0403 0x6010 0x0403 0x6014
Debug: 32 29 command.c:146 script_debug(): command - ftdi_channel 0
Debug: 34 29 command.c:146 script_debug(): command - ftdi_layout_init 0x0008 0xf00b
Debug: 36 29 command.c:146 script_debug(): command - ftdi_layout_signal LED -data 0x1000
Debug: 38 29 command.c:146 script_debug(): command - ftdi_layout_signal LED2 -data 0x2000
Debug: 40 29 command.c:146 script_debug(): command - ftdi_layout_signal LED3 -data 0x4000
Debug: 42 29 command.c:146 script_debug(): command - ftdi_layout_signal LED4 -data 0x8000
Debug: 44 29 command.c:146 script_debug(): command - reset_config none
Debug: 46 29 command.c:146 script_debug(): command - adapter speed 20000
Debug: 48 29 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 49 29 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 50 29 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 51 29 options.c:63 configuration_output_handler(): adapter speed: 20000 kHz
User : 52 29 options.c:63 configuration_output_handler(): 
Debug: 53 29 configuration.c:97 find_file(): found C:\Users\tonyj\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/target/esp32.cfg
Debug: 54 29 command.c:146 script_debug(): command - transport select jtag
Debug: 55 30 configuration.c:97 find_file(): found C:\Users\tonyj\.espressif_4.4\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/target/esp_common.cfg
Debug: 56 30 command.c:146 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock are optional
Debug: 58 30 command.c:1115 help_add_command(): added 'program_esp' help text
Debug: 59 30 command.c:146 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [restore_clock]
Debug: 61 30 command.c:1141 help_add_command(): added 'program_esp' usage text
Debug: 62 30 command.c:146 script_debug(): command - add_help_text program_esp_bins write all the images at address specified in flasher_args.json generated while building idf project
Debug: 64 30 command.c:1115 help_add_command(): added 'program_esp_bins' help text
Debug: 65 30 command.c:146 script_debug(): command - add_usage_text program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress] [restore_clock]
Debug: 67 30 command.c:1141 help_add_command(): added 'program_esp_bins' usage text
Debug: 68 30 command.c:146 script_debug(): command - add_help_text esp_get_mac Print MAC address of the chip. Use a `format` argument to return formatted MAC value
Debug: 70 30 command.c:1115 help_add_command(): added 'esp_get_mac' help text
Debug: 71 30 command.c:146 script_debug(): command - add_usage_text esp_get_mac [format]
Debug: 73 30 command.c:1141 help_add_command(): added 'esp_get_mac' usage text
Debug: 74 30 command.c:146 script_debug(): command - jtag newtap esp32 cpu0 -irlen 5 -expected-id 0x120034e5
Debug: 75 30 tcl.c:572 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu0, Dotted: esp32.cpu0, 4 params
Debug: 76 30 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 77 30 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
Debug: 78 30 core.c:1488 jtag_tap_init(): Created Tap: esp32.cpu0 @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 79 30 command.c:146 script_debug(): command - jtag newtap esp32 cpu1 -irlen 5 -expected-id 0x120034e5
Debug: 80 30 tcl.c:572 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu1, Dotted: esp32.cpu1, 4 params
Debug: 81 30 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
[... cut ...]
Debug: 1526 44219 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 1527 44219 xtensa.c:2533 xtensa_target_deinit(): start
Debug: 1528 44220 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
Debug: 1529 44220 target.c:1964 print_wa_layout():    0x40090000-0x40093fff (16384 bytes)
Debug: 1530 44220 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 1531 44220 xtensa.c:2533 xtensa_target_deinit(): start
Debug: 1532 44221 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
```

tonyjackson-srt, 2022-03-11:
By comparison, from v0.10.0-esp32-20210401 (which does work):

```
Debug: 1358 20463 esp_xtensa.c:317 esp_xtensa_get_size(): esp_xtensa_get_size size 0x1000000
Info : 1359 20463 esp_xtensa.c:883 esp_xtensa_probe(): Auto-detected flash bank 'esp32.cpu0.flash' size 16384 KB
Info : 1360 20463 esp_xtensa.c:885 esp_xtensa_probe(): Using flash bank 'esp32.cpu0.flash' size 16384 KB
Debug: 1361 20463 esp_xtensa.c:902 esp_xtensa_probe(): allocated 4096 sectors
Debug: 1362 20463 esp_xtensa.c:828 esp_xtensa_probe(): Flash size = 0 KB @ 0x00000000 'esp32.cpu0' - 'halted'
Debug: 1363 20463 esp_xtensa.c:212 esp_xtensa_flasher_image_init(): base=00000000 set=0
```

erhankur, 2022-04-28:
@tonyjackson-srt I wonder is this the same [issue](https://github.com/espressif/openocd-esp32/issues/223#issuecomment-1094085181)

Any chance to try with the attached version?

erhankur, 2022-07-06:
Closed due to inactivity.

closed 2022-07-06 by erhankur