espressif/openocd-esp32 #215: ❌ Error: Failed to get flash size!

florenso, 2022-03-15:
### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows

### Using an IDE ?

vscode

### OpenOCD command line

flashing with jtag

### ESP-IDF version

V4.4

### Problem Description

I have my jtag debugger working with the blink example. but when my device is flashed with my own project, i get strange errors.
in my_project i use BLE with nimble.

1. flash blink project with uart (works)
2. flash blink project with jtag (works)
3. flash my_project with jtag (works)
4. flash my_project with jtag (error)
5. flash blink project with jtag (error)

(Before I switch between projects I stop OpenOCD.)

### Debug Logs

```shell
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 20000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing d:/ontwikkeling/esp32/ble_gateway/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x4011D8F2, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x4011D8F2, debug_reason=00000000
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Warn : Empty flash mapping!
Warn : Failed to get flash mappings (0)!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
❌ Error: Failed to get flash size!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
❌ Error: Failed to get flash size!
Error: Failed t❌ o probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32.cpu0.flash'!
** Flashing Failed **
-1
```

### Expected behavior

i expect flashing with jtag should always work, no matter what program is loaded into the eps.

florenso, 2022-03-15:
verbose log:

User : 14 3 options.c:63 configuration_output_handler(): debug_level: 3
User : 15 3 options.c:63 configuration_output_handler(): 
Debug: 16 3 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 17 3 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 18 3 options.c:246 add_default_dirs(): exepath=D:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 19 3 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 20 3 configuration.c:42 add_script_search_dir(): adding D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
Debug: 21 6 configuration.c:42 add_script_search_dir(): adding C:/Users/koen/AppData/Roaming/OpenOCD
Debug: 22 6 configuration.c:42 add_script_search_dir(): adding D:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 23 6 configuration.c:42 add_script_search_dir(): adding D:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 24 6 configuration.c:97 find_file(): found D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/board/esp32-wrover-kit-3.3v.cfg
Debug: 25 8 configuration.c:97 find_file(): found D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg
Debug: 26 8 command.c:146 script_debug(): command - adapter driver ftdi
Debug: 28 8 command.c:146 script_debug(): command - ftdi_vid_pid 0x0403 0x6010 0x0403 0x6014
Debug: 30 8 command.c:146 script_debug(): command - ftdi_channel 0
Debug: 32 8 command.c:146 script_debug(): command - ftdi_layout_init 0x0008 0xf00b
Debug: 34 8 command.c:146 script_debug(): command - ftdi_layout_signal LED -data 0x1000
Debug: 36 8 command.c:146 script_debug(): command - ftdi_layout_signal LED2 -data 0x2000
Debug: 38 8 command.c:146 script_debug(): command - ftdi_layout_signal LED3 -data 0x4000
Debug: 40 8 command.c:146 script_debug(): command - ftdi_layout_signal LED4 -data 0x8000
Debug: 42 8 command.c:146 script_debug(): command - reset_config none
Debug: 44 8 command.c:146 script_debug(): command - adapter speed 20000
Debug: 46 8 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 47 8 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 48 8 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 49 8 configuration.c:97 find_file(): found D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/target/esp32.cfg
Debug: 50 8 command.c:146 script_debug(): command - transport select jtag
Debug: 51 9 configuration.c:97 find_file(): found D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/target/esp_common.cfg
Debug: 52 9 command.c:146 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock are optional
Debug: 54 9 command.c:1115 help_add_command(): added 'program_esp' help text
Debug: 55 9 command.c:146 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [restore_clock]
Debug: 57 9 command.c:1141 help_add_command(): added 'program_esp' usage text
Debug: 58 9 command.c:146 script_debug(): command - add_help_text program_esp_bins write all the images at address specified in flasher_args.json generated while building idf project
Debug: 60 9 command.c:1115 help_add_command(): added 'program_esp_bins' help text
Debug: 61 9 command.c:146 script_debug(): command - add_usage_text program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress] [restore_clock]
Debug: 63 9 command.c:1141 help_add_command(): added 'program_esp_bins' usage text
Debug: 64 9 command.c:146 script_debug(): command - add_help_text esp_get_mac Print MAC address of the chip. Use a `format` argument to return formatted MAC value
Debug: 66 9 command.c:1115 help_add_command(): added 'esp_get_mac' help text
Debug: 67 9 command.c:146 script_debug(): command - add_usage_text esp_get_mac [format]
Debug: 69 9 command.c:1141 help_add_command(): added 'esp_get_mac' usage text
Debug: 70 9 command.c:146 script_debug(): command - jtag newtap esp32 cpu0 -irlen 5 -expected-id 0x120034e5
Debug: 71 9 tcl.c:572 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu0, Dotted: esp32.cpu0, 4 params
Debug: 72 9 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 73 9 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
Debug: 74 9 core.c:1488 jtag_tap_init(): Created Tap: esp32.cpu0 @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 75 9 command.c:146 script_debug(): command - jtag newtap esp32 cpu1 -irlen 5 -expected-id 0x120034e5
Debug: 76 9 tcl.c:572 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu1, Dotted: esp32.cpu1, 4 params
Debug: 77 9 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 78 9 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
Debug: 79 9 core.c:1488 jtag_tap_init(): Created Tap: esp32.cpu1 @ abs position 1, irlen 5, capture: 0x1 mask: 0x3
Debug: 80 9 command.c:146 script_debug(): command - target create esp32.cpu0 esp32 -endian little -chain-position esp32.cpu0 -coreid 0 -rtos FreeRTOS
[... cut ...]
Debug: 1438 28452 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 1439 28452 xtensa.c:2533 xtensa_target_deinit(): start
Debug: 1440 28452 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
Debug: 1441 28452 target.c:1964 print_wa_layout():    0x40090000-0x40093fff (16384 bytes)
Debug: 1442 28452 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 1443 28452 xtensa.c:2533 xtensa_target_deinit(): start
Debug: 1444 28452 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas

florenso, 2022-03-15:
Hi @gerekon

I am using the "Single facotry app, no OTA" partition table, with an offset of 0x8000. MD5 checksum is enabled.

content of ./build/partition-table-flash-args :
`--flash_mode dio --flash_freq 40m --flash_size 2MB
0x8000 partition_table/partition-table.bin
`

i am not sure how to open the partition table itself. But [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html) say:

>  ESP-IDF Partition Table
> | Name     | Type | SubType | Offset  | Size   | Flags |
> |----------|------|---------|---------|--------|-------|
> | nvs      | data | nvs     | 0x9000  | 0x6000 |       |
> | phy_init | data | phy     | 0xf000  | 0x1000 |       |
> | factory  | app  | factory | 0x10000 | 1M     |       |

davidlehrian, 2022-04-23:
I am getting this exact same error and am using the same idf version and openocd version but I'm using Eclipse. I can debug the Blink example app just fine but when I try to debug the BluFi example app I get this same error. 

esp-idf version 4.4
IDE Ecliipse
\espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin

> Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> adapter speed: 20000 kHz
> 
> Flashing C:/Users/lehrian/Documents/ESP32/BluFi/build/partition_table/partition-table.bin at 0x8000
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
> Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
> Info : clock speed 20000 kHz
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : esp32.cpu0: Debug controller was reset.
> Info : esp32.cpu0: Core was reset.
> Info : esp32.cpu1: Debug controller was reset.
> Info : esp32.cpu1: Core was reset.
> Info : starting gdb server for esp32.cpu0 on 3333
> Info : Listening on port 3333 for gdb connections
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : esp32.cpu0: Target halted, PC=0x40140226, debug_reason=00000000
> Info : Set GDB target to 'esp32.cpu0'
> Info : esp32.cpu1: Target halted, PC=0x40140226, debug_reason=00000000
> Info : esp32.cpu0: Debug controller was reset.
> Info : esp32.cpu0: Core was reset.
> Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
> Info : esp32.cpu0: Core was reset.
> Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
> Info : esp32.cpu1: Debug controller was reset.
> Info : esp32.cpu1: Core was reset.
> Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
> Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
> Warn : Empty flash mapping!
> Warn : Failed to get flash mappings (0)!
> Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
> Error: Failed to get flash size!
> Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
> Error: Failed to get flash size!
> Error: Failed to probe flash, size 0 KB
> Error: auto_probe failed
> Error: Failed to find bank 'esp32.cpu0.flash'!
> ** Flashing Failed **
> -1
> Started by GNU MCU Eclipse
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : accepting 'gdb' connection on tcp/3333
> Warn : No symbols for FreeRTOS!
> Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
> Warn : Empty flash mapping!
> Warn : Failed to get flash mappings (0)!
> Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
> Error: Failed to get flash size!
> Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
> Error: Failed to get flash size!
> Error: Failed to probe flash, size 0 KB
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected
> shutdown command invoked

erhankur, 2022-08-16:
Fixed in the https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20220706

closed 2022-08-16 by erhankur