espressif/openocd-esp32 #216: esp32-c3内置jatg  openocd运行失败

derskop, 2022-03-22:
### Development Kit

ESP32C3-Dekitv2

### Module or chip used

ESP32C3-Dekitv2

### Debug Adapter

内置jatg

### OpenOCD version

 Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)

### Operating System

windows

### Using an IDE ?

vscode

### OpenOCD command line

"board/esp32c3-builtin.cfg

### JTAG Clock Speed

默认

### ESP-IDF version

v4.4

### Problem Description

我按照文档要求，设置esp32C3为built-in-jtag模式，然后D+,D-分别使用19，18连接。而且将固件更新设置为jatg，运行openocd出现如下错误：

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_get_string_descriptor_ascii() failed with -9

### Debug Logs

```shell
openocd : Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
所在位置 行:1 字符: 1
+ openocd -d3 -f board/esp32-wrover-kit-3.3v.cfg 2>&1 | tee openocd.log
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (Open On-Chip De...21-12-20-15:43):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 10 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 10 options.c:63 configuration_output_handler(): 
Debug: 15 11 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 11 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 11 options.c:246 add_default_dirs(): exepath=d:/ESPconf/tool/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 18 11 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 12 configuration.c:42 add_script_search_dir(): adding d:\ESPconf\tool\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
Debug: 20 12 configuration.c:42 add_script_search_dir(): adding C:/Users/26064/AppData/Roaming/OpenOCD
Debug: 21 12 configuration.c:42 add_script_search_dir(): adding d:/ESPconf/tool/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
[... 252 lines cut ...]
Debug: 295 167 ftdi.c:650 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Error: 296 205 mpsse.c:216 open_matching_device(): no device found
Error: 297 206 mpsse.c:366 mpsse_open(): unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: 298 259 mpsse.c:216 open_matching_device(): no device found
Error: 299 259 mpsse.c:366 mpsse_open(): unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'
Debug: 300 261 command.c:629 run_command(): Command 'init' failed with error code -100
User : 301 261 command.c:694 command_run_line(): 
Debug: 302 261 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 303 262 xtensa.c:2533 xtensa_target_deinit(): start
Assertion failed!

Program: d:\ESPconf\tool\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0
Debug: 304 262 server.c:609 sig_handler(): Terminating on Signal 22
```

### Expected behavior

我希望上述配置完成之后，即可实现下载调试。

### Screenshots

[image]

erhankur, 2022-04-07:
@derskop Debug logs you shared are not for ESP32-C3. 

Are you trying to connect from usb-jtag or from the jtag pins?

erhankur, 2022-04-28:
@derskop if your issue continues can you please answer my question. Otherwise we can close the issue.

erhankur, 2022-05-06:
Closing this issue due to inactivity.

closed 2022-05-06 by erhankur