espressif/openocd-esp32 #217: Design Consideration: using USB-JTAG vs. external JTAG on ESP32-S3

franz-ms-muc, 2022-03-23:
### Development Kit

ESP32-S3-DEVKITC-1-N8

### Module or chip used

ESP32-S3-WROOM-1-N8

### Debug Adapter

ESP-PROG

### OpenOCD version

all

### Operating System

all

### Using an IDE ?

Eclipse

### OpenOCD command line

all

### JTAG Clock Speed

all

### ESP-IDF version

V4.4

### Problem Description

Hello, 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html

it show 2 ways of JTAG: 

- Configure ESP32-S3 built-in JTAG Interface
- Configure Other JTAG Interface

Question: 
are same Features available on both ways ? 
like same Number of Breakpoints ? 

same Speed on the JTAG interface ? 
same overall "user Experience" ?

### Expected behavior

Get an answer about the Question.

franz-ms-muc, 2022-04-06:
Hello, 

any news about this ?

Spritetm, 2022-04-06:
The internal debug structure is the same, so the user experience and breakpoints etc are the same. Speed is hard to compare, as it heavily depends on the JTAG adapter you would otherwise use plus the things you use it for. Do note that most of the [limitations](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-serial-jtag-console.html) for the serial part also apply to the JTAG part.

igrr, 2022-04-06:
Debugging speed can be characterized in different ways, but the simplest to measure is probably the speed of flashing the code over JTAG.

Here's a comparison between ESP32 with FT2232H bridge and ESP32-C3 built-in USB_SERIAL_JTAG.

1. Generate a 1 megabyte file: `dd if=/dev/zero of=/tmp/1mb.bin bs=1024 count=1024`
2. Flash ESP32 with FT2232H:
   `openocd -f board/esp32-ethernet-kit-3.3v.cfg -c 'init; reset halt; program_esp /tmp/1mb.bin 0x100000'`
   Result from the log:
   `Info : PROF: Data transferred in 3398.64 ms @ 301.297 KB/s`
3. Flash ESP32-C3 with USB_SERIAL_JTAG:
   `openocd -f board/esp32c3-builtin.cfg -c 'init; reset halt; program_esp /tmp/1mb.bin 0x100000'`
   Result from the log:
   `Info : PROF: Data transferred in 10371.2 ms @ 98.7348 KB/s`

So flashing the program using ESP32-C3 built-in USB_SERIAL_JTAG is currently about 3 times slower than flashing over an external FT2232H.

There may be some room for improvement, as even 300kB/sec is a throughput which theoretically can be sustained over the 12Mbps USB1.1 interface of USB_SERIAL_JTAG. But at least with the current version of OpenOCD you can expect such performance difference.

For other types of practical usage, such as stepping and setting breakpoints, the difference in latency will likely be smaller than 3 times.

igrr, 2022-04-06:
One more note, I have deliberately not used the `compress` option of `program_esp` command to show the difference in JTAG throughput.

In practical usage, you would instead run `program_esp <filename> <address> compress verify`, and compressed flashing will happen faster.

closed 2022-04-06 by igrr

franz-ms-muc, 2022-04-08:
thanks so much !