espressif/openocd-esp32 #218: ❌ Error: Failed to get flash maps / ❌ Error: Failed to get flash size!

brentalanmiller, 2022-03-29:
### Development Kit

ESP32-S3-DevKitC-1

### Module or chip used

ESP32-S3-WROOM-2

### Debug Adapter

ESP32-S3 Built-In

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows 10 and Ubuntu 20.04

### Using an IDE ?

VS Code 1.65.2 / ESP-IDF Plugin 1.4.0

### OpenOCD command line

openocd -d3 -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

4.4

### Problem Description

1. I've been following the VSCode tutorial at https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/basic_use.md
2. When I select JTAG as the flashing method and try to flash, I see the following message in the openocd console:
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000001
Info : esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Flashing /home/skycatch/tmp/blink/blink/build/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x500000EF, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
Info : esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
❌ Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
❌ Error: Failed to get flash size!
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
❌ Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32s3.cpu0.flash'!
** Flashing Failed **
-1
```
3. UART flashing works fine.
4. If I flash via UART, then try to debug via JTAG, I just get a pop-up error message saying "Failed to init Debug Adaptor", and the openocd console shows:
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000001
Info : esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
❌ Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
❌ Error: Failed to get flash size!
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
❌ Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
❌ Error: attempted 'gdb' connection rejected
```
5. I thought this might be related to #215, but I JTAG flashing and debugging fails 100% of the time with the blink project.

### Debug Logs

```shell
When I manually run OpenOCD as `openocd -d3 -f board/esp32s3-builtin.cfg` and try to debug with gdb the logs are too long for github, but can be downloaded here https://drive.google.com/file/d/1ArEICuVxiwI2_Hi_CGY3RdiQSSlcBjDK/view?usp=sharing
```

### Expected behavior

To be able to flash and debug code in VSCode.

erhankur, 2022-04-07:
@brentalanmiller What is the flash size of your board?

brentalanmiller, 2022-04-07:
@erhankur It's the 32MB version.

erhankur, 2022-04-07:
I assume it is `esp32-s3-wroom-2-n32r8v` with the octal flash. If yes, 32MB octal flash support added to master branch and not released yet. We are planning to release soon.  

If you don't want to wait for release I can send the binary for your tests. (Just tell me which OS are you working on) 
Or you can easily build from the source on linux.

brentalanmiller, 2022-04-07:
@erhankur thanks for the info and yes, I have a ESP32-S3-WROOM-2-N32R8V. (Sorry, I should have included that in the initial report.)

I just checked out and built master at 554a6c26cfa61fff241a9a0eefae6bdf64abd72d and now I can flash and debug via JTAG.

Thanks again! Appreciate the help.

closed 2022-04-07 by brentalanmiller

erhankur, 2022-04-09:
For those who have a building issue from the source on windows.

[openocd-esp32-win32-0.11.0-esp32-20211220-1338-g554a6c26c.zip](https://github.com/espressif/openocd-esp32/files/8457507/openocd-esp32-win32-0.11.0-esp32-20211220-1338-g554a6c26c.zip)

erhankur, 2022-05-05:
@dtangBE [openocd-esp32-macos-arm64-0.11.0-esp32-20220411-7-g4589234da.tar.gz](https://github.com/espressif/openocd-esp32/files/8630400/openocd-esp32-macos-arm64-0.11.0-esp32-20220411-7-g4589234da.tar.gz)