espressif/openocd-esp32 #219: Error: esp_usb_jtag: could not find or open device (ESP32-S3-DevKitC-1)

niliha, 2022-03-30:
### Development Kit

ESP32-S3-DevKitC-1

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

builtin

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Linux, Fedora Workstation 35

### Using an IDE ?

no

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v4.4

### Problem Description

1. Connect ESP32-S3-DevKitC-1 using JTAG USB port to your PC.
2. Run `openocd -f board/esp32s3-builtin.cfg`
3. Observe `Error: esp_usb_jtag: could not find or open device!`

### Debug Logs

```shell
See attachment
```

### Expected behavior

openocd connects to the board.

niliha, 2022-03-30:
Running `openocd -f board/esp32s3-builtin.cfg ` outputs:

```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!

openocd: ../src/jtag/core.c:343: jtag_checks: Assertion `jtag_trst == 0' failed.
[1]    12211 IOT instruction (core dumped)  openocd -f board/esp32s3-builtin.cfg
```

Debug output with log level 4:
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/8384325/openocd_log.txt)

Privileges seem to be alright. I set the udev rules and can access the board, e.g. through `cat /dev/ttyACM0`.

`dmesg` outputs
```
[ 2979.017957] usb 1-4: new full-speed USB device number 10 using xhci_hcd
[ 2979.147889] usb 1-4: New USB device found, idVendor=303a, idProduct=4001, bcdDevice= 1.00
[ 2979.147902] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2979.147907] usb 1-4: Product: Espressif Device
[ 2979.147911] usb 1-4: Manufacturer: Espressif Systems
[ 2979.147915] usb 1-4: SerialNumber: 123456
[ 2979.150604] cdc_acm 1-4:1.0: ttyACM0: USB ACM device
```

`lsusb` lists the board as
```
Bus 001 Device 010: ID 303a:4001 Espressif Systems Espressif Device
```

Thank you! I appreciate any help.

Spritetm, 2022-03-31:
Sorry, this still very much feels like a permission problem. Note that being able to access /dev/ttyACM0 does not mean that the USB permissions are set correctly, as generally Linux distros give this device the permissions belonging to a serial port, not the USB device it stems from. If you want to make sure this is not a permission problem, you can try running OpenOCD as root (note that normally you should never do this! but for a check it should be OK); if that does work, your issue is likely permissions.

niliha, 2022-03-31:
Hey, thanks for the quick reply!
Unfortunately, running openOCD as root does not change anything.
I also tried a different USB cable.

Spritetm, 2022-03-31:
Ah, I think I see something... that PID/VID does not match the USB serial-JTAG converter, and that serial number also doesn't make sense. Could it be that you have firmware in the ESP32S3 that uses the USB OTG peripheral? If so, that takes over the USB port, and you can't do JTAG debugging using that.

niliha, 2022-04-01:
Wow, this did indeed fix the issue! I don't know what was on the esp before but now its working.
`dmesg` now looks like this:
```
[  526.643224] usb 1-1: new full-speed USB device number 7 using xhci_hcd
[  526.771338] usb 1-1: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[  526.771352] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  526.771358] usb 1-1: Product: USB JTAG/serial debug unit
[  526.771363] usb 1-1: Manufacturer: Espressif
[  526.771367] usb 1-1: SerialNumber: <some hex serial number>
```

Thanks a lot for your help!

closed 2022-04-01 by niliha

Spritetm, 2022-04-01:
Nice :) Btw, the <some hex serial number> is the base MAC address of the ESP32S3.

fahrettinaslan, 2022-04-08:
Hi,

I have a same problem.
Switching to JTAG after loading an empty project did not fix it.

LCKway, 2022-09-16:
`PS F:\JLPro\Mars_Plan\Procedure\LED_M2\JATG\blink> openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!

Assertion failed!

Program: D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 338

Expression: jtag_trst == 0`

windos 10 Have the same problem

rdlaner, 2023-06-19:
@niliha @Spritetm 
Apologies for commenting on a closed thread. However, I'm having the same issue and I suspect that the app running on my esp32s3 is enabling USB-OTG and therefore preventing JTAG from working.

How did you update your app to prevent it from enabling USB-OTG? Is it an update via menuconfig or something?

Henry586, 2023-11-01:
> `PS F:\JLPro\Mars_Plan\Procedure\LED_M2\JATG\blink> openocd -f board/esp32s3-builtin.cfg Open On-Chip Debugger v0.11.0-esp32-20220411 (2022-04-11-08:48) Licensed under GNU GPL v2 For bug reports, [...]

For me, just put esp32s3 to download mode by press boot button, press reset, release boot button to enter jtag mode.