espressif/openocd-esp32 #220: Not able to debug with ESP32c3 builtin JTAG, linux and VisualStudio .

jiauka, 2022-03-31:
### Development Kit

Custom board

### Module or chip used

ESP32-C3-MINI-1

### Debug Adapter

embedded JTAG

### OpenOCD version

 v0.11.0-esp32-20211220

### Operating System

Ubuntu 20.04

### Using an IDE ?

Visual Studio code

### OpenOCD command line

 openocd  -d4 -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

40000 

### ESP-IDF version

v4.4 and master

### Problem Description

I can not debug with Visual Studio Code.
I can stgart debugging, then debugger stop at app_main as usual, I do setup a break point somewhere inside my code, then press "play", it runs until teh break point and I can not do anything else, no step over, no step in. The CPU is halted but I can not debug it.

### Debug Logs

```shell
Debug: 17 3 log.c:251 handle_log_output_command(): set log_output to "openocd.log"
Debug: 18 3 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 19 3 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 20 3 options.c:246 add_default_dirs(): exepath=/home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 21 3 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 22 3 configuration.c:42 add_script_search_dir(): adding /home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
Debug: 23 3 configuration.c:42 add_script_search_dir(): adding /home/jaume/.config/openocd
Debug: 24 3 configuration.c:42 add_script_search_dir(): adding /home/jaume/.openocd
Debug: 25 3 configuration.c:42 add_script_search_dir(): adding /home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 26 3 configuration.c:42 add_script_search_dir(): adding /home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 27 3 configuration.c:97 find_file(): found /home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg
Debug: 28 3 command.c:146 script_debug(): command - debug_level 2
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [0] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 194 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 605 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 194 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 605 KB
Info : Using flash bank 'esp32c3.irom' size 608 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 194 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 605 KB
Info : Using flash bank 'esp32c3.drom' size 196 KB
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 194 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 605 KB
Info : Using flash bank 'esp32c3.irom' size 608 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 194 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 605 KB
Info : Using flash bank 'esp32c3.drom' size 196 KB
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : esp32c3: Detected debug stubs @ 0x3fc94724
Warn : Not full dbg stub table 1 of 3
```

### Expected behavior

Be able to debug

jiauka, 2022-04-09:
Hi and thanks.

I haven't tried the Visual Code debug for a few day. And now it is even worse.
It says "Failed to Init Debug adapter"
launch.json
`{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}`

ESP doctor
`---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.13.0-39-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 /home/jaume/.espressif/python_env/idf5.0_py3.8_env 
System environment variable PATH 
 /home/jaume/esp/esp-idf5/components/esptool_py/esptool:/home/jaume/esp/esp-idf5/components/espcoredump:/home/jaume/esp/esp-idf5/components/partition_table:/home/jaume/esp/esp-idf5/components/app_update:/home/jaume/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/jaume/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/jaume/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/jaume/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/jaume/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/jaume/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/jaume/.espressif/tools/cmake/3.20.3/bin:/home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/jaume/.espressif/tools/ninja/1.10.2/:/home/jaume/.espressif/python_env/idf5.0_py3.8_env/bin:/home/jaume/esp/esp-idf5/tools:/home/jaume/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.65.2 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/jaume/esp/esp-idf5
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /home/jaume/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/jaume/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/jaume/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/jaume/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/jaume/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/jaume/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/jaume/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to /home/jaume/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/jaume/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/jaume/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/jaume/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to /home/jaume/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/jaume/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.25.1
ESP-IDF version 5.0
Python version 3.8.10
Python's pip version 21.3.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.4
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2021.10.8
cffi version: 1.15.0
chardet version: 4.0.0
charset-normalizer version: 2.0.12
click version: 8.0.4
[... 115 lines cut ...]
Debug: 17 2 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 18 2 options.c:246 add_default_dirs(): exepath=/home/jaume/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 19 2 options.c:247 add_default_dirs(): bin2data=../share/openocd
[... cut ...]
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections

`

erhankur, 2022-04-28:
@jiauka Sorry for the late answer. I still couldn't see the logs with debug_level.  
`openocd -d3 -f board/esp32-wrover-kit-3.3v.cfg 2>&1 | tee openocd.log`

Or if you can send me an example problematic application, I can reproduce the issue and help you faster.

pipiv, 2022-05-07:
Hi,

got a similar issue on linux, I can't start debugger I got this issue:
[image]
[openocd.log](https://github.com/espressif/openocd-esp32/files/8645598/openocd.log)

volkmarnissen, 2022-09-30:
I ahd the same issue. 
Solution/workarount: Press the boot button on your board.

erhankur, 2023-06-12:
If you still have similar issue, please try with the latest OpenOCD release and report again.

closed 2023-06-12 by erhankur