espressif/openocd-esp32 #222: ESP32-C3 LIBUSB error (openocd, probe.rs) (IDFGH-7149)

TheButlah, 2022-04-08:


erhankur, 2022-04-11:
@TheButlah Did you try with the correct cfg file as Roland mentioned. `board/esp32c3-builtin.cfg`

TheButlah, 2022-04-11:
The issue is that libusb_open did not work, this is not specifically an openocd issue, it's a driver level issue I believe.

Would the `board/esp32c3-builtin.cfg` have anything to do with that?

igrr, 2022-04-11:
It may help, since the device OpenOCD will be trying to open (VID/PID) is specified in the config file. The one you have been using is for an FT2233H based board, so VID/PID pair is different.

TheButlah, 2022-04-13:
Trying to test openocd. I click "open esp-idf terminal" in the vscode extension, and the type `openocd`. I get:
```
PS D:\Ryan\Programming\esp32c3_experiments\espidf_example> openocd
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)      
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:26: Error: Can't find openocd.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Debug Adapter has to be specified, see "adapter driver" command
embedded:startup.tcl:26: Error:        
in procedure 'script' 
at file "embedded:startup.tcl", line 26
```

TheButlah, 2022-04-13:
I see two hardware devices:
[image]
[image]

This is what openocd gives when trying to start it through the vscode button on the toolbar. Not that it is different from the first error message I posted. I may have set the wrong board target at that time, but *this* time I can confirm that it is targeting the ESP32-C3, and the flash type is JTAG:

```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed❌  with LIBUSB_ERROR_NOT_SUPPORTED
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
❌ Error: esp_usb_jtag: could not find or open device!

[Stopped] : OpenOCD Server
```

I installed all the components except the IDE itself through [this](https://github.com/espressif/idf-installer/releases/tag/online-2.15-beta-01) installer (aka I am running espressif's drivers).

TheButlah, 2022-04-13:
Probe-rs also fails with `LIBUSB_ERROR_NOT_FOUND`

TheButlah, 2022-04-13:
WHAT??? It magically fixed itself somehow.

Now, openocd attaches, and also the device still has a COM port, and also probe-rs via libusb attaches. How???

Here is what I did:
I went back to messing with zadig drivers. First I tried replacing the usbser driver (the image I posted with the hardware ID ending in MI_00, it was listed as a COM device) with WinUSB via zadig. I tried testing it out with probe-rs - no dice, it still gave a LIBUSB_ERROR_NOT_FOUND. So I reverted the MI_00 device to the usbser driver inside of zadig - this is what it was using originally.

Next I tried installing winusb on the other device (the second image, listed as MI_02). Didn't work. I tried libusbK too. Didn't work, this time crashing the program with an access violation. So I went into device manager and uninstalled it, like this:
[image]
This reverted the driver to the espressif one:
[image]
Its important to note that to the absolute best of my knowledge, this device was *already* using the espressif driver. And yet, for some reason, it started working. I can start openocd, and the example probe-rs code runs. I can also still flash the device, because there is still the COM device listed too.

OpenOCD:
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
```

Probe-rs:
```
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
 DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
 DEBUG jaylink                          > libusb 1.0.24.11584
 DEBUG jaylink                          > libusb has capability API: true       
 DEBUG jaylink                          > libusb has HID access: true
 DEBUG jaylink                          > libusb has hotplug support: false     
 DEBUG jaylink                          > libusb can detach kernel driver: false
DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 12346, idProduct: 4097, bcdDevice: 257, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
[ESP JTAG (VID: 303a, PID: 1001, Serial: 84:F7:03:33:50:60, EspJtag)]
 TRACE probe_rs::probe::cmsisdap::tools > Attempting to open device matching 303a:1001:84:F7:03:33:50:60
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 002: ID 0b05:18f3
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 001: ID 046d:085e
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 000: ID 1022:149c
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 002: ID 1532:0086
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 005: ID 0bda:5418
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 000: ID 1022:149c
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 003: ID 2109:2815
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 006: ID 303a:1001
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 001: ID 2109:0815
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 002 Device 001: ID 8087:0025
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 003 Device 000: ID 1022:149c
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 002: ID 0bda:0411
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 003: ID 0bda:5411
 TRACE probe_rs::probe::cmsisdap::tools > Error opening: NotSupported
 TRACE probe_rs::probe::cmsisdap::tools > Trying device Bus 001 Device 004: ID 1209:2201
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 303a:1001 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::espusbjtag::protocol  > Acquired libusb context.
 DEBUG probe_rs::probe::espusbjtag::protocol  > Aquired handle for probe
[... cut ...]
 TRACE probe_rs::probe::espusbjtag::protocol  > Interface 2
 TRACE probe_rs::probe::espusbjtag::protocol  > Endpoint 2: 2
 TRACE probe_rs::probe::espusbjtag::protocol  > Endpoint 3: 131
 DEBUG probe_rs::probe::espusbjtag::protocol  > Claiming interface 2 with IN EP 131 and OUT EP 2.
 DEBUG probe_rs::probe::espusbjtag::protocol  > [1, 10, 1, 8, 64, 31, 1, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 DEBUG probe_rs::probe::espusbjtag::protocol  > Protocol version: 1
 DEBUG probe_rs::probe::espusbjtag::protocol  > Succesfully attached to ESP USB JTAG.
```

TheButlah, 2022-04-13:
I've gone ahead and uninstalled all the espressif drivers, to test if the issue is the use of the default windows ones. As expected, the default usbser driver + default windows-provided WinUSB fails. 

To clarify, this is what I mean by the default windows WinUSB:
[image]

This causes openocd and probe-rs to fail in the previously discussed ways, since libusb cannot open the device.

**The problem then is that when I try to install the espressif drivers, it doesn't actually replace the windows one.**
I use the windows installer, and have everything unselected except drivers, since the rest of esp-idf is still installed:
[image]
[image]
Installation almost instantly finishes, and yet, the windows driver remains, *not* the espressif one.

How do I get the espressif driver for device with VID: 303A, PID: 1001, MI_02 (does that mean interface 2?). This is the second device, from [this](https://github.com/espressif/openocd-esp32/issues/222#issuecomment-1097590211) message.

TheButlah, 2022-04-13:
OK, I can install the WinUSB driver through zadig, instead of espressif's installer. This is an acceptable workaround. For those coming across this in the future, be sure that your drivers look like this:

Device 1: USB Serial
[image]
This device will remain as the default windows usbser driver, we don't need to change it.

Device 2: Regular USB
[image]
note that if it shows its signed by espressif instead of libwdi, thats fine, it means the espressif installer worked and you don't need zadig.

This second device is the one that needs to be installed with Zadig. I used WinUSB as the driver. Be sure you select "install driver" instead of "install WCID driver" in the dropdown. It should look like this:
[image]

TheButlah, 2022-04-13:
@igrr I think this confirms that the issue is actually with the espressif driver installer, in some cases it doesn't overwrite the windows default driver with the libwdi one.

I would probably transfer back to `esp-idf`