espressif/openocd-esp32 #224: Error while flashing ESP32-C3-DevKitM-1 via builtin JTAG interface

Wasabinary, 2022-04-14:
### Development Kit

ESP32-C3-DevKitM-1

### Module or chip used

ESP32-C3-MINI-1

### Debug Adapter

builtin usb-jtag

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows

### Using an IDE ?

Eclipse

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg 

### JTAG Clock Speed

default

### ESP-IDF version

v4.4

### Problem Description

1. Create ESP-IDF "hello-world" template project for ESP32-C3
2. Remove R1 and R4 resistors from ESP32-C3-DevKitM-1 to disconnect CP2102N USB to Serial chip from ESP32-C3-MINI-1
3. Place R2 and R3 resistors to link USB connector straight to GPIO18 and GPIO19
4. Flash board via JTAG using Eclipse IDE and *openocd -f board/esp32c3-builtin.cfg* command

### Debug Logs

```shell
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Users/rbruneau/eclipse-workspace/hello_world/build/partition_table/partition-table.bin at 0x8000
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32c3.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Warn : target esp32c3 examination failed
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32c3.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Error: Unsupported DTM version: 15
```

### Expected behavior

Successful flashing of the board

erhankur, 2022-05-06:
@Wasabinary there is similar topic in the ESP32 [forum](https://esp32.com/viewtopic.php?t=21007) Can you please verify your modifications?

Wasabinary, 2022-05-09:
Hello @erhankur,

This topic is precisely the one I followed to get the hardware modifications right. It looks OK since the board is recognized as a COM port, so I can flash it using *idf.py -p COMx flash monitor*. Still, the  *openocd -f board/esp32c3-builtin.cfg* command fails with the error message above... Any suggestions ?

erhankur, 2023-01-19:
@Wasabinary I couldn't help a lot, but I wonder did you manage to fix the issue?

Wasabinary, 2023-01-19:
@erhankur No I actually gave up. I think I don't really understand how the built-in JTAG interface works...

erhankur, 2023-01-19:
@Spritetm maybe you can be help in the hardware design.

Marc-Aurele, 2023-03-27:
Hello

Don't you have to solder C4 and C5 also ? Maybe the issue could come from this ? (as they seem to be unmounted)
I'm also interested in this feature but I didn't test it yet !

regards

closed 2023-12-13 by erhankur