espressif/openocd-esp32 #225: ESP32-WROVER-E module GDB auto detached

GavinChung, 2022-04-20:
### Development Kit

ESP32-WROVER-E module

### Module or chip used

ESP32-D0WD-V3

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows

### Using an IDE ?

VS Code

### OpenOCD command line

openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

20000K

### ESP-IDF version

V4.4

### Problem Description

The everything was working fine before I add more functions.
It was able to hit break point at app_main and stop at break point for me to hit run, but right now it just show "[Inferior 1 (Remote target) detached]".
Even I rollback to old version of code still encounter same issue.

### Debug Logs

```shell
undefinedC:\espressif\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Running executable
Remote debugging using :3333
0x40000400 in ?? ()
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Debug controller was reset.
esp32.cpu1: Core was reset.
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Register cache flushed.
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Debug controller was reset.
esp32.cpu1: Core was reset.
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
PROF: Data transferred in 3038.7 ms @ 218.515 KB/s
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
** Programming Finished **
** Verify Started **
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
** Verified OK **
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Debug controller was reset.
esp32.cpu1: Core was reset.
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
PROF: Data transferred in 32.608 ms @ 122.669 KB/s
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
** Programming Finished **
** Verify Started **
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
**** Verification failure! ****
target_hash e3b0c4...52b855
file_hash: 372dfd...fa2862
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'program_esp'
in procedure 'program_error' called at file "C:/espressif/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 135
at file "embedded:startup.tcl", line 530
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Debug controller was reset.
esp32.cpu1: Core was reset.
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
PROF: Data transferred in 47.948 ms @ 250.271 KB/s
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
** Programming Finished **
** Verify Started **
esp32.cpu0: Target halted, PC=0x40092AEE, debug_reason=00000001
** Verified OK **
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Debug controller was reset.
esp32.cpu1: Core was reset.
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Register cache flushed.
Hardware assisted breakpoint 1 at 0x400d6db0: file ../main/main.c, line 906.
Continuing.
esp32.cpu0: Target halted, PC=0x400D6DB0, debug_reason=00000001
Set GDB target to 'esp32.cpu0'
esp32.cpu1: Target halted, PC=0x401398A6, debug_reason=00000000
[New Thread 1073468800]
[New Thread 1073466144]
[New Thread 1073456692]
[New Thread 1073461140]
[New Thread 1073455228]
[New Thread 1073443928]
[Switching to Thread 1073465796]

Thread 1 hit Temporary breakpoint 1, app_main () at ../main/main.c:906
906	{
[Inferior 1 (Remote target) detached]
```

### Expected behavior

It t should hit app_main and stop in app_main for debug.

### Screenshots

[image]
This is launch.json file

erhankur, 2022-05-06:
@GavinChung can you please share  the openocd log with debug level 3. Explained [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb)

erhankur, 2022-07-06:
Closed due to inactivity.