espressif/openocd-esp32 #226: OpenOCD can't debug BluFi example...

davidlehrian, 2022-04-23:
### Development Kit

ESP32 DevkitV1

### Module or chip used

ESP32

### Debug Adapter

ESP-PROG

### OpenOCD version

C:\Users\lehrian\espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/openocd.exe

### Operating System

Windows 10

### Using an IDE ?

Eclipse

### OpenOCD command line

C:\Users\lehrian\espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin/openocd.exe

### JTAG Clock Speed

default - 20000kHz

### ESP-IDF version

4.4.1

### Problem Description

Create new project from BluFi example.
Build and flash device.
Create debug configuration and change Board: to ESP32 chip (via ESP-PROG)
select the new configuration and try to run debug
I get the following:

Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Flashing C:/Users/lehrian/Documents/ESP32/blufi2/build/partition_table/partition-table.bin at 0x8000
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x40175982, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x40175982, debug_reason=00000000
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
[... 314 lines cut ...]
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Warning:
Cannot insert breakpoint 3.
Cannot access memory at address 0x400d70b4
Cannot insert breakpoint 1.
Cannot access memory at address 0x400d7294
Cannot insert breakpoint 4.
Cannot access memory at address 0x400d7297

Command aborted.
```

### Expected behavior

Expect to be able to run the demo app in a debugger.

davidlehrian, 2022-04-23:
It seems that now that it has failed once, if I attempt to debug it again I get the following:

Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Flashing C:/Users/lehrian/Documents/ESP32/blufi2/build/partition_table/partition-table.bin at 0x8000
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000001
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Too many flash mappings -1! Must be 2.
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Failed to get flash size!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32.cpu0.flash'!
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Too many flash mappings -1! Must be 2.
Warn : Failed to get flash mappings (-4)!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Failed to get flash size!
Info : esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked

The the following is displayed in a popup:

Error in final launch sequence
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
Remote communication error.  Target disconnected.: (undocumented errno 10061).
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
Remote communication error.  Target disconnected.: (undocumented errno 10061).
Remote communication error.  Target disconnected.: (undocumented errno 10061).

erhankur, 2022-04-23:
@davidlehrian Looks like you have the same [issue](https://github.com/espressif/openocd-esp32/issues/223#issuecomment-1094085181)

Can you please try with the attached version in the link?

davidlehrian, 2022-04-23:
That seems to fix the issue. Thank you so much for the quick turnaround!

erhankur, 2022-08-16:
[release](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20220706) has fixed the issue.

closed 2022-08-16 by erhankur