espressif/openocd-esp32 #227: Trouble building from sources

brentonjudge, 2022-05-03:
### Development Kit

ESP32-C3

### Module or chip used

ESP32-C3-WROOM-02

### Debug Adapter

UART

### OpenOCD version

20220411

### Operating System

Windows

### Using an IDE ?

No

### OpenOCD command line

N/A

### JTAG Clock Speed

n/A

### ESP-IDF version

N/A

### Problem Description

When attempting to build under MingW32 am getting an error with jimtcl 

make  all-recursive
make[1]: Entering directory '/home/brenton/esp/openocd-esp32'
Making all in jimtcl
make[2]: Entering directory '/home/brenton/esp/openocd-esp32/jimtcl'
make[2]: *** No rule to make target 'all'.  Stop.
make[2]: Leaving directory '/home/brenton/esp/openocd-esp32/jimtcl'
make[1]: *** [Makefile:5157: all-recursive] Error 1
make[1]: Leaving directory '/home/brenton/esp/openocd-esp32'
make: *** [Makefile:2092: all] Error 2
 

### Expected behavior

Should compile

gerekon, 2022-05-03:
Hi,

Did you do the following steps before building?
1) `git submodule update --init --recursive`
2) `./bootstrap`
3) `configure <any opts>`

What MinGW distribution do you use? What shell do you use for building: msys, cmd, powershell?

brentonjudge, 2022-05-04:
Thanks for the response.

1-3 yes. following instructions and options from https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/building-openocd-windows.html (pasting all parts), so configure was
./configure --disable-doxygen-pdf --enable-ftdi --enable-jlink --enable-ulink --build=i686-w64-mingw32 --host=i686-w64-mingw32

msys2-x86_64-20220503 and using mingw32 shell.

$ ./configure --disable-doxygen-pdf --enable-ftdi --enable-jlink --enable-ulink --build=i686-w64-mingw32 --host=i686-w64-mingw32
configure: loading site script /etc/config.site
checking for makeinfo... no
configure: WARNING: Info documentation will not be built.
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for i686-w64-mingw32-gcc... i686-w64-mingw32-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether i686-w64-mingw32-gcc accepts -g... yes
checking for i686-w64-mingw32-gcc option to enable C11 features... none needed
checking whether i686-w64-mingw32-gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of i686-w64-mingw32-gcc... gcc3
checking for i686-w64-mingw32-ranlib... no
checking for ranlib... ranlib
checking for i686-w64-mingw32-pkg-config... /mingw32/bin/i686-w64-mingw32-pkg-config
checking pkg-config is at least version 0.23... yes
checking build system type... i686-w64-mingw32
checking host system type... i686-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by i686-w64-mingw32-gcc... C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw32/bin/nm -B
checking the name lister (/mingw32/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert i686-w64-mingw32 file names to i686-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert i686-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw32/i686-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for i686-w64-mingw32-objdump... no
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for i686-w64-mingw32-dlltool... no
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for i686-w64-mingw32-ar... no
checking for ar... ar
checking for archiver @FILE support... @
checking for i686-w64-mingw32-strip... no
checking for strip... strip
checking for i686-w64-mingw32-ranlib... ranlib
checking command to parse /mingw32/bin/nm -B output from i686-w64-mingw32-gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for i686-w64-mingw32-mt... no
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
[... 207 lines cut ...]
ST-Link Programmer                      yes (auto)
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes
Altera USB-Blaster II Compatible        yes (auto)
Bitbang mode of FT232R based devices    yes (auto)
Versaloon-Link JTAG Programmer          yes (auto)
TI XDS110 Debug Probe                   yes (auto)
CMSIS-DAP v2 Compliant Debugger         yes (auto)
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Olimex ARM-JTAG-EW Programmer           yes (auto)
Raisonance RLink JTAG Programmer        yes (auto)
USBProg JTAG Programmer                 yes (auto)
Andes JTAG Programmer                   yes (auto)
Espressif JTAG Programmer               yes (auto)
CMSIS-DAP Compliant Debugger            no
Nu-Link Programmer                      no
Cypress KitProg Programmer              no
Altera USB-Blaster Compatible           no
ASIX Presto Adapter                     no
OpenJTAG Adapter                        no
Linux GPIO bitbang through libgpiod     no
SEGGER J-Link Programmer                yes
Bus Pirate                              no
Use Capstone disassembly framework      no

brentonjudge, 2022-05-04:
Hi, external downloaded the jimtcl libraries and then copied them over seems to have resolved the issue.  Not sure what the difference was but it resolved the problem.  Unfortunately it didn't resolve the issue in programming the ESP32C3 chips.

closed 2022-05-04 by brentonjudge

gerekon, 2022-05-04:
Нуыб Ш сруслув нщг сщYes, I checked your config and build went successful. Looks like you had problems with updating `jimtcl` submodule.