espressif/openocd-esp32 #228: Unable to program blank ESP32-C3 chips

brentonjudge, 2022-05-04:
### Development Kit

Custom board

### Module or chip used

ESP

### Debug Adapter

On board USB/Built-In

### OpenOCD version

esp32-win32-0.11.0-esp32-20220411 and earlier

### Operating System

Windows 10

### Using an IDE ?

N/A

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg -c "program_esp partition-table.bin 0x8000 verify exit"

### JTAG Clock Speed

40000KHz (default)

### ESP-IDF version

v.4.4.1

### Problem Description

Run the command  and then get the output detailed blow.  The modules have been assembled onto boards and appear to be functioning but were not programmed before being assembled.  I had wondered if it was a limitation of openocd to program unprogrammed chips.

Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : esp_usb_jtag: serial (7C:DF:A1:BA:81:90)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Found 8 triggers
Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
 
I thought it may be related to https://github.com/espressif/openocd-esp32/issues/176 and so I tried the fix that chuanjinpang made but it is still unable to resolve the issue.  It then can't detect the size still from the probe of flash.  I then fixed this at 4096x1024 which passed these parts, but it still failed to program the flash even after these steps - the output is shown below

### Expected behavior

Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : clock speed 5000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 27 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 73 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
** Programming Started **
Info : PROF: Data transferred in 146.681 ms @ 136.35 KB/s
** Programming Finished **
** Verify Started **
** Verified OK **
shutdown command invoked

brentonjudge, 2022-05-04:
Wouldn't allow me to do the whole debug log so including the tail here

Debug: 1940 10343 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381a00
Debug: 1941 10350 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381a80
Debug: 1942 10359 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381b00
Debug: 1943 10366 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381b80
Debug: 1944 10375 riscv-013.c:2858 log_mem_access_result(): Succeeded to write memory via system bus.
Debug: 1945 10380 target.c:2466 target_write_buffer(): writing buffer of 268 byte at 0x40381c00
Debug: 1946 10387 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381c00
Debug: 1947 10394 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381c80
Debug: 1948 10401 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x40381d00
Debug: 1949 10409 riscv-013.c:2858 log_mem_access_result(): Succeeded to write memory via system bus.
Debug: 1950 10414 esp_algorithm.c:320 algorithm_load_func_image(): addr 0x00000000, sz 769, flags 0
Debug: 1951 10419 esp_algorithm.c:352 algorithm_load_func_image(): DATA sec size 769 -> 772
Debug: 1952 10424 esp_algorithm.c:357 algorithm_load_func_image(): BSS sec size 289 -> 292
Debug: 1953 10429 target.c:2116 alloc_working_area_try_do(): allocated new working area of 1064 bytes at address 0x3fc84000
Debug: 1954 10437 target.c:1983 print_wa_layout():  * 0x3fc84000-0x3fc84427 (1064 bytes)
Debug: 1955 10440 target.c:1983 print_wa_layout():    0x3fc84428-0x3fca3fff (130008 bytes)
Debug: 1956 10445 target.c:2466 target_write_buffer(): writing buffer of 512 byte at 0x3fc84000
Debug: 1957 10452 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x3fc84000
Debug: 1958 10459 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x3fc84080
Debug: 1959 10468 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x3fc84100
Debug: 1960 10476 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x3fc84180
Debug: 1961 10485 riscv-013.c:2858 log_mem_access_result(): Succeeded to write memory via system bus.
Debug: 1962 10490 target.c:2466 target_write_buffer(): writing buffer of 257 byte at 0x3fc84200
Debug: 1963 10498 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x3fc84200
Debug: 1964 10506 riscv-013.c:3680 write_memory_bus_v1(): transferring burst starting at address 0x3fc84280
Debug: 1965 10515 riscv-013.c:2858 log_mem_access_result(): Succeeded to write memory via system bus.
Debug: 1966 10521 riscv-013.c:2908 mem_should_skip_sysbus(): Skipping mem write via system bus - unsupported size.
Debug: 1967 10529 riscv-013.c:3815 write_memory_progbuf(): writing 1 words of 1 bytes to 0x3fc84300
Debug: 1968 10534 riscv-013.c:800 execute_abstract_command(): command=0x221008; access register, size=32, postexec=0, transfer=1, write=0, regno=0x1008
Debug: 1969 10541 riscv-013.c:1504 register_read_direct(): {0} s0 = 0x0
Debug: 1970 10547 riscv-013.c:800 execute_abstract_command(): command=0x221009; access register, size=32, postexec=0, transfer=1, write=0, regno=0x1009
Debug: 1971 10557 riscv-013.c:1504 register_read_direct(): {0} s1 = 0x0
Debug: 1972 10561 program.c:35 riscv_program_write(): debug_buffer[00] = DASM(0x00940023)
Debug: 1973 10567 program.c:35 riscv_program_write(): debug_buffer[01] = DASM(0x00140413)
Debug: 1974 10572 program.c:35 riscv_program_write(): debug_buffer[02] = DASM(0x00100073)
Debug: 1975 10577 riscv-013.c:4381 riscv013_write_debug_buffer(): cache hit for 0x100073 @2
Debug: 1976 10581 riscv-013.c:3872 write_memory_progbuf(): writing until final address 0x000000003fc84301
Debug: 1977 10585 riscv-013.c:3875 write_memory_progbuf(): transferring burst starting at address 0x000000003fc84300
Debug: 1978 10590 riscv-013.c:1315 register_write_direct(): {0} s0 <- 0x3fc84300
Debug: 1979 10596 riscv-013.c:800 execute_abstract_command(): command=0x231008; access register, size=32, postexec=0, transfer=1, write=1, regno=0x1008
Debug: 1980 10604 riscv-013.c:800 execute_abstract_command(): command=0x271009; access register, size=32, postexec=1, transfer=1, write=1, regno=0x1009
Debug: 1981 10614 batch.c:91 riscv_batch_run(): Ignoring empty batch.
Debug: 1982 10619 riscv-013.c:3955 write_memory_progbuf(): successful (partial?) memory write
Debug: 1983 10625 riscv-013.c:1315 register_write_direct(): {0} s1 <- 0x0
Debug: 1984 10629 riscv-013.c:800 execute_abstract_command(): command=0x231009; access register, size=32, postexec=0, transfer=1, write=1, regno=0x1009
Debug: 1985 10637 riscv-013.c:1315 register_write_direct(): {0} s0 <- 0x0
Debug: 1986 10642 riscv-013.c:800 execute_abstract_command(): command=0x231008; access register, size=32, postexec=0, transfer=1, write=1, regno=0x1008
Debug: 1987 10650 program.c:35 riscv_program_write(): debug_buffer[00] = DASM(0x0000100f)
Debug: 1988 10664 program.c:35 riscv_program_write(): debug_buffer[01] = DASM(0x0000000f)
Debug: 1989 10669 program.c:35 riscv_program_write(): debug_buffer[02] = DASM(0x00100073)
Debug: 1990 10673 riscv-013.c:4381 riscv013_write_debug_buffer(): cache hit for 0x100073 @2
[... cut ...]
Debug: 4041 21296 riscv-013.c:800 execute_abstract_command(): command=0x230c10; access register, size=32, postexec=0, transfer=1, write=1, regno=0xc10
Debug: 4042 21306 riscv.c:3501 riscv_set_register(): [esp32c3] wrote 0x3 to hpmcounter16 valid=0
Debug: 4043 21311 esp_riscv.c:481 esp_riscv_wait_algorithm(): restore priv
Debug: 4044 21316 riscv.c:3906 register_set(): [esp32c3] write 0x03 to priv (valid=0)
Debug: 4045 21320 riscv.c:3482 riscv_set_register(): [esp32c3] priv <- 3
Debug: 4046 21323 riscv-013.c:4115 riscv013_set

brentonjudge, 2022-05-04:
=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                           Set this bit to vdd spi pin function as gpio       = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enable btlc gpio                                   = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                            Set this bit to enable power glitch function       = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)                       Sample delay configuration of power glitch         = 0 R/W (0b00)
DIS_LEGACY_SPI_BOOT (BLOCK0)                       Disables Legacy SPI boot mode                      = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)                        Selects the default UART for printing boot msg     = UART0 R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)                         Force ROM code to send a resume command during SPI = False R/W (0b0)
                                                    bootduring SPI boot
BLOCK_USR_DATA (BLOCK3)                            User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Flash Config fuses:
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms

Identity fuses:
SECURE_VERSION (BLOCK0)                            Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                                   ure)
MAC (BLOCK1)                                       Factory MAC Address
   = 7c:df:a1:ba:81:90 (OK) R/W
WAFER_VERSION (BLOCK1)                             WAFER version                                      = 3 R/W (0b011)
PKG_VERSION (BLOCK1)                               Package version                                    = ESP32-C3 R/W (0b000)
BLOCK1_VERSION (BLOCK1)                            BLOCK1 efuse version                               = 0 R/W (0b000)
OPTIONAL_UNIQUE_ID (BLOCK2)                        Optional unique 128-bit ID
   = 57 2f 06 11 5c b9 e1 df de 35 fa 7c 33 13 d2 0e R/W
BLOCK2_VERSION (BLOCK2)                            Version of BLOCK2                                  = 7 R/W (0b111)
CUSTOM_MAC (BLOCK3)                                Custom MAC Address
   = 00:00:00:00:00:00 (OK) R/W

Jtag Config fuses:
SOFT_DIS_JTAG (BLOCK0)                             Software disables JTAG. When software disabled, JT = 0 R/W (0b000)
                                                   AG can be activated temporarily by HMAC peripheral
DIS_PAD_JTAG (BLOCK0)                              Permanently disable JTAG access via pads. USB JTAG = False R/W (0b0)
                                                    is controlled separately.

Security fuses:
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)               Disables flash encryption when in download boot mo = False R/W (0b0)
                                                   des
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables encryption and decryption, when an SPI boo = Disable R/W (0b000)
                                                   t mode is set. Enabled when 1 or 3 bits are set,di
                                                   sabled otherwise
SECURE_BOOT_KEY_REVOKE0 (BLOCK0)                   If set, revokes use of secure boot key digest 0    = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE1 (BLOCK0)                   If set, revokes use of secure boot key digest 1    = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE2 (BLOCK0)                   If set, revokes use of secure boot key digest 2    = False R/W (0b0)
KEY_PURPOSE_0 (BLOCK0)                             KEY0 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_1 (BLOCK0)                             KEY1 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_2 (BLOCK0)                             KEY2 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_3 (BLOCK0)                             KEY3 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_4 (BLOCK0)                             KEY4 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_5 (BLOCK0)                             KEY5 purpose                                       = USER R/W (0x0)
[... cut ...]
                                                   led separately
DIS_USB_DEVICE (BLOCK0)                            Disables USB DEVICE                                = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)                     Disables use of USB in UART download boot mode     = False R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = False R/W (0b0)

gerekon, 2022-05-04:
Hi @brentonjudge 

Do you see this problem if chip's flash is not empty? I mean to try flash with `idf.py` and then try flashing with OpenOCD.

brentonjudge, 2022-05-04:
Same issue.  If I do it immediately after executing "esptool.py write_flash 0x8000 build/partition_table/partition-table.bin" then there is a different message, but if I unplug the USB and plug it back in (external supply so should be no power reset), it goes back to the same message.  

The 'different message' is:
am_esp partition-table.bin 0x8000 verify exit"
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Error: Abstract command ended in error 'not supported' (abstractcs=0x10001202)
Error: Timed out after 5s waiting for busy to go low (abstractcs=0x10001202). Increase the timeout with riscv set_command_timeout_sec.
[repeating until crash]

gerekon, 2022-05-04:
Could you do the following
1) Flash example project (e.g. `blink`) using `idf.py flash`
2) Run OpenOCD `openocd -d 3 -f boards/esp32c3-builtin.cfg` -c 'halt;flash probe 0'.
3) Post OpenOCD output here

brentonjudge, 2022-05-05:
Output from the bootloader (in case of assistance)

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380526
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16bc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (24) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (24) boot: compile time 00:12:27
I (24) boot: chip revision: 3
I (26) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=06cb8h ( 27832) map
I (96) esp_image: segment 1: paddr=00016ce0 vaddr=3fc8a200 size=0146ch (  5228) load
I (101) esp_image: segment 2: paddr=00018154 vaddr=40380000 size=07ec4h ( 32452) load
I (114) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=12a04h ( 76292) map
I (129) esp_image: segment 4: paddr=00032a2c vaddr=40387ec4 size=02244h (  8772) load
I (131) esp_image: segment 5: paddr=00034c78 vaddr=50000010 size=00010h (    16) load
I (138) boot: Loaded app from partition at offset 0x10000
I (141) boot: Disabling RNG early entropy source...
I (157) cpu_start: Pro cpu up.
I (166) cpu_start: Pro cpu start user code
I (166) cpu_start: cpu freq: 160000000
I (166) cpu_start: Application information:
I (169) cpu_start: Project name:     hello_world
I (174) cpu_start: App version:      1
I (179) cpu_start: Compile time:     May  5 2022 00:10:42
I (185) cpu_start: ELF file SHA256:  2398be86643e1537...
I (191) cpu_start: ESP-IDF:          v4.4.1-dirty
I (196) heap_init: Initializing. RAM available for dynamic allocation:
I (203) heap_init: At 3FC8C4D0 len 00033B30 (206 KiB): DRAM
I (209) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (216) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (223) spi_flash: detected chip: generic
I (227) spi_flash: flash io: dio
I (231) sleep: Configure to isolate all GPIO pins in sleep state
I (238) sleep: Enable automatic switching of GPIO sleep configuration
I (245) cpu_start: Starting scheduler.

gerekon, 2022-05-05:
@brentonjudge Could you try Windows distro from [this message](https://github.com/espressif/openocd-esp32/issues/223#issuecomment-1094085181)? Looks like you can have similar issue.

gerekon, 2022-05-05:
@brentonjudge sorry, we need to prepare a bit different distro. That one fixes issue on ESP32 only. I' ll post new distro here later on

erhankur, 2022-05-05:
@brentonjudge  Temporary solution for all supported chips. 

[openocd-esp32-win32-0.11.0-esp32-20220411-7-g4589234da.zip](https://github.com/espressif/openocd-esp32/files/8630434/openocd-esp32-win32-0.11.0-esp32-20220411-7-g4589234da.zip)

brentonjudge, 2022-05-05:
With WinUSB

C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin>openocd -f board/esp32c3-builtin.cfg -c "program_esp partition-table.bin 0x8000 verify exit"
Open On-Chip Debugger  v0.11.0-esp32-20220411-7-g4589234d (2022-05-05-10:20)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (7C:DF:A1:BA:81:8C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Error: Trying to use configured scan chain anyway...
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
[... 56 lines cut ...]
Error: missing data from bitq interface
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Warn : Bypassing JTAG setup events due to errors
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: Unsupported DTM version: 8
Error: Unsupported DTM version: 8
[crash]

erhankur, 2022-05-06:
@brentonjudge Can you follow the steps in the [link](https://github.com/espressif/idf-installer/issues/133#issuecomment-1098132212) Driver issue is reported by the user and we are currently working on it. Sorry for the inconvenience.