espressif/openocd-esp32 #230: Failed to debug

FanhuaCloud, 2022-05-20:
### Development Kit

No devkit

### Module or chip used

ESP32-S3-WROOM-1-N8R8

### Debug Adapter

builtin usb jtag

### OpenOCD version

v0.11.0-esp32-20220411

### Operating System

Windows

### Using an IDE ?

vscode

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

v5.0

### Problem Description

OpenOCD server failed to start because of Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED

### Debug Logs

```shell
OpenOCD server failed to start because of Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Error: libusb_
```

### Expected behavior

This shouldn't be a problem

erhankur, 2022-05-23:
@FanhuaCloud Can you please make sure the usb-jtag driver installed correctly? 

You can install the driver package from [here](https://dl.espressif.com/dl/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15.zip) 

I suggest to remove all previous drivers for the usb-jtag device. Then you can make a clean installation.  

If your issue still continues, please check the registry entries. Should be similar to this. 

[image]

Under `VID_303A&PID_1001&MI_02` Device Parameters section must have `DeviceInterfaceGUIDs` key.

FanhuaCloud, 2022-05-23:
Yes, I reinstalled the driver and checked the UUID, but it still doesn't work, could it be a problem with the idf plugin?
The following is the log of openocd:
```
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (F4:12:FA:43:4D:70)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

```

FanhuaCloud, 2022-05-23:
And, this is my regedit
[image]

erhankur, 2022-05-23:
OK. So you don't have a driver issue anymore. Can you load esp-idf hello_world example and test it again?  

And if there is still an issue, please share full OpenOCD logs described in [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb).

FanhuaCloud, 2022-05-23:
> OK. So you don't have a driver issue anymore. Can you load esp-idf hello_world example and test it again?
>
> And if there is still an issue, please share full OpenOCD logs described in [...]

Thank you, after my attempts, the problem does not exist with ESP-IDF4.4.1, I am currently reinstalling back to the master branch to confirm whether it is a problem with ESP-IDF

FanhuaCloud, 2022-05-23:
Ok, probably it is a problem with the master branch, the following is the log of openocd
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/8755296/openocd_log.txt)

FanhuaCloud, 2022-05-23:
> OK. So you don't have a driver issue anymore. Can you load esp-idf hello_world example and test it again?
>
> And if there is still an issue, please share full OpenOCD logs described in [...]

Ok, probably it is a problem with the master branch, the following is the log of openocd
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/8755302/openocd_log.txt)

erhankur, 2022-05-23:
Does your application working as expected with master branch? Is it possible to send the binary for my test?

erhankur, 2022-05-24:
I don't expect to see error messages in the serial terminal. I am trying to understand in which case you are starting to see debugging issues . Monitoring/not monitoring,  after app restart.
Btw, I am trying with esp-idf hello-world example. I didn't modify the code. Is this the same with your test app?

FanhuaCloud, 2022-05-25:
> I don't expect to see error messages in the serial terminal. I am trying to understand in which case you are starting to see debugging issues . Monitoring/not monitoring, after app restart. Btw, I [...]

Yes, it is the same

closed 2022-05-31 by FanhuaCloud

FanhuaCloud, 2022-06-03:
> Still did not helped me. openocd cannot start and cannot find jtag.

Please check whether the corresponding driver is installed and the JTAG configuration used is correct

pravinkpatel, 2022-06-05:
Same issue facing

Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Flashing D:/ESP/Projects/softAP/build/partition_table/partition-table.bin at 0x8000
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'
** OpenOCD init failed **
shutdown command invoked
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Assertion failed!

Program: D:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0

suskach, 2022-06-06:
It looks like on my side the dos window which I need to run openocd has to run as administrator,