espressif/openocd-esp32 #231: Failed to get flash maps - esp32c3-mini-1

Ardelean-Calin, 2022-06-17:
### Development Kit

No development kit - custom board

### Module or chip used

ESP32C3-Mini-1 (Rev 3) with 4MB Flash

### Debug Adapter

Integrated USB-JTAG

### OpenOCD version

v0.11.0-esp32-20220411

### Operating System

Linux

### Using an IDE ?

No

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

ESP-IDF v5.0-dev-3485-g20f5e180ee

### Problem Description

Using the[ ESP32-C3-MINI-1 ](https://datasheet.lcsc.com/lcsc/2108141930_Espressif-Systems-ESP32-C3-MINI-1-N4_C2838502.pdf)module on a custom board. The integrated JTAG + UART interface (via USB) is detected by my operating system, however I cannot debug any application via openocd, not even the blinky example attached here. 

I can successfully flash the device using [esptool.py](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/index.html), but when I try to flash via OpenOCD I get:

```bash
[calin@compilebox blink]$ openocd -f board/esp32c3-builtin.cfg -c "program_esp build/blink.bin 0x10000 verify exit"
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : esp_usb_jtag: serial (84:F7:03:67:B5:A8)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Found 8 triggers
Error: Failed to get flash maps (4294967295)!
Warn : Failed to get flash mappings (-4)!
Error: Failed to get flash size!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
```

The same error happens if I run
`openocd -f board/esp32c3-builtin.cfg`

and then try to connect via gdb in another terminal
`riscv32-esp-elf-gdb -ex "target extended-remote :3333"`

There seems to be some problem with openocd getting the flash maps for the ESP32­-C3-­MINI-­1 module. 

I have tried compiling openocd from sources and running it via a debugger, but I can't seem to be able to debug the `stub_flasher.c` file and the culprit seems to be there in the `stub_flash_get_map()` function. Any help on how to debug that function so that I can get more details would be highly appreciated!

I have attached the log from the following command:
`openocd -d3 -l openocd.log -f board/esp32c3-builtin.cfg -c "program_esp build/blink.bin 0x10000 verify exit"`

**Log:** [openocd.log](https://github.com/espressif/openocd-esp32/files/8927887/openocd.log)
**Binary files**: [blink.zip](https://github.com/espressif/openocd-esp32/files/8927903/blink.zip)

### Debug Logs

```shell
I have attached the logs as attachment so as to not flood this issue with a super long text.
```

### Expected behavior

I would expect to get a list of flash mappings and a successful connection via gdb.

erhankur, 2022-06-17:
I tried with the latest OpenOCD master and your idf branch. Also with your binary file. There is no issue on my side. Tested ESp32c3-MINI-1 from the usb jtag port. 

Your commands also look correct. 

I could ask to disable memory protection, but during the programming , OpenOCD first resets the chip and protection will be disabled. 

If you need to debug the target without programming it, you can disable  flash support by setting ESP_FLASH_SIZE to 0.

I will turn back if I have something new in my mind to try.

Ardelean-Calin, 2022-06-17:
Thank you for trying on your end, as well! If you say that on your end everything works as expected, then the problem might be with the ESP32C3 Mini Module I got from my parts supplier (JLCPCB) or worst case with my hardware design (even though I have just a simple two wire connection to the Data+ and Data- USB lines as seen in the picture below)
[image]

I can try and offer more information regarding the module found on my board, for example, this is the output of
`esptool.py get_security_info`:

[image]

Ardelean-Calin, 2022-06-17:
Sorry for the spam, I just wanted to attach the output of `espefuse.py summary`, as well. Maybe it helps...
[fuses.txt](https://github.com/espressif/openocd-esp32/files/8928845/fuses.txt)

erhankur, 2022-07-06:
@Ardelean-Calin did you add 1K5 resistor between D+ and 3V3. I couldn't see from the picture.

erhankur, 2022-09-01:
Closing due to inactivity.  if you have still an issue feel free to re-open.

closed 2022-09-01 by erhankur