espressif/openocd-esp32 #233: ESP32-C3: failing to connect GDB

tcbennun, 2022-06-27:
### Development Kit

N/A

### Module or chip used

ESP32-C3-MINI-1

### Debug Adapter

Built-in USB

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Debian Linux (kernel 5.18)

### Using an IDE ?

N/A

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

Any (tried default, tried 1 MHz)...

### ESP-IDF version

v4.4.1

### Problem Description

1. Connect module to PC via USB cable.
2. Run `openocd -l openocd.log -d3 -f board/esp32c3-builtin.cfg` in a new terminal with ESP-IDF already exported.
3. Create file named `gdbinit` with the following contents:
```
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
```
4. Run `riscv32-esp-elf-gdb -ex "set remotelogfile gdb.log" -x gdbinit build/blecent.elf` in another new terminal with ESP-IDF already exported.

The result is a failure for GDB to connect to the device. OpenOCD initially connects correctly, but problems occur when GDB is invoked. See logs.

### Debug Logs

```shell
OpenOCD log: https://gist.github.com/tcbennun/e1625a2c6fd85873721a51334ed65098

GDB log:

w +$qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df
r +<Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -+$vMustReplyEmpty#3a
r <Timeout: 2 seconds>
w $vMustReplyEmpty#3a
r <Timeout: 2 seconds>
w $vMustReplyEmpty#3a
r <Timeout: 2 seconds>
w $vMustReplyEmpty#3a
r <Timeout: 2 seconds><Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -+
End of log
```

### Expected behavior

Expected GDB to connect to OpenOCD and for the program to stop at a temporary breakpoint at the start of `app_main`.

erhankur, 2022-06-27:
Do you see the same issue with hello_world example?

tcbennun, 2022-06-27:
I get the same in the hello_world example, @erhankur.

However, as it turns out, this is apparently fixed by forcing the chip into Download Boot by pulling GPIO9 low on powerup. (Not yet sure whether this is a permanent fix, but I've managed to hit the entry point successfully.)

Is this usually a requirement?

tcbennun, 2022-06-27:
I'm now finding that GDB freezes after hitting that initial breakpoint - I can't enter any commands at all. If I interrupt with Ctrl+C, I can enter commands but nothing appears to actually happen...? OpenOCD reports nothing out of sorts?

tcbennun, 2022-06-27:
... Yeah, so going into Download Boot means that the program doesn't execute at all... the builtin JTAG doesn't seem to notice and just freezes as if execution is continuing.

gerekon, 2022-06-28:
@tcbennun 

> `openocd -l openocd.log -d3 -f board/esp32c3-builtin.cfg`

Could you post collected `openocd.log`?

gerekon, 2022-06-28:
Sorry. Can not find attached file with openocd log. I see only GDB log posted as a part of the initial message.

> Does your program uses WIFI/BLE?

In IDF 4.4.1 USB JTAG operation can be corrupted due USB PHY is disabled (https://github.com/espressif/esp-idf/issues/8046, https://github.com/espressif/arduino-esp32/issues/6264). 
To overcome this you can 
- use add a fix described [here](https://github.com/espressif/arduino-esp32/pull/6287) 
- or config builtin USB/CDC to be used as a console (see [ESP_PHY_ENABLE_USB](https://github.com/espressif/esp-idf/blob/v4.4.1/components/esp_phy/Kconfig#L107))

tcbennun, 2022-06-28:
@gerekon 

The gist link is there... it's in the verbose-styled part of the issue just under the "Debug Logs" header. Here's the link again: https://gist.github.com/tcbennun/e1625a2c6fd85873721a51334ed65098

> Does your program uses WIFI/BLE?

It does use BLE. I just retried hello_world with success this time - so I assume it's PHY related after all.

I already had ESP_PHY_ENABLE_USB enabled, and it looks like `esp_phy_enable()` calls `phy_bbpll_en_usb(true)` already in ESP-IDF 4.4.1.

erhankur, 2023-12-13:
@tcbennun If you still have an issue with the latest OpenOCD fee free to re-open.

closed 2023-12-13 by erhankur