espressif/openocd-esp32 #234: ESP32S3 + JLink not working

higaski, 2022-07-05:
### Development Kit

ESP32-S3-DevKitC-1

### Module or chip used

ESP32-S3-WROOM1

### Debug Adapter

J-Link EDU

### OpenOCD version

v0.11.0-esp32-20220411

### Operating System

Linux

### Using an IDE ?

no

### OpenOCD command line

openocd -c "adapter speed 1000" -s share/openocd/scripts -f interface/jlink.cfg -f target/esp32s3.cfg

### JTAG Clock Speed

1000 kHz

### ESP-IDF version

ESP-IDF v5.0-dev-3654-gc2ccc383da

### Problem Description

1. Connect ESP32-S3-DevKitC-1 to JLink
2. Make sure fuses are set correctly (either set DIS_USB_JTAG or JTAG_SEL_ENABLE)
3. Run OpenOCD

### Debug Logs

```shell
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 1 options.c:63 configuration_output_handler(): 
Debug: 5 1 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 1 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 1 options.c:246 add_default_dirs(): exepath=/home/vinci/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin
Debug: 8 1 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 1 configuration.c:44 add_script_search_dir(): adding /home/vinci/.config/openocd
Debug: 10 1 configuration.c:44 add_script_search_dir(): adding /home/vinci/.openocd
Debug: 11 1 configuration.c:44 add_script_search_dir(): adding /home/vinci/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin/../share/openocd/site
Debug: 12 1 configuration.c:44 add_script_search_dir(): adding /home/vinci/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin/../share/openocd/scripts
Debug: 13 1 command.c:166 script_debug(): command - adapter speed 1000
Debug: 14 1 adapter.c:180 adapter_config_khz(): handle adapter khz
Debug: 15 1 adapter.c:144 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 16 1 adapter.c:144 adapter_khz_to_speed(): convert khz to adapter specific speed value
User : 17 1 options.c:63 configuration_output_handler(): adapter speed: 1000 kHz
User : 18 1 options.c:63 configuration_output_handler(): 
Debug: 19 1 command.c:166 script_debug(): command - ocd_find interface/jlink.cfg
Debug: 20 1 configuration.c:99 find_file(): found share/openocd/scripts/interface/jlink.cfg
Debug: 21 1 command.c:166 script_debug(): command - adapter driver jlink
Debug: 22 1 command.c:166 script_debug(): command - ocd_find target/esp32s3.cfg
Debug: 23 1 configuration.c:99 find_file(): found share/openocd/scripts/target/esp32s3.cfg
[... 406 lines cut ...]
Debug: 536 370 command.c:166 script_debug(): command - nand init
Debug: 537 370 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 538 370 command.c:166 script_debug(): command - pld init
Debug: 539 370 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 540 370 command.c:166 script_debug(): command - tpiu init
Info : 541 370 gdb_server.c:3742 gdb_target_start(): starting gdb server for esp32s3.cpu0 on 3333
Info : 542 370 server.c:303 add_service(): Listening on port 3333 for gdb connections
```

### Expected behavior

It is currently not possible to connect to an ESP32-S3 using OpenOCD v0.11.0-esp32-20220411 and JLink V7.66e.

For whatever reason OpenOCD always errors out ->
```sh
Error: 443 368 core.c:1117 jtag_examine_chain_check(): JTAG scan chain interrogation failed: all ones
Error: 444 368 core.c:1118 jtag_examine_chain_check(): Check JTAG interface, timings, target power, etc.
Error: 445 368 core.c:1548 jtag_init_inner(): Trying to use configured scan chain anyway...
Debug: 446 368 core.c:1358 jtag_validate_ircapture(): IR capture validation scan
Error: 456 368 core.c:1410 jtag_validate_ircapture(): esp32s3.cpu0: IR capture error; saw 0x1f not 0x01
```

I found suggestions before that this can be bad wiring or connections. This is defiantly not the case. I've also tried to debug a DevKitM which doesn't work either.

I've even tried to set different fuses on the two dev kits I own. On the DevKitC board I've only burned the `JTAG_SEL_ENABLE` fuse which seems to work as documented. On this board I can at least still use the internal USB-JTAG when pulling GPIO3 high. On the DevKitM board I've burned `DIS_USB_JTAG` and `DIS_USB_SERIAL_JTAG`.

erhankur, 2022-07-06:
@higaski Currently I don't have my jlink with me and can not try your configuration. 

Did you try debugging hello_world example with both boards? 
I assume, jtag pins are empty and you don't configure them from your application. 
Can you please send your jlink - devkit pin connections.

higaski, 2022-07-07:
I did try debugging the hello_world example (removed the reset though) with both boards yes.

What exactly do you mean which "configure them"? What else besides setting fuses is there? As far as I know the JTAG pins don't require any special initialization in software?

My pin connections are as follows
| ARM 2x5 debug connector | ESP32-S3-DevKitC/M |
| ----------------------------------- | -------------------------- | 
| VTref 1                               | 3V3 any                   |
| TMS 2                                | MTMS 42                 |
| GND 3                               | G any                        |
| TCK 4                                | MTCK 39                   |
| TDO 6                                | MTDO 40                  |
| TDI 8                                 | MTDI 41                     |

erhankur, 2022-07-07:
I meant, if the jtag pins configured as GPIO, there will be an issue during the OpenOCD communication.  But since you tried with the hello_world app, this is not the case. Your connection also looks good. 
Do you power the devkit from the jlink? Just to be sure, did you enable the Jlink power out functionality. I remember by default  it is disabled. 

I will give a try with Jlink EDU and turn back.

higaski, 2022-07-07:
I power both modules through the UART connector. To be honest, I didn't even know the JLink EDU can output power... :laughing:

erhankur, 2022-07-18:
Ok. Here is my test outputs.  I don't see an issue with my [Jlink-EDU(20 pin)](https://www.segger.com/products/debug-probes/j-link/technology/interface-description/)

I tried with the latest OpenOCD release and idf 5.0 hello_world example. JTAG_SEL_ENABLE is true in my ESP32S3-DevkitC and nothing connected to the GPIO3. 

IO42 - TMS - JLINK7
IO41 - TDI - JLINK5
IO40 - TDO - JLINK13
IO39 - TCK - JLINK9
ANY 3V3 - VTREF
ANY GND - ANY GND

For the next thing to do, maybe you can connect a logic analyzer and see what is going on with the jtag pins. 

```
$OOCD_DIR/src/openocd -s $OOCD_DIR/tcl -f interface/jlink.cfg -f target/esp32s3.cfg -c "adapter speed 1000" -d3
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-18-15:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:63 configuration_output_handler(): debug_level: 3
User : 4 2 options.c:63 configuration_output_handler():
Debug: 5 2 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 6 2 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 7 2 options.c:246 add_default_dirs(): exepath=/Users/erhan/dev/openocd-esp32/src
Debug: 8 2 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 2 configuration.c:44 add_script_search_dir(): adding /Users/erhan/Library/Preferences/org.openocd
Debug: 10 2 configuration.c:44 add_script_search_dir(): adding /Users/erhan/.config/openocd
Debug: 11 2 configuration.c:44 add_script_search_dir(): adding /Users/erhan/.openocd
Debug: 12 2 configuration.c:44 add_script_search_dir(): adding /Users/erhan/dev/openocd-esp32/src/../share/openocd/site
Debug: 13 2 configuration.c:44 add_script_search_dir(): adding /Users/erhan/dev/openocd-esp32/src/../share/openocd/scripts
Debug: 14 2 command.c:166 script_debug(): command - ocd_find interface/jlink.cfg
Debug: 15 2 configuration.c:99 find_file(): found /Users/erhan/dev/openocd-esp32/tcl/interface/jlink.cfg
Debug: 16 2 command.c:166 script_debug(): command - adapter driver jlink
Debug: 17 3 command.c:166 script_debug(): command - ocd_find target/esp32s3.cfg
Debug: 18 3 configuration.c:99 find_file(): found /Users/erhan/dev/openocd-esp32/tcl/target/esp32s3.cfg
Debug: 19 3 command.c:166 script_debug(): command - transport select jtag
Debug: 20 3 command.c:166 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 21 3 configuration.c:99 find_file(): found /Users/erhan/dev/openocd-esp32/tcl/bitsbytes.tcl
Debug: 22 3 command.c:166 script_debug(): command - expr 1 << $x
Debug: 23 3 command.c:166 script_debug(): command - expr $x + 1
Debug: 24 3 command.c:166 script_debug(): command - expr 1 << $x
Debug: 25 3 command.c:166 script_debug(): command - expr $x + 1
Debug: 26 3 command.c:166 script_debug(): command - expr 1 << $x
Debug: 27 3 command.c:166 script_debug(): command - expr $x + 1
Debug: 28 3 command.c:166 script_debug(): command - expr 1 << $x
Debug: 29 3 command.c:166 script_debug(): command - expr $x + 1
Debug: 30 3 command.c:166 script_debug(): command - expr 1 << $x
Debug: 31 3 command.c:166 script_debug(): command - expr $x + 1
Debug: 32 3 command.c:166 script_debug(): command - expr 1 << $x
Debug: 33 3 command.c:166 script_debug(): command - expr $x + 1
Debug: 34 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 35 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 36 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 37 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 38 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 39 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 40 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 41 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 42 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 43 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 44 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 45 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 46 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 47 12 command.c:166 script_debug(): command - expr $x + 1
Debug: 48 12 command.c:166 script_debug(): command - expr 1 << $x
Debug: 49 13 command.c:166 script_debug(): command - expr $x + 1
Debug: 50 13 command.c:166 script_debug(): command - expr 1 << $x
Debug: 51 13 command.c:166 script_debug(): command - expr $x + 1
Debug: 52 13 command.c:166 script_debug(): command - expr 1 << $x
Debug: 53 13 command.c:166 script_debug(): command - expr $x + 1
[... 562 lines cut ...]
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI D6 pad                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI D7 pad                                         = 0 R/W (0b000000)

Usb Config fuses:
DIS_USB (BLOCK0)                                   Disables the USB OTG hardware                      = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
EXT_PHY_ENABLE (BLOCK0)                            Enables external USB PHY                           = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enables BTLC GPIO                                  = 0 R/W (0b00)
DIS_USB_JTAG (BLOCK0)                              Disable usb_serial_jtag-to-jtag function           = False R/W (0b0)
[... cut ...]
VDD_SPI_TIEH (BLOCK0)                              The VDD_SPI power supply voltage at reset          = Connect to 1.8V LDO R/W (0b0)
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD_SPI R/W (0b1)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 1.8V by efuse.
```

higaski, 2022-07-18:
Ok, I have no idea why but all of the sudden it started working for me too.

```sh
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
```

The only thing that changed on my system in the meantime is the SEGGER version:
```sh
JLinkExe --version
SEGGER J-Link Commander V7.66f (Compiled Jul  5 2022 15:23:22)
DLL version V7.66f, compiled Jul  5 2022 15:23:08
```

When I initially posted this issue it has been 7.66e.
Maybe it has been "their" issue all along?

What JLink version have you been using?

erhankur, 2022-07-18:
This is what I have. 
```
C:\ProgramData\Microsoft\Windows\Start Menu\Programs\SEGGER - J-Link V6.95d>"J-Link Commander V6.95d.lnk"
SEGGER J-Link Commander V6.95d (Compiled Feb 12 2021 16:39:44)
DLL version V6.95d, compiled Feb 12 2021 16:38:31

Connecting to J-Link via USB...O.K.
Firmware: J-Link V10 compiled Feb  4 2021 12:58:41
Hardware version: V10.10
S/N: 260106593
License(s): FlashBP, GDB
OEM: SEGGER-EDU
VTref=0.000V
```

> ok nevermind, I've tried downgrading to 7.66e and its still working... I apologize for wasting your time...

No problem. I am closing this issue.

closed 2022-07-18 by erhankur

mfeil-leviton, 2023-07-31:
Did you ever find out what the original issue was?

higaski, 2023-08-01:
Not really no sry.