espressif/openocd-esp32 #235: ESP32-S3 JTAG device error?

petemillermultimedia, 2022-07-06:
### Development Kit

ESP-S3-DevKitC-1

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

ESP-ROG

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows

### Using an IDE ?

VSCode

### OpenOCD command line

default VSCode command

### JTAG Clock Speed

default

### ESP-IDF version

4.4.1

### Problem Description

Flash with JTAG from within VSCode produces error:

unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

I've followed the instructions in various Espressif docs and help forums, but the main clue is that my Windows system lists the USB JTAG with vendor ID 303A; changing the driver doesn't seem to change this. Is this a mismatch with the command (seems to expect ID 0403)?

### Debug Logs

```shell
User : 14 5 options.c:63 configuration_output_handler(): debug_level: 3
User : 15 5 options.c:63 configuration_output_handler(): 
Debug: 16 5 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 17 5 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 18 5 options.c:246 add_default_dirs(): exepath=C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 19 5 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 20 5 configuration.c:42 add_script_search_dir(): adding C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
Debug: 21 5 configuration.c:42 add_script_search_dir(): adding C:/Users/pmiller/AppData/Roaming/OpenOCD
Debug: 22 5 configuration.c:42 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 23 5 configuration.c:42 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 24 6 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/board/esp32s3-ftdi.cfg
Debug: 25 9 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg
Debug: 26 9 command.c:146 script_debug(): command - adapter driver ftdi
Debug: 28 9 command.c:146 script_debug(): command - ftdi_vid_pid 0x0403 0x6010 0x0403 0x6014
Debug: 30 9 command.c:146 script_debug(): command - ftdi_channel 0
Debug: 32 9 command.c:146 script_debug(): command - ftdi_layout_init 0x0008 0xf00b
Debug: 34 9 command.c:146 script_debug(): command - ftdi_layout_signal LED -data 0x1000
Debug: 36 9 command.c:146 script_debug(): command - ftdi_layout_signal LED2 -data 0x2000
Debug: 38 9 command.c:146 script_debug(): command - ftdi_layout_signal LED3 -data 0x4000
Debug: 40 9 command.c:146 script_debug(): command - ftdi_layout_signal LED4 -data 0x8000
Debug: 42 9 command.c:146 script_debug(): command - reset_config none
Debug: 44 9 command.c:146 script_debug(): command - adapter speed 20000
Debug: 46 9 core.c:1822 jtag_config_khz(): handle jtag khz
[... 191 lines cut ...]
Debug: 249 18 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mepc' (157) does not exist
Debug: 250 18 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'meps' (158) does not exist
Debug: 251 18 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mesave' (159) does not exist
Debug: 252 18 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mesr' (160) does not exist
Debug: 253 18 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mecr' (161) does not exist
Debug: 254 18 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mevaddr' (162) does not exist
Debug: 255 18 ftdi.c:650 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Error: 256 63 mpsse.c:216 open_matching_device(): no device found
Error: 257 63 mpsse.c:366 mpsse_open(): unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: 258 101 mpsse.c:216 open_matching_device(): no device found
Error: 259 101 mpsse.c:366 mpsse_open(): unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'
Debug: 260 103 command.c:629 run_command(): Command 'init' failed with error code -100
User : 261 103 command.c:694 command_run_line(): 
Debug: 262 103 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 263 103 xtensa.c:2533 xtensa_target_deinit(): start
Debug: 264 118 server.c:609 sig_handler(): Terminating on Signal 22
```

### Expected behavior

JTAG flash runs without errors.

### Screenshots

[image]

erhankur, 2022-07-06:
@petemillermultimedia Can you follow the instructions written here?  https://github.com/espressif/openocd-esp32/issues/230#issuecomment-1134393392

petemillermultimedia, 2022-07-06:
My registry settings did not match, so I uninstalled the USB drivers and installed the recommended driver above. That produced registry settings that match those shown above, but even after a full restart it doesn't enable OpenOCD to run.

erhankur, 2022-07-06:
Btw, if you want to debug from builtin usb-jtag, you need to run openocd with `-f board/esp32s3-builtin.cfg`

closed 2022-07-07 by petemillermultimedia

B20221019, 2022-11-11:
I'm experiencing this behavior, can someone elaborate on how to recover my bricked ESP32-S3?