espressif/openocd-esp32 #236: Open the OpenOCD and flash with jtag occur error

WangShuoran, 2022-07-07:
### Development Kit

ESP32C3

### Module or chip used

ESP32C3

### Debug Adapter

Built-in JTAG

### OpenOCD version

v0.11.0-esp32-20211220

### Operating System

Windows

### Using an IDE ?

Vscode

### OpenOCD command line

board/esp32c3-builtin.cfg (vscode default)

### JTAG Clock Speed

default

### ESP-IDF version

v4.4.1

### Problem Description

在我用Vscode插件打开OpenOCD后，选择烧录模式为JTAG时，进行烧录，会报错：
When I open openocd with vscade plug-in and select JTAG as the burning mode, an error will be reported:

### Debug Logs

```shell
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
❌ Error: Algorithm timed out after 40001 ms.
❌ Error: ra = 0x40383c12
❌ Error: sp = 0x3fc847b0
❌ Error: gp = 0x0
❌ Error: tp = 0x0
❌ Error: t0 = 0x0
❌ Error: t1 = 0x40383bee
❌ Error: t2 = 0x0
❌ Error: s0 = 0x3fc94000
❌ Error: s1 = 0x1702aa49
❌ Error: a0 = 0x0
❌ Error: a1 = 0x0
❌ Error: a2 = 0x4
❌ Error: a3 = 0x0
❌ Error: a4 = 0x0
❌ Error: a5 = 0x0
❌ Error: a6 = 0x0
❌ Error: a7 = 0x403816e4
❌ Error: s2 = 0x66
❌ Error: s3 = 0x1
❌ Error: s4 = 0x3
❌ Error: s5 = 0x0
❌ Error: s6 = 0x0
❌ Error: s7 = 0xa0
❌ Error: s8 = 0x0
❌ Error: s9 = 0xa0
❌ Error: s10 = 0x14
❌ Error: s11 = 0x8
❌ Error: t3 = 0x0
❌ Error: t4 = 0x0
❌ Error: t5 = 0x0
❌ Error: t6 = 0x0
❌ Error: pc = 0x403803da
❌ Error: mstatus = 0x201800
❌ Error: mepc = 0x0
❌ Error: mcaus❌ e = 0x0
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
Warn : Failed to get flash mappings (-302)!
❌ Error: Target is already running ❌ an algorithm
Error: Failed to start algorithm (-4)!
Error: Failed to run flasher stub (-4)!
❌ Error: Target is already running an algorithm
Error: Failed to start algor❌ ithm (-4)!
Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
** Flashing Failed **
-1
```

### Expected behavior

在以前的v4.3.3中，就可以正常下载，并且可以编译，但是在v4.4.1中就有报错
In the previous v4.3.3, it can be downloaded normally and compiled, but there are errors in v4.4.1

erhankur, 2022-07-07:
@WangShuoran can you please try with [latest](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20220411) OpenOCD release? 

We are about the release the new version. If you also can try with master branch (attached), it could be helpful. 

[openocd-esp32-win32-0.11.0-esp32-20220706.zip](https://github.com/espressif/openocd-esp32/files/9061637/openocd-esp32-win32-0.11.0-esp32-20220706.zip)

WangShuoran, 2022-07-10:
我已经升级好这个最新版本：

我使用的是VScode的插件，插件的launch.json使用的我MS C插件试过，你们的适配器也试过，我先把launch.json发出来：
```
// MS C/C++ Extension
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "GDB",
            "type": "cppdbg",
            "request": "launch",
            "cwd": "${workspaceFolder}/build",
            "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
            "miDebuggerPath": "C:\\Espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32s3-elf\\bin\\xtensa-esp32s3-elf-gcc.exe",
            "setupCommands": [
                {
                    "text": "target remote :3333"
                },
                {
                    "text": "set remote hardware-watchpoint-limit 2"
                },
                {
                    "text": "mon reset halt"
                },
                {
                    "text": "thb app_main"
                },
                {
                    "text": "flushregs"
                }
            ]
            // "preLaunchTask": "OpenOCD-Running"
        }
    ]
}
// ESP-IDF Debug Adapter
// {
//     "version": "0.2.0",
//     "configurations": [
//         {
//             "type": "espidf",
//             "name": "Launch",
//             "request": "launch",
//             "debugPort": 9998,  // 请问这个DEBUG端口应该是什么?我该如何看
//             "logLevel": 2,
//             "mode": "manual",
//             "verifyAppBinBeforeDebug": false,
//             "tmoScaleFactor": 1,  // 这个我尝试过延时很长时间，没有用
//             "initGdbCommands": [
//                 "target remote :3333",
//                 "symbol-file /path/to/program.elf",  // 如果这个我不知道写 build/app.elf 还是 app.elf 还是该怎么写这个文件，我这个launch.json放在项目的.vscode文件夹，这个文件夹和build同一层
//                 "mon reset halt",
//                 "flushregs",
//                 "thb app_main"
//             ],
//             "env": {
//                 "CUSTOM_ENV_VAR": "SOME_VALUE"
//             }
//         }
//     ]
// }
```

使用MS C的调试器会报错为：
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
0
Info : Hart 0 unexpectedly reset!
Info : Reset cause (21) - (USB (UART) core reset)
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)

[image]

WangShuoran, 2022-07-13:
@erhankur

erhankur, 2022-07-18:
Is it OK when you load the application with the `esptool or idf.py flash`?  Can you attach the binaries and `flasher_args.json` file?

WangShuoran, 2022-07-22:
I am use vscode to flash application to board. that is default.

WangShuoran, 2022-07-22:
我能录个视频给你们看一下吗？这样可以快点找到问题所在，我看很多问题都不了了之，DEBUG功能在使用操作系统的场景比较重要，我不知道为什么我v4.4.1和最新版openocd不能DEBUG，而且OpenOCD还可以使用systemview等高阶任务处理程序，这对我也会是一个更好的成长过程。
Can I record a video for you to see? In this way, we can quickly find the problem. I think many problems are not solved. The debug function is more important in the scene of using the operating system. I don't know why I can't debug v4.4.1 and the latest version of openocd, and openocd can also use high-level task handlers such as Systemview, which will be a better growth process for me.

gerekon, 2022-07-28:
Hi @WangShuoran 

> // MS C/C++ Extension
1) According to your config you use `MS C/C++ Extension`, not ESP-IDF plugin for debugging. This is not default.

>             "miDebuggerPath": "C:\\Espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32s3-elf\\bin\\xtensa-esp32s3-elf-gcc.exe",
2) In your config you specify GCC instead of GDB as debugger. Also for some reason you use `xtensa-esp32s3-elf` toolchain which is for ESP32-S3 (Xtensa arch chip) not for ESP32-C3 (Riscv arch). 

> ** Verify OK **
3) From your log I see that flashing works well. Does debugging work after that?

gerekon, 2022-07-28:
@WangShuoran 
BTW, I see that you placed `phy_init_data.bin` at the default factory app location `0x10000`. Looks like you use custom partition table. By default OpenOCD chooses the first app image from it for debugging, but if you moved partition table too you need to tell OpenOCD where your app.bin is placed in flash directly via command `appimage_offset` as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints). You can do this by either 
- Specifying this command on OpenOCD command line. @brianignacio5 Could you tell us if it is possible to customize OpenOCD command line in VsCode IDF plugin for such cases?
- (Not sure that it works reliably). Execute it via GDB startup commands in config file (`initGdbCommands`), e.g.
   ```
   target remote :3333"
   mon esp appimage_offset <new_app_offset>
   disconnect
   target remote :3333"
   mon reset halt"
   thb app_main"
   ```

WangShuoran, 2022-08-06:
请问这个gdbinit中的配置和在手册中的openocd的配置在概念上一样吗？需要两个地方都要配置吗？
Is the configuration in gdbinit conceptually the same as the openocd configuration in the manual? Do you need to configure both places?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints
[image]

WangShuoran, 2022-08-06:
Now I want to use ESP-IDF Debug Adapter
My config is:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ESPIDF",
            "type": "espidf",
            "request": "launch",
            "mode": "auto",
            "debugPort": 43476,
            "logLevel": 2,
            "appOffset": "0x100000",
            "verifyAppBinBeforeDebug": false,
            "tmoScaleFactor": 10,
            "initGdbCommands": [
                "target remote :3333",
                "mon esp appimage_offset 0x100000",
                "set remote hardware-watchpoint-limit 2",
                "mon reset halt",
                "thb app_main",
                "flushregs"
            ],
            "env": {}
        }
    ]
}
```
I don't know how to config debugger and program

so, I get an error:
[image]
and ESP-IDF Debug Adapter:
DEBUG_ADAPTER_STARTED
2022-08-06 14:30:40,707 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2022-08-06 14:30:40,710 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
and openocd output:
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : Flash mapping 0: 0x100020 -> 0x3c080020, 97 KB
Info : Flash mapping 1: 0x120020 -> 0x42000020, 504 KB
Info : Using flash bank 'esp32c3.irom' size 508 KB
Info : Flash mapping 0: 0x100020 -> 0x3c080020, 97 KB
Info : Flash mapping 1: 0x120020 -> 0x42000020, 504 KB
Info : Using flash bank 'esp32c3.drom' size 100 KB
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)

WangShuoran, 2022-08-06:
还有，通过ESP-IDF Debug Adapter使用好在哪里？
Also, what are the advantages of using the esp-idf debug Adapter?

gerekon, 2022-08-12:
> target remote :3333"
> mon esp appimage_offset <new_app_offset>
> disconnect
> target remote :3333"
> mon reset halt"
> thb app_main"

You missed command in your debug adapter config. It should be like
```json
            "initGdbCommands": [
                "target remote :3333",
                "mon esp appimage_offset 0x100000",
                // reconnection is necessary to make GDB re-reading updated memory map
                "disconnect",
                "target remote :3333",
                "set remote hardware-watchpoint-limit 2",
                "mon reset halt",
                "thb app_main",
                "flushregs"
            ],
```

> Is the configuration in gdbinit conceptually the same as the openocd configuration in the manual? Do you need to configure both places?

Yes the config via GDB commands does the same thing.

> Also, what are the advantages of using the esp-idf debug Adapter?

Initially we had some problems with stepping using MS C/C++ debugger extension. So we had to implemented our own adapter. But now I've heard that people use MS one w/o problems, so maybe we can consider testing it again.
@brianignacio5  What do you think? Can we do this?

WangShuoran, 2022-08-13:
I have any error with an simple project test esp-debug-adapter, please analysis:
I use esp32c3 board, openocd open with vscode plugin. my launch.json:
```c
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ESPIDF",
            "type": "espidf",
            "request": "launch",
            "mode": "auto",
            "debugPort": 43474,
            "logLevel": 2,
            // "appOffset": "0x100000",
            "verifyAppBinBeforeDebug": false,
            "tmoScaleFactor": 10,
            "initGdbCommands": [
                // "mon esp appimage_offset 0x100000",
                "disconnect",
                "target remote :3333",
                "set remote hardware-watchpoint-limit 2",
                "mon reset halt",
                "thb app_main",
                "flushregs"
            ],
            "env": {}
        }
    ]
}
```
and openoce log is:
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
0
Info : Hart 0 unexpectedly reset!
Info : Reset cause (21) - (USB (UART) core reset)
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 73 KB
Info : Using flash bank 'esp32c3.irom' size 76 KB
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 73 KB
Info : Using flash bank 'esp32c3.drom' size 28 KB
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)

Debug adapter log is:
```c
DEBUG_ADAPTER_STARTED
2022-08-13 09:10:30,135 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2022-08-13 09:10:30,137 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
```

WangShuoran, 2022-08-13:
and my screen is:
[image]

gerekon, 2022-08-14:
> I have any error with an simple project test esp-debug-adapter, please analysis: I use esp32c3 board, openocd open with vscode plugin. my launch.json:
>
> ```c
> {
>     "version": "0.2.0", [...]

I do not see any problems in the logs and screens you sent. What the problem do you have?

At first I'd recommend you to test with one of our examples as described in [IDF VsCode Extension tutorials](https://github.com/espressif/vscode-esp-idf-extension#tutorials) w/o modifying adapter config file.

WangShuoran, 2022-08-15:
> I do not see any problems in the logs and screens you sent. What the problem do you have?

My screen show the debuger can not enter the breakpoint, I can not use next button etc...

> At first I'd recommend you to test with one of our examples as described in [IDF VsCode Extension tutorials](https://github.com/espressif/vscode-esp-idf-extension#tutorials) w/o modifying adapter config file.

Yes I see this and follow those show behavior. But I do not know why I have any error.

This is my example project:
[debug_c3.zip](https://github.com/espressif/openocd-esp32/files/9334853/debug_c3.zip)

brianignacio5, 2022-08-15:
You could increment the `logLevel` in the launch.json for the `espidf` debug adapter to see if there is any error in the debug adapter itself (logLevel 2 doesn't show much information) and there is `idf.openOcdDebugLevel` to increment the OpenOCD log level (if you are launching openOCD from the extension).

 I thought that `"appOffset": "0x100000",` in launch.json for the debug adapter's app offset launch argument for those custom partition table solved this case but we should add a setting to customize openOCD launch arguments.

Using the esp-idf adapter has the advantage that we can provide support for it and we are implementing more debugging features. The `cppdbg ` adapter is not maintained by us.

WangShuoran, 2022-08-17:
@brianignacio5 , thank you give me any tips. I follow those and show my log, and my vscode show:
[image]
It still not into breakpoint.

Debug Adapter:
```
2022-08-17 08:41:34,855 - Debug Adapter (main) - DEBUG - START_TIME_2022_08_17_08-41-34
2022-08-17 08:41:34,855 - Debug Adapter (main) - DEBUG - Init errors:None
2022-08-17 08:41:34,855 - Debug Adapter (main) - INFO - Working directory: C:\Users\xr\AppData\Local\Programs\Microsoft VS Code
2022-08-17 08:41:34,856 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': 'd:\\test_study\\debug_c3\\esp-idf-vscode-generated.gdb',
    'core_file': (),
    'debug': 5,
    'developer_mode': 'none',
    'device_name': 'esp32c3',
    'elfpath': ('d:\\test_study\\debug_c3\\build\\main.elf',),
    'log_file': 'd:\\test_study\\debug_c3\\debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 10,
    'toolchain_prefix': 'riscv32-esp-elf-'}
DEBUG_ADAPTER_STARTED
2022-08-17 08:41:34,856 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2022-08-17 08:41:34,856 - Debug Adapter (main) - INFO - Starting. Cmd: c:\Users\xr\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\debug_adapter_main.py -d 5 -e d:\test_study\debug_c3\build\main.elf -l d:\test_study\debug_c3\debug.log -p 43474 -dn esp32c3 -a 0x10000 -t riscv32-esp-elf- -x d:\test_study\debug_c3\esp-idf-vscode-generated.gdb -tsf 10

2022-08-17 08:41:34,857 - Debug Adapter (main) - INFO - Listening on port 43474
DEBUG_ADAPTER_READY2CONNECT
2022-08-17 08:41:34,857 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
2022-08-17 08:41:34,859 - Debug Adapter (main) - DEBUG - Got connection
2022-08-17 08:41:34,860 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_STARTED\n", "source": {}}, "seq": 0}

2022-08-17 08:41:34,860 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Starting. Cmd: c:\\Users\\xr\\.vscode\\extensions\\espressif.esp-idf-extension-1.4.0\\esp_debug_adapter\\debug_adapter_main.py -d 5 -e d:\\test_study\\debug_c3\\build\\main.elf -l d:\\test_study\\debug_c3\\debug.log -p 43474 -dn esp32c3 -a 0x10000 -t riscv32-esp-elf- -x d:\\test_study\\debug_c3\\esp-idf-vscode-generated.gdb -tsf 10\n\n", "source": {}}, "seq": 1}

2022-08-17 08:41:34,860 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Listening on port 43474\n", "source": {}}, "seq": 2}

2022-08-17 08:41:34,860 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT\n", "source": {}}, "seq": 3}

2022-08-17 08:41:34,860 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got connection\n", "source": {}}, "seq": 4}

2022-08-17 08:41:34,875 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 453\\r\\n'<<

2022-08-17 08:41:34,876 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2022-08-17 08:41:34,876 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "zh-cn",
        "pathFormat": "path",
        "supportsArgsCanBeInterpretedByShell": true,
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
[... 438 lines cut ...]
}
2022-08-17 08:41:41,563 - Debug Adapter(Command Processor) - DEBUG - Processed command: configurationDone
2022-08-17 08:41:41,563 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {},\n    \"command\": \"configurationDone\",\n    \"seq\": 5,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 23}

2022-08-17 08:41:41,563 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "response", "request_seq": 5, "success": true, "command": "configurationDone", "seq": 24}

2022-08-17 08:41:41,563 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Processed command: configurationDone\n", "source": {}}, "seq": 25}

2022-08-17 08:41:41,609 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 46\\r\\n'<<

2022-08-17 08:41:41,609 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2022-08-17 08:41:41,609 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "command": "threads",
    "seq": 6,
    "type": "request"
}
2022-08-17 08:41:41,610 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"command\": \"threads\",\n    \"seq\": 6,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 26}

2022-08-17 08:41:41,610 - Debug Adapter(Command Processor) - DEBUG - Processed command: threads
2022-08-17 08:41:41,610 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "response", "request_seq": 6, "success": true, "command": "threads", "body": {"threads": []}, "seq": 27}

2022-08-17 08:41:41,610 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Processed command: threads\n", "source": {}}, "seq": 28}

```

WangShuoran, 2022-08-17:
[openocd-log.txt](https://github.com/espressif/openocd-esp32/files/9355483/openocd-log.txt)

WangShuoran, 2022-08-17:
My log level: openocd-->3, adapter-->5

WangShuoran, 2022-08-18:
Can you reproduce my mistakes?
@brianignacio5 @gerekon

brianignacio5, 2022-08-18:
Could you provide a minimal example project to test the issue @WangShuoran ?

WangShuoran, 2022-08-18:
I update my vscode extention with 1.5.0, the error still exists

brianignacio5, 2022-08-18:
Hi @WangShuoran 

Without modification to your code I've tested your example project in MacOS with ESP32-C3-DevKitM-1 :
[image]
 This was using ESP-IDF v5.0, Python 3.9.6 and openOCD v0.11.0-esp32-20220706
 
 With ESP-IDF v4.4.2 and openOCD v0.11.0-esp32-20220411 seems to be failing too. Could you try using ESP-IDF master and latest openOCD to see if it works ?

gerekon, 2022-08-18:
> [openocd-log.txt](https://github.com/espressif/openocd-esp32/files/9355483/openocd-log.txt)

Looks like flash has no app image at `0x10000`. Is your app at that offset?

brianignacio5, 2022-08-18:
Also tested in Windows with ESP-IDF github master and openOCD v0.11.0-esp32-20220706 and is it working as shown here:
[image]

Only changes I've made were to change launch.json like this:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ESPIDF",
            "type": "espidf",
            "request": "launch",
            "mode": "auto",
            "logLevel": 2,
        }
    ]
}
```

WangShuoran, 2022-08-19:
> > [openocd-log.txt](https://github.com/espressif/openocd-esp32/files/9355483/openocd-log.txt)
> 
> Looks like flash has no app image at `0x10000`. Is your app at that offset?

I think it is. Because default (without patitions.txt and menuconfig change) project.bin is at 0x10000.

This is my burn with JTAG output:
```
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
Info : only one transport option; autoselect 'jtag'
[OpenOCD]
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
[OpenOCD]
Warn : Transport "jtag" was already selected
[OpenOCD]
Info : Listening on port 6666 for tcl connections
[OpenOCD]
Info : Listening on port 4444 for telnet connections
[OpenOCD]
❌ Error: libusb_open() failed with[OpenOCD]
 LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (60:55:F9:77:43:38)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[OpenOCD]
Info : datacount=2 progbufsize=16
[OpenOCD]
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
[OpenOCD]
Info : start[OpenOCD]
ing gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
[OpenOCD]
Flashing d:/test_study_v2/[OpenOCD]
debug_c3/build/bootloader/bootloader.bin at 0x0
[OpenOCD]
Info : JTAG tap: [OpenOCD]
esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[OpenOCD]
Info : [esp32c3] Found 8 triggers
[OpenOCD]
Info : Flash mapping 0: 0[OpenOCD]
x10020 -> 0x3c080020, 75 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 470 KB
[OpenOCD]
Info : Auto-detected fla[OpenOCD]
sh bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
[OpenOCD]
** Programming Started **
[OpenOCD]
Info : PROF: Data t[OpenOCD]
ransferred in 365.424 ms @ 54.7309 KB/s
[OpenOCD]
** Programming Finished **
[OpenOCD]
** Verify Start[OpenOCD]
ed **
[OpenOCD]
** Verified OK **
[OpenOCD]
** Flashing done for bootloader/bootloader.bin **
[OpenOCD]
Flashin[OpenOCD]
g d:/test_study_v2/debug_c3/build/main.bin at 0x10000
[OpenOCD]
Info : JTAG tap: esp3[OpenOCD]
2c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[OpenOCD]
** Programming Started **
[OpenOCD]
Info : PROF: Data tran[OpenOCD]
sferred in 2483.18 ms @ 59.601 KB/s
[OpenOCD]
** Programming Finished **
[OpenOCD]
** Verify Started **
[OpenOCD]
** Verified OK **
[OpenOCD]
** Flashing done for main.bin **
[OpenOCD]
Flashing [OpenOCD]
d:/test_study_v2/debug_c3/build/partition_table/partition-table.bin at 0x8000
[OpenOCD]
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 ([OpenOCD]
Espressif Systems), part: 0x0005, ver: 0x0)
[OpenOCD]
** Programming [OpenOCD]
Started **
[OpenOCD]
Info : PROF: Data transferred i[OpenOCD]
n 94.032 ms @ 42.5387 KB/s
[OpenOCD]
** Programming Finished **
[OpenOCD]
** Verify Started **
[OpenOCD]
** Verified OK **
[OpenOCD]
** Flashing done for partition_table/partition-table.bin **
[OpenOCD]
** Resetting Target **
[OpenOCD]
Info : JTAG tap: esp32c3.cpu tap/device found: 0x000[OpenOCD]
05c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[OpenOCD]
0

```

WangShuoran, 2022-08-19:
When I update vscode extension in v1.5.0. JTAG burn can jump to monitor. I don't know why.

WangShuoran, 2022-08-19:
> 

Now I use you config. It is OK. I don't know why I config detail but output error.

WangShuoran, 2022-08-26:
https://github.com/espressif/openocd-esp32/issues/236#issuecomment-1220201625

@brianignacio5 Can you meet this situation?

brianignacio5, 2022-08-26:
I'm sorry @WangShuoran , you mean after you flash you can't open the monitor? Or what is your issue ?

erhankur, 2024-01-15:
Closed due to inactivity.

closed 2024-01-15 by erhankur