espressif/openocd-esp32 #237: Many errors occured when debugging

wanggaoteng, 2022-07-14:
### Development Kit

ESP32

### Module or chip used

ESP32-WROOM-32

### Debug Adapter

CMSIS-DAP

### OpenOCD version

openocd-esp32-win32-0.11.0-esp32-20220411

### Operating System

windows

### Using an IDE ?

platformio

### OpenOCD command line

openocd -f board/esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v4.2

### Problem Description

In the end of #198 , wuxx shows a picture that CMSIS-DAP cannot debug ESP32, there are some errors. I had the same problem.
So the latest OpenOCD-esp32 doesn’t have that support compiled in in the offical binary
Can Expressif solve this problem? Let the people use CMSIS-DAP debugging ESP32.

### Expected behavior

Let the people use CMSIS-DAP debugging ESP32.

### Screenshots

[image]

erhankur, 2022-07-18:
We are fully aligned with the upstream. Also we started to create a patch for the Xtensa chips. I wonder if there is same issue. Any chance to try with the master branch? https://github.com/openocd-org/openocd

wanggaoteng, 2022-07-19:
> I am not familiar with the cmsis-dap adapters, yet. I will order one. Which one do you use?

Hi, thanks for reply.
I use the cmsis-dap made by [wuxx](https://github.com/wuxx/nanoDAP).
Best regards.

erhankur, 2022-08-16:
Now I have CMSIS-DAP adapter. I will return with the test results or fixes soon.

erhankur, 2022-08-18:
@wanggaoteng @wuxx  here is my test results.

- I have nano-dap, ordered from https://www.aliexpress.com/item/1005002334339283.html
- Tested on MacOS with the OpenOCD master branch. 
- I used ESP32-S2 as target with hello-world example
- Added 2 lines into the cmsis-dap.cfg
```
adapter driver cmsis-dap
cmsis_dap_vid_pid 0xc251 0xf001
cmsis_dap_backend hid
```
- Disabled watchdog timers from the idf menuconfig. My adapter communicates through wifi, so direct connected adapters might not have seen wdt reset issue. In that case wdt can stay enabled.
- Again due to the slowness, I have increased the reset timeout to the 1000ms from the code.  https://github.com/espressif/openocd-esp32/blob/master/src/target/espressif/esp32s2.c#L479

And the result

```
$OOCD_DIR/src/openocd -s $OOCD_DIR/tcl -f interface/cmsis-dap.cfg -f target/esp32s2.cfg -c "adapter speed 5000" 
Open On-Chip Debugger  v0.11.0-esp32-20220706-81-gffce4306 (2022-08-18-22:16)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: FW Version = 000000-170636
Info : CMSIS-DAP: Serial# = 00000000-00674F2E
Info : CMSIS-DAP: Interface Initialised (JTAG)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s2 on 3333
Info : Listening on port 3333 for gdb connections
```

reset command from telnet
```
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : cmsis-dap JTAG TLR_RESET
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Target halted, PC=0x400258A2, debug_reason=00000000
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x00000000, debug_reason=00000000
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
```

So far, cmsis-dap looks fine. I would like to continue to test full functionality but through wifi,  speed is not enough to run all commands. For example I couldn't upload a binary.  (Increasing LIBUSB_TIMEOUT might be helpful) 

```
> halt
[esp32s2] Target halted, PC=0x400258A2, debug_reason=00000000
>
> flash probe 0
CMSIS-DAP command 0x14 not implemented
CMSIS-DAP command CMD_DAP_JTAG_SEQ failed.
Connection closed by foreign host.
```

@wuxx Did you try any arm target from the jtag?  I wonder how this speed affects the whole debug process.

wuxx, 2022-08-19:
hello, it seems that the openocd send a CMSIS-DAP command 0x14, actually it was support by CMSIS-DAP standard, may be slow down the speed may helps? can you try a normal DAP Programmer like here for compare.
https://www.aliexpress.com/item/1005001376821868.html?spm=a2g0o.store_pc_groupList.8148356.5.6df89b43bWuknY&pdp_npi=2%40dis%21USD%21US%20%245.80%21US%20%245.80%21US%20%245.80%21%21%21%21%402132a21416608748885276320e1240%2112000016385698710%21sh&gatewayAdapt=4itemAdapt

erhankur, 2022-08-19:
@wuxx Can you try with normal DAP programmer and let me know if there is an issue during programming binaries? 

For now I can say ESP32 devices compatible to cmsis-dap adapter. (timeout issues are another story)

@wanggaoteng Can you also re-test and share your results?

erhankur, 2022-08-25:
@wuxx @wanggaoteng Since, I don't see a problem with the cmsis-dap protocol support, I am closing this issue. You can create another one if you see a problem during debugging or loading application.

closed 2022-08-25 by erhankur