espressif/openocd-esp32 #238: esp-idf gdb openocd DEBUGGING FILED

netanbi, 2022-07-17:
Hi,
I am trying to start work with the esp32-c3-devkitm-1 with the eclipse pluge in ,
first i test if i can burn the FW like gatt_server to board with the original build in usb and i seccueded ,
then i see that to work with debug i need to connect gpio18+19 ,gnd ,5v to usb pc 
i connect and run the command openocd -f board/esp32c3-builtin.cfg
from the terminal and secssed to get :

user-name@computer-name:~/esp/esp-idf$ openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.10.0-esp32-20201202-26-g05a036c2 (2021-03-15-17:18)
Licensed under GNU GPL v2
For bug reports, read
    https://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections

but the problem is if i run debuge then i get a lot of errors 

Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 6 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 6 options.c:63 configuration_output_handler(): 
Debug: 15 6 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 6 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 7 options.c:246 add_default_dirs(): exepath=C:/Users/nbitan/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 18 7 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 7 configuration.c:42 add_script_search_dir(): adding C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts
Debug: 20 8 configuration.c:42 add_script_search_dir(): adding C:/Users/nbitan/AppData/Roaming/OpenOCD
Debug: 21 8 configuration.c:42 add_script_search_dir(): adding C:/Users/nbitan/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 22 9 configuration.c:42 add_script_search_dir(): adding C:/Users/nbitan/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 23 9 command.c:146 script_debug(): command - gdb_port 3333
Debug: 25 10 command.c:146 script_debug(): command - telnet_port 4444
Debug: 27 10 command.c:146 script_debug(): command - tcl_port 6666
Debug: 29 10 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/board/esp32c3-builtin.cfg
Debug: 30 14 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/interface/esp_usb_jtag.cfg
Debug: 31 15 command.c:146 script_debug(): command - adapter driver esp_usb_jtag
Info : 33 15 transport.c:117 allow_transports(): only one transport option; autoselect 'jtag'
Debug: 34 16 command.c:146 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 36 16 esp_usb_jtag.c:899 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 37 16 command.c:146 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 39 16 esp_usb_jtag.c:912 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 40 17 command.c:146 script_debug(): command - adapter speed 40000
Debug: 42 17 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 43 17 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 44 17 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 45 18 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/target/esp32c3.cfg
Debug: 46 19 command.c:146 script_debug(): command - transport select jtag
Warn : 47 19 transport.c:286 jim_transport_select(): Transport "jtag" was already selected
Debug: 48 19 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/bitsbytes.tcl
Debug: 49 20 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/memory.tcl
Debug: 50 21 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/mmr_helpers.tcl
Debug: 51 22 configuration.c:97 find_file(): found C:\Users\nbitan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/target/esp_common.cfg
Debug: 52 23 command.c:146 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock are optional
Debug: 54 23 command.c:1115 help_add_command(): added 'program_esp' help text
Debug: 55 24 command.c:146 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [restore_clock]
Debug: 57 24 command.c:1141 help_add_command(): added 'program_esp' usage text
Debug: 58 24 command.c:146 script_debug(): command - add_help_text program_esp_bins write all the images at address specified in flasher_args.json generated while building idf project
Debug: 60 25 command.c:1115 help_add_command(): added 'program_esp_bins' help text
Debug: 61 25 command.c:146 script_debug(): command - add_usage_text program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress] [restore_clock]
[... 362 lines cut ...]
Error: 462 21439 libusb_helper.c:417 jtag_libusb_find_device(): libusb_open() failed with LIBUSB_ERROR_ACCESS
Debug: 463 21607 libusb_helper.c:411 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:4-22
Error: 464 21607 libusb_helper.c:417 jtag_libusb_find_device(): libusb_open() failed with LIBUSB_ERROR_ACCESS
Debug: 465 21783 libusb_helper.c:411 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:4-22
Error: 466 21783 libusb_helper.c:417 jtag_libusb_find_device(): libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: 468 21941 esp_usb_jtag.c:349 esp_usb_jtag_revive_device(): esp_usb_jtag: device not found!
Error: 469 21942 esp_usb_jtag.c:456 esp_usb_jtag_send_buf(): esp_usb_jtag: failed to revive USB device!
Error: 470 21944 bitq.c:285 bitq_execute_queue(): missing data from bitq interface
Error: 471 21945 riscv.c:399 dtmcontrol_scan(): failed jtag scan: -104
Debug: 472 21947 riscv.c:986 riscv_examine(): dtmcontrol=0xffffff98
Debug: 473 21948 riscv.c:988 riscv_examine():   version=0x8
Error: 474 21949 riscv.c:424 get_target_type(): Unsupported DTM version: 8
Debug: 475 21950 command.c:146 script_debug(): command - esp32c3 invoke-event examine-fail
Debug: 476 21951 command.c:146 script_debug(): command - esp32c3 invoke-event reset-assert-pre
Debug: 477 21953 command.c:146 script_debug(): command - transport select
Debug: 478 21954 command.c:146 script_debug(): command - esp32c3 cget -chain-position
Debug: 479 21956 command.c:146 script_debug(): command - jtag tapisenabled esp32c3.cpu
Debug: 480 21957 command.c:146 script_debug(): command - esp32c3 arp_reset assert 1
Debug: 481 21957 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
Debug: 482 21957 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas
Debug: 483 21958 riscv.c:1140 riscv_assert_reset(): [0]
Error: 484 21958 riscv.c:424 get_target_type(): Unsupported DTM version: 8
Debug: 485 21958 riscv.c:3209 riscv_invalidate_register_cache(): [0]

can someone help with that ???

erhankur, 2022-07-18:
Is this same with the hello_world example? It is worth trying to make sure your usb cable works well. 

And please can you also try with the latest OpenOCD [release](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20220706)

erhankur, 2022-08-16:
@netanbi Can you comment if your issue continues?

netanbi, 2022-08-16:
Hi after upgrade the OCD library its solved

closed 2022-08-16 by erhankur