espressif/openocd-esp32 #239: Debugger stepping into code on ESP32-S3 from Eclipse when I press F6 (step over)

project-x51, 2022-07-20:
### Development Kit

ESP32-S3-EYE

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

Direct via USB

### OpenOCD version

v0.11.0-esp32-20211220 (2021-12-20-15:43)

### Operating System

Windows 11

### Using an IDE ?

Eclipse CDT (Espressif-IDE)

### OpenOCD command line

Config Options in Eclipse -s ${openocd_path}/share/openocd/scripts -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default - 40000 kHz

### ESP-IDF version

V4.4

### Problem Description

From Eclipse
1. Create a project based on Hello World Example
2. Create a debug config for the project
3. Start debugging
4. Single step through the code using F6. 
 
The debugger will correctly step over the 1st printf("Hello world!\n); It will then step into the esp_chip_info(&chip_info); function even though we are single stepping. It then steps over the next printf calls before stepping into the spi_flash_get_chip_size() function. It skips the next printf call even though it has the esp_get_minimum_free_heap_size() function. It continues stepping over the next few calls before stepping into the vTaskDelay() function. 

Once it steps into the vTaskDelay() function it never leaves. Using F7 (step return) doesn't work and the debugging session ends or hangs. 

The process is 100% repeatable, every time. Similar issues occur in other examples.

 

### Debug Logs

```shell
Sorry no debug logs. Adding logging to the OpenOCD Config Options inside eclipse causes the debug session to timeout.

-l openocd_log.txt -d3 -s ${openocd_path}/share/openocd/scripts -f board/esp32s3-builtin.cfg

Error in services launch sequence
Starting OpenOCD timed out.
```

### Expected behavior

The code should not step into a function when I use the Step Over (F6) button. 

If I have stepped into a function the Step Return (F7) button should return me to the line/expression after the call.

In other words the debugger should work as all other debuggers do.

eren-terzioglu, 2022-07-26:
Seems it is working on Linux based environment. Did you download idf and idf-tools from Eclipse? If yes could you try these steps:
1) Remove Espressif add-on and its idf version
2) Make a clean Espressif add-on install
3) Install idf and idf-tools from Eclipse
4) Compile hello_world based project from Eclipse
5) Debug the sample again

If problem still happens, please notify.

erhankur, 2022-08-16:
@project-x51 Do you still have the issue?

project-x51, 2022-08-17:
Yes, sorry I have been using VSCode to make forward progress with my project and haven't had the time to re-visit the Eclipse plugin. I will attempt to do this once I have my revised boards off for manufacture in the next week or two.

erhankur, 2023-02-11:
Closed due to inactivity.

closed 2023-02-11 by erhankur