espressif/openocd-esp32 #240: Profiling crashes openocd: address_space >= 2

xobs, 2022-08-07:
### Development Kit

N/A

### Module or chip used

ESP32-S3

### Debug Adapter

USB

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

Windows

### Using an IDE ?

N/A

### OpenOCD command line

openocd -d2 -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

master

### Problem Description

1. Launch openocd
2. Telnet to port 4444
3. Run `profile 1 log.gconv`

### Debug Logs

```shell
Debug: 888 22316 xtensa.c:792 xtensa_smpbreak_set(): [esp32s3.cpu1] set smpbreak=0, state=2
Debug: 889 22318 xtensa.c:1200 xtensa_prepare_resume(): [esp32s3.cpu1] current=1 address=0x00000000, handle_breakpoints=0, debug_execution=0)
Debug: 890 22321 xtensa.c:572 xtensa_write_dirty_registers(): [esp32s3.cpu1] start
Debug: 891 22323 esp32s3.c:328 esp32s3_queue_write_dirty_user_regs(): esp32s3.cpu1: user regs writing is not implemented!
Debug: 892 22327 xtensa.c:650 xtensa_write_dirty_registers(): [esp32s3.cpu1] Writing back reg a3 value 00000000, num =4
Debug: 893 22330 xtensa.c:833 xtensa_core_status_check(): [esp32s3.cpu1] DSR (8000CC11)
Debug: 894 22332 xtensa.c:776 xtensa_smpbreak_write(): [esp32s3.cpu1] write smpbreak set=0x30000 clear=0x600000
Debug: 895 22335 xtensa.c:792 xtensa_smpbreak_set(): [esp32s3.cpu1] set smpbreak=30000, state=2
Debug: 896 22338 esp_xtensa_smp.c:342 esp_xtensa_smp_resume_cores(): [esp32s3.cpu1] begin
Debug: 897 22340 esp_xtensa_smp.c:369 esp_xtensa_smp_resume(): [esp32s3.cpu0] smp_break=0x30000
Debug: 898 22342 xtensa.c:776 xtensa_smpbreak_write(): [esp32s3.cpu0] write smpbreak set=0x0 clear=0x630000
Debug: 899 22345 xtensa.c:792 xtensa_smpbreak_set(): [esp32s3.cpu0] set smpbreak=0, state=2
Debug: 900 22347 xtensa.c:1200 xtensa_prepare_resume(): [esp32s3.cpu0] current=1 address=0x00000000, handle_breakpoints=0, debug_execution=0)
Debug: 901 22351 xtensa.c:572 xtensa_write_dirty_registers(): [esp32s3.cpu0] start
Debug: 902 22353 esp32s3.c:328 esp32s3_queue_write_dirty_user_regs(): esp32s3.cpu0: user regs writing is not implemented!
Debug: 903 22356 xtensa.c:650 xtensa_write_dirty_registers(): [esp32s3.cpu0] Writing back reg a3 value 00000000, num =4
Debug: 904 22360 xtensa.c:833 xtensa_core_status_check(): [esp32s3.cpu0] DSR (8000CC11)
Debug: 905 22362 xtensa.c:776 xtensa_smpbreak_write(): [esp32s3.cpu0] write smpbreak set=0x30000 clear=0x600000
Debug: 906 22365 xtensa.c:792 xtensa_smpbreak_set(): [esp32s3.cpu0] set smpbreak=30000, state=2
Debug: 907 22367 xtensa.c:1249 xtensa_do_resume(): [esp32s3.cpu0] start
Debug: 908 22370 xtensa.c:833 xtensa_core_status_check(): [esp32s3.cpu0] DSR (8000CC01)
Debug: 909 22372 target.c:1860 target_call_event_callbacks(): target event 2 (resumed) for core esp32s3.cpu0
Debug: 910 22375 esp32s3.c:561 esp32s3_handle_target_event(): 2
Debug: 911 22376 esp_xtensa_smp.c:608 esp_xtensa_smp_handle_target_event(): 2
Debug: 912 22378 esp_xtensa.c:80 esp_xtensa_handle_target_event(): 2
Debug: 913 22380 xtensa.c:2392 xtensa_handle_target_event(): 2
Debug: 914 22381 xtensa.c:1249 xtensa_do_resume(): [esp32s3.cpu1] start
Debug: 915 22383 xtensa.c:833 xtensa_core_status_check(): [esp32s3.cpu1] DSR (8000CC01)
Debug: 916 22385 target.c:1860 target_call_event_callbacks(): target event 2 (resumed) for core esp32s3.cpu1
Debug: 917 22388 esp32s3.c:561 esp32s3_handle_target_event(): 2
Debug: 918 22390 esp_xtensa_smp.c:608 esp_xtensa_smp_handle_target_event(): 2
Debug: 919 22392 esp_xtensa.c:80 esp_xtensa_handle_target_event(): 2
Debug: 920 22394 xtensa.c:2392 xtensa_handle_target_event(): 2
Debug: 921 22395 target.c:1860 target_call_event_callbacks(): target event 4 (resume-end) for core esp32s3.cpu1
Debug: 922 22398 esp32s3.c:561 esp32s3_handle_target_event(): 4
Debug: 923 22400 esp_xtensa_smp.c:608 esp_xtensa_smp_handle_target_event(): 4
Debug: 924 22401 esp_xtensa.c:80 esp_xtensa_handle_target_event(): 4
Debug: 925 22403 xtensa.c:2392 xtensa_handle_target_event(): 4
Info : 926 22421 target.c:2455 target_profiling_default(): Profiling completed. 4 samples.
Assertion failed!

Program: C:\Users\Sean\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin\openocd.exe
File: ../src/target/target.c, Line 4318

Expression: address_space >= 2
Debug: 927 22430 server.c:609 sig_handler(): Terminating on Signal 22
```

### Expected behavior

Profiling should not crash openocd

xobs, 2022-08-07:
Profiling has successfully completed one time, however most of the time it fails with the above assert.

xobs, 2022-08-08:
Note that specifying a `start` and `stop` appears to solve this crash:

```
openocd -f board/esp32s3-builtin.cfg -c 'init' -c 'profile 15 gmon.log 0x42000000 0x44000000'
```

Note, however, that I need to patch the linker script to rename the text section to `.text`, due to a limitation in how gprof works:

```patch
diff --git a/components/esp_system/ld/esp32s3/sections.ld.in b/components/esp_system/ld/esp32s3/sections.ld.in
index 13741b45ef..499346d8df 100644
--- a/components/esp_system/ld/esp32s3/sections.ld.in
+++ b/components/esp_system/ld/esp32s3/sections.ld.in
@@ -241,7 +241,7 @@ SECTIONS

   ASSERT(((_bss_end - ORIGIN(dram0_0_seg)) <= LENGTH(dram0_0_seg)), "DRAM segment data does not fit.")

-  .flash.text :
+  .text :
   {
     _stext = .;
     _instruction_reserved_start = ABSOLUTE(.);
@@ -275,16 +275,16 @@ SECTIONS
   } > default_code_seg

   /**
-   * This dummy section represents the .flash.text section but in default_rodata_seg.
+   * This dummy section represents the .text section but in default_rodata_seg.
    * Thus, it must have its alignment and (at least) its size.
    */
   .flash_rodata_dummy (NOLOAD):
   {
     _flash_rodata_dummy_start = ABSOLUTE(.);
-    /* Start at the same alignment constraint than .flash.text */
-    . = ALIGN(ALIGNOF(.flash.text));
-    /* Create an empty gap as big as .flash.text section */
-    . = . + SIZEOF(.flash.text);
+    /* Start at the same alignment constraint than .text */
+    . = ALIGN(ALIGNOF(.text));
+    /* Create an empty gap as big as .text section */
+    . = . + SIZEOF(.text);
     /* Prepare the alignment of the section above. Few bytes (0x20) must be
      * added for the mapping header. */
     . = ALIGN(0x10000) + 0x20;
```

erhankur, 2022-08-16:
Can you send your test application to reproduce the issue?

xobs, 2022-09-01:
Did you want me to send a compiled ELF file? I experience the crash eventually when running blink from https://github.com/espressif/esp-idf/tree/dfafa619c72865e769546f04157b7ad1648e24b2/examples/get-started/blink

You may need to run the program for longer, since it seems to be an issue of executing code that is in RAM as opposed to on flash. Or something along those lines. This should tickle the bug, but you may need to run it several times:

```
[9:45:48 am] ~/esp/esp-idf/examples/get-started/blink> openocd -f board/esp32s3-builtin.cfg -c 'init' -c 'profile 1 gmon.log' -c 'shutdown'
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : esp_usb_jtag: serial (F4:12:FA:47:27:CC)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : Starting profiling. Halting and resuming the target as often as we can...
Info : [esp32s3.cpu1] Target halted, PC=0x403793F2, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu1'
Info : [esp32s3.cpu0] Target halted, PC=0x403793F2, debug_reason=00000000
Info : [esp32s3.cpu1] Target halted, PC=0x403793F2, debug_reason=00000000
[...]
Info : [esp32s3.cpu0] Target halted, PC=0x403793F2, debug_reason=00000000
Info : [esp32s3.cpu1] Target halted, PC=0x403793F2, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu1'
Info : [esp32s3.cpu0] Target halted, PC=0x403793F2, debug_reason=00000000
Info : Profiling completed. 16 samples.
Assertion failed!

Program: C:\Users\Sean\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin\openocd.exe
File: ../src/target/target.c, Line 4318

Expression: address_space >= 2
[9:45:51 am] ~/esp/esp-idf/examples/get-started/blink> 
```

erhankur, 2024-01-13:
Today while I was looking at another profiler issue (https://github.com/espressif/openocd-esp32/issues/312) I tested the blink application with the latest OpenOCD. I didn't specify start and stop addresses. There's no issue so far. The generated report is something like this; 
```
xtensa-esp32s3-elf-gprof build/blink.elf gmon.out
Flat profile:

Each sample counts as 0.0769231 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  Ts/call  Ts/call  name    
 48.64      5.24     5.24                             gdma_append
 20.85      7.48     2.25                             esp_ptr_byte_accessible
 16.50      9.26     1.78                             gdma_default_tx_isr
 12.16     10.57     1.31                             esp_gpio_reserve_pins
  0.87     10.66     0.09                             esp_cpu_wait_for_intr
  0.48     10.72     0.05                             esp_task_wdt_init
  0.24     10.74     0.03                             esp_task_wdt_reset
  ```
  
  I am closing this issue. If you still have a problem, feel free to re-open.

closed 2024-01-13 by erhankur